morrownr/8821cu-20210916 #64: Slow connection and unpredictable behavior with a "Dual Band + Bluetooth USB adapter"

tuqueque, 2023-02-06:
Hello, I'm the user who reported issues in the previous/older driver page ( https://github.com/morrownr/8821cu-20210118/issues/89 )... Yesterday, I finally had a moment to uninstall the old driver and install this one. I had the small hope of fixing my issues with this new driver, but I'm having the same exact issues as before.

The main thing I'd love to solve is the slow connection. Let me know anything and thank you in advance!

morrownr, 2023-02-06:
Hi @tuqueque 

Yes, you are somewhere on my to-do list that somehow has grown out of control.

A little history: I have had users of this chipset that have multi-function adapters (wifi plus bt) that have had various issues in the past. The solution is to set a module parameter but I can't tell you the settings. You have to play with it.

Go to the driver directory and run:

sudo ./edit-options.sh

Look for the following section:

```
# Select RFE type ( rtw_RFE_type )
#
# 0 = (2-Ant, DPDT), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 1 = (1-Ant, SPDT@Ant1), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 2 = (1-Ant, SPDT@Ant1) , (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 3 = (1-Ant, DPDT@Ant2), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 4 = (1-Ant, DPDT@Ant2), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 5 = (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 6 = (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 7 = (1-Ant), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW) (try this setting first)
# 64 = this appears to be the default on adapters that do not support bluetooth
#
# Note: RFE Type is used to set antenna isolation and the BT coexistence
# mechanism. Some adapters require this setting and some do not. It may be
# necessary to try different settings to determine which setting is optimal
# for your adapter.
```

Read it. You will have to add `rtw_RFE_type=` to the `options` line near the beginning of the file. Try starting with:

rtw_RFE_type=7

Don't ask. Explaining rocket science would be easier.

Nick

tuqueque, 2023-02-06:
> Don't ask. Explaining rocket science would be easier.

Hahaha, loved this last remark!

So, let me repeat to see if I understood correctly:

1.- Run `sudo ./edit-options.sh`
2.- Enter somewhere `rtw_RFE_type=7`
3.- Profit.

I imagine if "7" doesn't work, I should try 1-6 to see if one of those other options fixes my issue, correct?

Also, I imagine I should reboot after running that script, correct?

morrownr, 2023-02-06:
1. yes, it is just a little script that starts a text editor and opens the correct file.
2. "somewhere" Okay, not good. There will be an uncommented line that starts with 'options. Add it to the end of that line.
3. I wish.

> I imagine if "7" doesn't work, I should try 1-6 to see if one of those other options fixes my issue, correct?

Exactly. It is called trial and error. Heavy on the error.

> Also, I imagine I should reboot after running that script, correct?

Correct. In fact, the script will try to get you to reboot.

tuqueque, 2023-02-06:
Great, everything is clear now!

I'll try all this in a couple of days when I'll have some spare time again. I'll let you know how it went.

Thank you VERY MUCH for your feedback and help!

xaxxontech, 2023-03-14:
I was having slowness issues with [this dongle](https://www.hardkernel.com/shop/wifi-module-5bk/) running ubuntu 22.04 kernel 5.18.0 
and this solution with rtw_RFE_type=4 seems to have solved things
-thanks

morrownr, 2023-03-17:
@xaxxontech 

Keep us posted on this if you have time.

It took some time to come up with this possible solution and it still is not obvious why Realtek would supply a chipset with this undesirable feature. 

This is only seen on multi-function (wifi + bt) adapters so I try to get the word out there to folks to try to find single-function adapters... which is not always easy with this chipset.

githunffh, 2023-03-18:
Hello I was able to connect my dongle but it was only Wi-Fi no Bluetooth using this GitHub pull request on my Ubuntu 22 desktop version this is my first time leaving a comment on here and I hope all my verbage is correct.

morrownr, 2023-03-18:
@githunffh 

No problem with the wordage.

It is fairly common that users of the multi-function (wifi and bt) version of this chipset have problems with either wifi or bt not working. Let me copy and paste from one of the messages above and you let me know if you are able to follow it:

```
Go to the driver directory and run:

sudo ./edit-options.sh

Look for the following section:

# Select RFE type ( rtw_RFE_type )
#
# 0 = (2-Ant, DPDT), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 1 = (1-Ant, SPDT@Ant1), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 2 = (1-Ant, SPDT@Ant1) , (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 3 = (1-Ant, DPDT@Ant2), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 4 = (1-Ant, DPDT@Ant2), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 5 = (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 6 = (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 7 = (1-Ant), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW) (try this setting first)
# 64 = this appears to be the default on adapters that do not support bluetooth
#
# Note: RFE Type is used to set antenna isolation and the BT coexistence
# mechanism. Some adapters require this setting and some do not. It may be
# necessary to try different settings to determine which setting is optimal
# for your adapter.
Read it. You will have to add rtw_RFE_type= to the options line near the beginning of the file. Try starting with:

rtw_RFE_type=7

Don't ask. Explaining rocket science would be easier.
```

This is trial and error at its best so you may be need to try several settings before you find the right one for your adapter.

morrownr, 2023-04-18:
@tuqueque 

Can I show this as solved?

tuqueque, 2023-04-19:
> @tuqueque
> 
> Can I show this as solved?

Hey, sorry for disappearing... Yes, I guess you can close this issue. Sadly, I couldn't make it work on my computer and I just bought another USB adapter with just WiFi... I still have the other one, maybe in several months I'll try it again.

Thank you very much for your time and effort working on this!

morrownr, 2023-04-19:
I understand. I encourage usb wifi adapter users to avoud multi-function adapters (wifi + bt) if they can and I think you understand why.

There is an in-kernel driver for this adapter now. It was merged in kernel 6.2 but had some performance problems. It is improving. I'd say that in a few months, it will be a good driver. You might give it a try in a few months.

I'm not sure if you are aware but this site has a Main Menu and several things are available.

https://github.com/morrownr/USB-WiFi

@morrownr

closed 2023-04-19 by morrownr