morrownr/8821cu-20210916 #65: Fun with Manjaro x86_64?

Jibun-no-Kage, 2023-02-07:
This was a bit of a surprise...

\# sudo ./install-driver.sh
Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.
Once the header files are properly installed, please run "sudo ./install-driver.sh"

So tried manual...

\# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.1-1-MANJARO/build M=/root/src/8821cu-20210916  modules
make[1]: Entering directory '/root/src/8821cu-20210916'
make[1]: *** /lib/modules/6.1.1-1-MANJARO/build: No such file or directory.  Stop.
make[1]: Leaving directory '/root/src/8821cu-20210916'
make: *** [Makefile:2501: modules] Error 2

Headers are installed...

\# pacman -Q | grep headers
linux-api-headers 5.18.15-1
linux419-headers 4.19.271-1
linux61-headers 6.1.9-1
vulkan-headers 1:1.3.240-1

\# uname -r 
6.1.1-1-MANJARO

\# ls -l /lib/modules/
total 12
drwxr-xr-x 3 root root 4096 Feb  6 21:21 4.19.271-1-MANJARO
drwxr-xr-x 4 root root 4096 Feb  6 21:39 6.1.9-1-MANJARO
drwxr-xr-x 2 root root 4096 Feb  6 21:28 extramodules-6.1-MANJARO

So something is not quite right, got a odd kernel version versus headers mismatch some how.

Jibun-no-Kage, 2023-02-07:
Anyway... appears that the delete of interface for wireless hangs...

\#ifconfig wls160u2 
wls160u2: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether d2:d0:c3:c3:1d:e3  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

\# iw dev wls160u2 del
^C^C^C^C^C^C^C

morrownr, 2023-02-07:
Hey @Jibun-no-Kage 

Since you seem to have 2 issues in one thread:

> Your kernel header files aren't properly installed.

That has popped up on me twice since we released this baby. Both times it was dead on. Once was on a freshly installed RasPiOS and a update/upgrade/reboot cleared the problem. It is nice having that in the script. Manjaro sometimes works in mysterious ways.

> # iw dev wls160u2 del

Should this go in the earlier issue about `del` hanging a system?

I haven't had time to explore this as I am neck deep trying to fix something else in another driver.

Jibun-no-Kage, 2023-02-07:
Yeah, I was thinking I had a new issue but not so.

closed 2023-02-07 by Jibun-no-Kage