morrownr/8821cu-20210916 #66: Information Request:  uEFI support on Raspberry Pi4 emerging?  Is this an issue we need to consider yet?

Jibun-no-Kage, 2023-02-10:
Information Request:  uEFI support on Raspberry Pi4 emerging?  Is this an issue we need to consider yet

https://github.com/pftf/RPi4

I only ask this in the context that users may ask same, so maybe we qualify our position on this now?  Just a one-liner in the documentation?  As I understand it, the maturity of uEFI on Pi4 is a long way from realization.

alkisg, 2023-02-11:
I've netbooted Raspberries using https://github.com/ipxe/pipxe in the past; that also uses the UEFI tianocore implementation.
I didn't notice anything different after the initial boot loader process, except for some glitches in the monitor colors. I don't think Wi-Fi modules will be affected, unless they start using secure boot...

morrownr, 2023-02-11:
> unless they start using secure boot...

I think secure boot is the problem that needs attention as far as this driver goes. dkms implements a solution and it works well as long as distros support the method it uses. Ubuntu and all the distros based on Ubuntu do support it and it works very well. Then there is the issue of the non-dkms installations. The installation script, as you know, currently checks if dkms is installed. If it is, it is used, if not, then a non-dkms installation is used and we currently do not have any support for secure boot if the non-dkms installation is used.

This is an issue that likely can be solved but time to concentrate on the issue is what is needed. One of the primary goals when I started this site was to help Linux users of as many distros as possible and you gents know that the lack of core standardization in many areas makes that a challenge.

Nick

Jibun-no-Kage, 2023-02-12:
I think maybe we just add a one-liner the documentation?  Just qualify what is or is not applicable to our solution?  Just was thinking of issues that we can address before a flood of questions.  Just trying to be a bit strategic versus tactical is all.

Jibun-no-Kage, 2023-02-12:
@alkisg As for PXE/iPXE, I helped design a PXE infrastructure and core standards for a Fortunate 10, before I retired.  Was a very interesting enterprise project/design, at the time we prototyped using forman, later integrated RedHat Satellite.  Was no small trick to get ESXi servers integrated to same, before VMware even thought that was an idea worth exploring.  But I digress.  I need to revamp my home lab iPXE staging, but plan to use it for my Pi4s of course.  I hope to some day have a simple Pi CM cluster frame to goofy around with, would love to use iPXE to spin it up and down as needed.  That will have to wait until Pi devices are not insanely priced... humm maybe use ThinClients until then?  Oops I digress yet again.

morrownr, 2023-02-19:
We can address this in the future. The monitor mode shit is priority for now. Closing.

closed 2023-02-19 by morrownr