morrownr/8821cu-20210916 #67: `rtw_acs_info_dump()` flooding kernel log

rahmanuh, 2023-02-16:
Hi,

The `rtw_acs_info_dump()` floods the kernel log with:
```
...
[56768.132025] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56768.139754] ========== ACS (VER-3) ==========
[56768.139762] Best 24G Channel:4
[56768.139766] Best 5G Channel:40
[56768.139769] ==================================
[56772.484590] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56772.492430] ========== ACS (VER-3) ==========
[56772.492432] Best 24G Channel:4
[56772.492433] Best 5G Channel:40
[56772.492434] ==================================
[56779.388062] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56779.396155] ========== ACS (VER-3) ==========
[56779.396158] Best 24G Channel:4
[56779.396159] Best 5G Channel:40
[56779.396160] ==================================
[56789.388269] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56789.396108] ========== ACS (VER-3) ==========
[56789.396111] Best 24G Channel:4
[56789.396112] Best 5G Channel:40
[56789.396113] ==================================
[56803.396043] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56803.405948] ========== ACS (VER-3) ==========
[56803.405956] Best 24G Channel:1
[56803.405960] Best 5G Channel:40
[56803.405963] ==================================
[56823.388166] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56823.396374] ========== ACS (VER-3) ==========
[56823.396377] Best 24G Channel:4
[56823.396379] Best 5G Channel:40
[56823.396380] ==================================
[56852.387975] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56852.396062] ========== ACS (VER-3) ==========
[56852.396071] Best 24G Channel:5
[56852.396075] Best 5G Channel:40
[56852.396079] ==================================
[56895.389173] RTW: wlp0s20f0u9- hw port(0) mac_addr =06:76:b7:4b:c3:ff
[56895.397378] ========== ACS (VER-3) ==========
[56895.397387] Best 24G Channel:3
[56895.397401] Best 5G Channel:40
...
```
I wonder if there is any specific purpose to show these logs for user? Would it be possible to suppress these logs?

morrownr, 2023-02-16:
Hi @rahmanuh 

Yes, pretty worthless. Let me see what I can do.

morrownr, 2023-02-16:
@rahmanuh 

I think that is the simplest overall solution. Let me test it and then I'll merge. Thanks.

closed 2023-02-17 by morrownr