morrownr/8821cu-20210916 #70: Loading of unsigned module is rejected, secureboot is disabled

vamzii, 2023-02-19:
hi mate, thanks for your awesome driver, was using from the past 5 months,
today after updating my pop os installation, wifi stopped working, and this is the dmesg output i am getting,
`[  +0.000006] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0.000002] usb 1-1: Product: 802.11ac NIC
[  +0.000002] usb 1-1: Manufacturer: Realtek
[  +0.000001] usb 1-1: SerialNumber: 123456
[  +0.032954] Loading of unsigned module is rejected
`

this is the device i am currently having
Bus 001 Device 013: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

did i screw anything?
secureboot is currently disabled.

vamzii, 2023-02-19:
sorry got confused between trusted computing and secureboot, disabled it and it started working :)
made a video for the same, will share if people are still facing this issue thanks dev for the driver, you can close the issue

morrownr, 2023-02-19:
Sure, share the video.

@morrownr

vamzii, 2023-05-25:
extremely sorry its been a while, took me time to edit it,
https://www.youtube.com/watch?v=ScfYxXrJZlM