morrownr/8821cu-20210916 #74: Extremly unstable work on arch linux, kernel 6.2, Mercusys mu6h wi-fi adapter

Eyjafja, 2023-03-08:
More often than not, I can't even connect even though the list of networks is available, but still, when I can connect to the network, it does not work for long

morrownr, 2023-03-09:
Hi @Eyjafja 

We need more info. See pinned issue called `(Info) Problem Reporting Guideline`.

Eyjafja, 2023-03-10:
> Hi @Eyjafja
> 
> We need more info. See pinned issue called `(Info) Problem Reporting Guideline`.
###  Sorry. `There is the install-driver.sh `output:

`: ---------------------------
: [install-driver.sh](https://github.com/morrownr/8821cu-20210916/issues/74) v20230227
: x86_64 (architecture)
: 12/12 (in-use/total processing units)
: 32742852 (total system memory)
: 6.2.2-arch1-1 (kernel version)
: gcc (GCC) 12.2.1 20230201
: dkms-3.0.10
: ---------------------------

Checking for previously installed drivers.
rtl8821cu/[5.12.0.4](https://github.com/morrownr/8821cu-20210916/issues/74), 6.2.2-arch1-1, x86_64: installed
: ---------------------------

Removing a dkms installation.
Module rtl8821cu-5.12.0.4 for kernel 6.2.2-arch1-1 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

8821cu.ko.zst:
 - Uninstallation
   - Deleting from: /usr/lib/modules/6.2.2-arch1-1/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod...
Deleting module rtl8821cu-5.12.0.4 completely from the DKMS tree.
Removing 8821cu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821cu-5.12.0.4
Removal complete.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4
Creating symlink /var/lib/dkms/rtl8821cu/[5.12.0.4/source](https://github.com/morrownr/8821cu-20210916/issues/74) -> /usr/src/rtl8821cu-5.12.0.4
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/lib/modules/6.2.2-arch1-1/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/[mok.pub](https://github.com/morrownr/8821cu-20210916/issues/74)

Building module:
Cleaning build area...
./[dkms-make.sh](https://github.com/morrownr/8821cu-20210916/issues/74)........
Signing module /var/lib/dkms/rtl8821cu/[5.12.0.4/build/8821cu.ko](https://github.com/morrownr/8821cu-20210916/issues/74)
Cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

8821cu.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /usr/lib/modules/6.2.2-arch1-1/updates/dkms/
depmod...
The driver was installed by dkms successfully.`

### And `sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get` output:
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 006: ID 2717:ff80 Xiaomi Inc. Mi/Redmi series (RNDIS)
Bus 001 Device 002: ID c0f4:02e0 USB usb keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
 Soft blocked: no
 Hard blocked: no
rtl8821cu/[5.12.0.4](https://github.com/morrownr/8821cu-20210916/issues/74), 6.2.2-arch1-1, x86_64: installed
phy#0
 Interface wlan0
  ifindex 3
  wdev 0x1
  addr 52:7c:d3:46:2c:bf
  type managed
  txpower -100.00 dBm
global
country 00: DFS-UNSET
 (2402 - 2472 @ 40), (6, 20), (N/A)
 (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
 (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
 (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
 (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
 (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
 (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
 (57240 - 63720 @ 2160), (N/A, 0), (N/A)

morrownr, 2023-03-10:
@Eyjafja 

Thanks for the additional information. It helps rule out many things. It looks like you have a solid installation so maybe we need to move on to the environment.

Are you using a desktop system?
If so, which port is the adapter plugged into? Front or back?
Do you use an extension cable?

Would you please install `wavemon`?

sudo apt install wavemon

That is the Debian/Ubunto/RasPiOS way but you did not say which distro you use.

You can run `wavemon` in a terminal by just typing `wavemon`.

What are your `link quality` and `signal level` numbers?

Eyjafja, 2023-03-10:
Yes, I'm using desktop system.

Adapter was In front port.

No, I don't use any extension cabels

I sayed in theme, that my system is arch and my adapter is mecrusys mu6h(AC650). 

I will install wavemon later, when i get home

P.S. i think some problems are caused by kernel. Because it all started with the fact that I did a complete system update, including the kernel.  After that, I got some wifi driver from the kernel, but it caused hangs in the KDE wireless network management menu.  And also an attempt to call ip link in the terminal cause the terminal freezing(and ctrl+c or ctrl+d couldn't help)

morrownr, 2023-03-10:
> i think some problems are caused by kernel. Because it all started with the fact that I did a complete system update, including the kernel. After that, I got some wifi driver from the kernel, but it caused freezes in the KDE wireless network management menu.

I think we may be making progress. Kernel 6.2 has an in-kernel driver for your adapter. My driver installation script will blacklist the internal driver as long as the system is rebooted after running the installation script. It will also do away with the blacklist if you use the remove-driver.sh script for removal.

This information along with the description of the problem tells me this could be a situation where the 2 drivers are in a fight. If you run `sudo sh remove-driver.sh` and reboot, it should allow the in-kernel driver to take over.

I have tested the in-kernel driver. Performance is not so good yet. I'm working the work on the linux-wireless mailing list and I think it will eventually be a good driver. You can try it by removing this driver. If the performance does not yet meet your needs, then install this driver and reboot. You can update you local folder with

$ git pull
then run
$ sudo sh install-driver.sh
then run
$ sudo reboot

Tell me what happens.

Eyjafja, 2023-03-10:
No, you misunderstood me.  I installed your driver already after problems with the driver in the kernel, and your driver is much better, than in-kernel. In-kernel cause freezes and i can't reboot or off my pc with working in-kernel driver. for this reason I disabled it and later installed your driver.

morrownr, 2023-03-10:
Ah. Go ahead and install `wavemon` when you are at home and let's see what we have.

Eyjafja, 2023-03-10:
> Ah. Go ahead and install `wavemon` when you are at home and let's see what we have.
[image]

morrownr, 2023-03-10:
F7 will allow you to change interface then F1 sends you back to the main screen.

Eyjafja, 2023-03-10:
I have only one interface

morrownr, 2023-03-10:
What does the following show?

iw dev

morrownr, 2023-03-10:
Connect to an access point and check `wavemon` again.

Eyjafja, 2023-03-10:
i can't connect, it's not work

morrownr, 2023-03-10:
This is not good.

The driver is loading or you would not have an interface when you execute `iw dev`.

Does Arch have a way to `forget connections`? If so, forget the old connections and try to connect again.

Eyjafja, 2023-03-10:
It doesn't help. 

I think powersave mode can cause this, how can i turn off powersave mode?

morrownr, 2023-03-10:
By the way, what channel are you trying to connect to?

Eyjafja, 2023-03-10:
i am trying connect to channel six, changing options doesn't help

Eyjafja, 2023-03-10:
idk how, i just connected to my wifi. Here is wavemon
[image]

morrownr, 2023-03-10:
Your signal level and link quality are certainly not a problem.

Can you attempt a 5 GHz band channel such as 36?

Eyjafja, 2023-03-10:
I attempted channel 132, which is also 5 GHz. Internet is only exists for a few seconds, just like before

morrownr, 2023-03-10:
Did you have a good connection before you upgraded the system?

I am at a loss for now.

Eyjafja, 2023-03-10:
I didn't have any drivers before system upgrade. 

I have dual boot arch with windows, on windows connection is good, so wi-fi adapter is ok

morrownr, 2023-03-10:
This might have been a major upgrade that messed some things up. Have you checked the Arch forums to see if anyone else is seeing this or something similar?

You might post about this while at the Arch forums are well. 

This repo is cloned over 200 times per day and, as you can see, there are not a lot of issues being reported so my best guess is that it could be something Arch specific.

Eyjafja, 2023-03-10:
I did not find anything like that, I found only that problem with kernel driver freezes (although at the time of writing those threads, it was not yet in the kernel), but it was also not common.

I will try to write about it on forum. 

Thank you for taking the time.

Eyjafja, 2023-04-12:
this problem was solved by updating to kernel 6.2.6, after that the driver was stable

closed 2023-04-12 by Eyjafja