morrownr/8821cu-20210916 #75: RTW: rtw_suspend_normal: ### ERROR #### driver in IPS ####ERROR###!!!

fdservices, 2023-03-10:
Recently started to get this message in the logs.

Arch Linux acer 6.2.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 03 Mar 2023 15:58:31 +0000 x86_64 GNU/Linux
Name             : rtl8821cu-morrownr-dkms-git
Installed         : 5.12.0.4.129.8c45ce6-1
Install Date    : Fri 24 Feb 2023 08:32:15 CET

Bluetooth/Wireless:
    0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC
    Realtek 8821cu Wireless LAN 802.11ac USB NIC

Is this due to the driver?

morrownr, 2023-03-10:
@fdservices 

> Is this due to the driver?

That is a good question. I am booted into kernel 6.2.1 but I am on an Ubuntu based distro. I am not seeing this.

I need you to clarify where you got this driver because this driver ends up all over the place as this repo seems to be the core upstream for every distro that repackages their own. The only problem is that this repo is updated very often and the downstream repackaged versions do not get updated very often.

If you got this from what I think is called AUR for Arch based distros, before I can help with any degree of certainty, I need you to remove the version from AUR and directly install the version here. I hope you understand.

@morrownr

fdservices, 2023-03-11:
Thank you for the quick reply. I uninstalled the AUR version and rebooted. The module was no longer listed. 

I then installed the git version from here, using the 'install-driver' script and kept the default settings. I rebooted and checked that the module was installed.

I then suspended the PC and checked the errors after waking from sleep. Still there I am afraid.

[Sat Mar 11 08:40:20 2023] RTW:  suspend start\n
[Sat Mar 11 08:40:20 2023] RTW: rtw_suspend_normal: ### ERROR #### driver in IPS ####ERROR###!!!
[Sat Mar 11 08:40:20 2023] RTW: rtw_dev_unload: driver in IPS-FWLPS
[Sat Mar 11 08:40:20 2023] RTW: rtw suspend success in 4 ms
[Sat Mar 11 08:40:22 2023] RTW: resume start
[Sat Mar 11 08:40:22 2023] RTW: rtw_resume_common:0 in 640 ms
[Sat Mar 11 08:40:28 2023] RTW: wlp0s20u2u2i2- hw port(0) mac_addr =1c:bf:ce:66:00:85
[Sat Mar 11 08:40:28 2023] RTW: rtw_set_802_11_connect(wlp0s20u2u2i2)  fw_state=0x00000008
[Sat Mar 11 08:40:29 2023] RTW: start auth

FWIW the AUR version downloads your code directly from your site, so it should be up to date.

morrownr, 2023-03-12:
Hi @fdservices 

> FWIW the AUR version downloads your code directly from your site, so it should be up to date.

Do you know how often it downloads?

Regarding the error: That code is complicated and I'm not seeing the error here so it might be best to test the settings to see if we can find one that works well for your system.

sudo ./edit-options.sh

Test the other settings for `rtw_power_mgnt=1` and see what happens.

If that does not help, then I have another idea.

fdservices, 2023-03-14:
@morrownr Copy of email follows:
>
>
> Do you know how often it downloads?
Every time I update the driver

> Test the other settings for rtw_power_mgnt=1
Checked 0 and 2, both gave the same error after a reboot

I assume that there is nothing in the fact that the error states "driver in IPS" and the next line in the log is "RTW: rtw_dev_unload: driver in IPS-FWLPS" ?

fdservices, 2023-04-18:
No, the error is still reported

morrownr, 2023-04-18:
This is going to require some investigation.

Have you tested on another system?

Can you run and post the results of the following:

$ ls /sys/module/8821cu/parameters/

For what it is worth, I just checked the usage data; This repo is averaging 500+ clones per day. If this was a general problem, I would expect to see a lot of similar reports so my best guess at this point is that the problem could be hardware related. In other words, some specific hardware may be causing something that is causing this. It could also be distro specific. I think our best bet is to look at all of the module parameters and maybe some of the related code to see if one of module parameters could mediate this.

fdservices, 2023-04-19:
Sorry but I have no way of testing on another system.
Although it is clearly reported as an error, I have not seen any negative effect.
[parameters.txt](https://github.com/morrownr/8821cu-20210916/files/11270255/parameters.txt)

morrownr, 2023-04-19:
> Although it is clearly reported as an error, I have not seen any negative effect.

If you are not seeing any negative effects then how about we leave it alone for now. I'll close the issue but if you do see negative effects, come back and let me know.

Keep in mind that there is now an in-kernel driver for this chipset. Maybe once you get to kernel 6.4 or later, you should give it a try.

@morrownr

closed 2023-04-19 by morrownr