morrownr/8821cu-20210916 #77: Authentication issues on RTL8811CU

greengeckowizard, 2023-03-15:
Hello, I am experiencing difficulties connecting to wifi networks on my Linux Mint 21.1 system running Linux kernel version 6.1.18 xanmod v3. I have also attempted to use other kernel versions in the 6 range, but they all seem to exhibit the same issue. I am using the UGREEN AC650 wifi adapter (https://www.amazon.com/UGREEN-Adapter-Desktop-Wireless-Computer/dp/B08M9FX4G3), which features the rtl8811CU chipset. I have confirmed the SSID and WPA2 authentication password, but have been unable to establish a connection. I have attempted to enable usb_modeswitch and also tried with it disabled, but this did not resolve the issue. While my adapter can detect all available networks, it remains unable to connect.

greengeckowizard, 2023-03-15:
I just saw the instructions here's my output when I run those commands.
[image]

greengeckowizard, 2023-03-15:
and for the output on the installation it installs just fine.
[image]

greengeckowizard, 2023-04-18:
> @greengeckowizard
> 
> I've been very bust lately. Has this been resolved?

Still having issues even with a different router I've tested.

greengeckowizard, 2023-04-18:
Basically it goes into am authentication loop where it doesn't want to establish a connection and it keeps saying the password is incorrect when I've tested with different ssids on my network and also passwords for those ssids and still is having the same issue. I have also confirmed the password and everything so it's not that.

morrownr, 2023-04-18:
Are you using the Cinnamon version of Mint? If so, the wifi support is not the greatest, I would know.

 If you select Network Connection from the little plug-in in the panel, then delete the WiFi connections. Sometimes it helps to "forget" connections.

On the wifi routers/APs you are using, do they have separate SSID's for 2.4 and 5 GHz bands? If not it would be a good idea to make that so.

What security level are you using in the router? WPA2/WPA3/Mixed

greengeckowizard, 2023-04-18:
Yes I'm using cinnamon version of Mint and I have also tested with Ubuntu still having the same issue. My router is using separate ssids for different bands 2.4ghz and 5ghz band. And the security level is WPA2

greengeckowizard, 2023-04-18:
I have also tested with Zorin OS same issue

morrownr, 2023-04-18:
Are you having the same issue on both bands?

greengeckowizard, 2023-04-18:
> Are you having the same issue on both bands?

Yes I have tried both bands and they still have the same issue

morrownr, 2023-04-18:
I just reread the earlier messages you posted and I cannot see anything that provides any help at this point. Everything I have read tells me the driver is installed and should be working. 

Generally the problem you are describing is related to security. I can confirm that this driver supports WPA2 and WPA3. Is there a problem with wpa_supplicant or IWD? Possibly. Is it an issue with settings in the routers? Possibly.

Run:

$ sudo save-log.sh

...and post the results in the file it creates.

You could also run and look for usb and rtw issues in:

$ sudo dmesg

What happens if you run:

$ iw reg set US

...replace the US with your country code and then see if you can connect.

Also, I have a section with recommended wifi router/AP settings in the README.md. You might want to check it out just in case.

greengeckowizard, 2023-04-18:
Here's my outputs when I run all of those commands.

[dmesg.txt](https://github.com/morrownr/8821cu-20210916/files/11266510/dmesg.txt)
[iw.txt](https://github.com/morrownr/8821cu-20210916/files/11266511/iw.txt)
[rtw.log](https://github.com/morrownr/8821cu-20210916/files/11266512/rtw.log)

morrownr, 2023-04-18:
hmmm...

dmesg is saying:

secureboot: Secure boot enabled

dkms is saying:

Secure Boot is not enabled on this system.

Somewhere in here is likely the truth behind the problem. You seem to be running a kernel that is not a Ubuntu signed kernel. 

A little advice: if you are going to leave secure boot on, you need to be very careful what you are doing or you will end up with things that do not work. If you were running a clean Ubuntu or Mint distro, install-driver.sh/dkms should have signed the driver for you during installation if secure boot was set up properly during installation of the distro. What you have is a little bit of a mess. The best advice I can give you is to backup the important files and do a clean Ubuntu or Mint installation and follow the instructions during the installation for setting up secure boot... or if you really don't need secure boot, do a clean installation with secure boot turned off.

I don't know the history of how things have been done so my general advice is the best I can do.

morrownr, 2023-04-18:
> What custom kernels work with this driver just curious (I would really like to use custom kernel)?

None if you are going to keep secure boot on. What you are doing will shut down a lot, if not all, drivers that need to be compiled. Well, you may be able to manually sign the drivers but that is beyond the scope of support I try to provide here.

My opinion is that you need to make a decision:

1. Go into the bios and turn secure boot off and play with custom kernels.
or
2. Keep secure boot on and forget about using custom kernels.

The issue here is secure boot. Secure boot was designed for specific purposes. You need to study up on secure boot so that you understand what you can and cannot do.

Good luck.

greengeckowizard, 2023-04-18:
> None if you are going to keep secure boot on. What you are doing will shut down a lot, if not all, drivers that need to be compiled. Well, you may be able to manually sign the drivers but that is [...]

I just turned secure boot off thank you I'll see if it works now.

morrownr, 2023-04-19:
Status?

greengeckowizard, 2023-04-19:
Hi its doing the same thing where it keeps trying to authenticate and tries to reconnect but it can't. I did make sure secure boot is disabled and it is disabled. I ran  sudo mokutil --sb-state  and it says secure boot is disabled. The issue is that it just won't connect. I'm on linux kernel 5.15.0-70 generic on Mint Cinnamon

greengeckowizard, 2023-04-19:
Here's my updated dmesg and logs 
[dmesg.txt](https://github.com/morrownr/8821cu-20210916/files/11277535/dmesg.txt)
[rtw.log](https://github.com/morrownr/8821cu-20210916/files/11277536/rtw.log)

greengeckowizard, 2023-04-20:
I have noticed the driver is kinda on and off like it works one time after the initial install reboot and then after you shut down or restart its sometimes it functions correctly like it can authenticate and then usually after the third reboot it works again very strange.

greengeckowizard, 2023-04-22:
The issue is also occurring in Fedora 38

morrownr, 2023-04-22:
I'm going to keep an eye out for this. It sure would be nice to have a surefire way to reproduce it.

greengeckowizard, 2023-04-22:
> I'm going to keep an eye out for this. It sure would be nice to have a surefire way to reproduce it.

I've noticed it's been doing it with even my other rtl8811cu adapter and that's not an branded one it's a cheap ebay adapter with that chipset

morrownr, 2023-04-22:
When you are able to run kernel 6.2 or later, uninstall this driver and run with the in-kernel driver to see if it happens there.

greengeckowizard, 2023-04-22:
> When you are able to run kernel 6.2 or later, uninstall this driver and run with the in-kernel driver to see if it happens there.

Sure I'll give it a test I'm on Fedora at the moment. Where would I install this in-kernel driver?

morrownr, 2023-04-23:
> Where would I install this in-kernel driver I dont see anything about it when I search for documentation on this?

You don't install it. That is why it is called in-kernel. Let me give you the link to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Scroll down and start reading with menu item 1, then 2, etc. There is a lot to learn.

I've been testing on kernel 6.3 and the improvements are coming rapidly.

greengeckowizard, 2023-04-23:
> You don't install it. That is why it is called in-kernel. Let me give you the link to the Main Menu of this site:
>
> https://github.com/morrownr/USB-WiFi
>
> Scroll down and start reading with menu item [...]

Thanks will test now.

morrownr, 2023-04-23:
You do have to run the following to uninstall this driver:

sudo ./remove-driver.sh

This driver shuts down the in-kernel driver to prevent conflicts.

greengeckowizard, 2023-04-23:
Hmm for my specific adapter I don't see anything for the in-kernel driver on that page.

evshiron, 2023-04-27:
Greetings.

I have the same UGREEN adapter and the same issue occurs to me too.

Somehow I manage to establish connections with several minutes' try and error each time, and it works just fine after that.

I am using Kubuntu without Secure Boot. Most of the time it prompts the password dialog again and again, and generates "No secrets were provided" notification when it fails.

A log of `dmesg` is attached and @morrownr might be interested.

[dmesg3.log](https://github.com/morrownr/8821cu-20210916/files/11348033/dmesg3.log)

morrownr, 2023-04-28:
I just installed this driver on a fresh installation of Ubuntu 23.04.
System is x86_64.
Kernel is 6.2
gcc is 12.2

I use the `save-log.sh` script that comes with the driver so as to only get `RTW` lines in the rtw.log:

```
RTW: module init start
 RTW: rtl8821cu v5.12.0.4-1-g9241a6516.20210916_COEX20200730-5151
 RTW: rtl8821cu BT-Coex version = COEX20200730-5151
 RTW: [HALMAC]55772M
 RTW: HW EFUSE
 RTW: 0x000: 29 81 00 BC  09 00 31 00  6F 04 A4 10  10 00 30 0B  
 RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF  
 RTW: 0x020: FF FF 29 29  2C 2B 24 24  24 23 23 23  22 22 22 22  
 RTW: 0x030: 22 FF FF FF  FF FF 10 FF  FF FF 2B 2B  2B 2B 2B 2B  
 RTW: 0x040: 2C 2C 2C 2C  2C 23 FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 1A 1A 00  FF FF FF FF  
 RTW: 0x0C0: FF 21 00 41  00 00 00 00  00 FF 22 FF  FF FF FF FF  
 RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x100: DA 0B 11 C8  E3 46 02 E8  4E 06 89 29  3F 09 03 52  
 RTW: 0x110: 65 61 6C 74  65 6B 0E 03  38 30 32 2E  31 31 61 63  
 RTW: 0x120: 20 4E 49 43  08 03 31 32  33 34 35 36  FF FF FF FF  
 RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x140: FF FF FF FF  00 00 31 0F  FF FF FF FF  FF FF FF FF  
 RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
 RTW: hal_com_config_channel_plan chplan:0x7F
 RTW: [RF_PATH] ver_id.RF_TYPE:RF_1T1R
 RTW: [RF_PATH] HALSPEC's rf_reg_trx_path_bmp:0x11, rf_reg_path_avail_num:1, max_tx_cnt:1
 RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
 RTW: [RF_PATH] Registry's trx_path_bmp:0x00, tx_path_lmt:0, rx_path_lmt:0
 RTW: [RF_PATH] HALDATA's trx_path_bmp:0x11, max_tx_cnt:1
 RTW: [RF_PATH] HALDATA's rf_type:RF_1T1R, NumTotalRFPath:1
 RTW: [TRX_Nss] HALSPEC - tx_nss:1, rx_nss:1
 RTW: [TRX_Nss] Registry - tx_nss:0, rx_nss:0
 RTW: [TRX_Nss] HALDATA - tx_nss:1, rx_nss:1
 RTW: txpath=0x1, rxpath=0x1
 RTW: txpath_1ss:0x1, num:1
 RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
 RTW: rtw_ndev_init(wlan0) if1 mac_addr=e8:4e:06:89:29:3f
 RTW: module init ret=0
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: wlxe84e0689293f- hw port(0) mac_addr =e8:4e:06:89:29:3f
 RTW: rtw_set_802_11_connect(wlxe84e0689293f)  fw_state=0x00000008
 RTW: start auth
 RTW: auth success, start assoc
 RTW: assoc success
 RTW: ============ STA [04:92:26:8c:2a:f4]  ===================
 RTW: mac_id : 0
 RTW: wireless_mode : 0x44
 RTW: mimo_type : 0
 RTW: static smps : N
 RTW: bw_mode : 80MHz, ra_bw_mode : 80MHz
 RTW: rate_id : 10
 RTW: rssi : -1 (%), rssi_level : 0
 RTW: is_support_sgi : Y, is_vht_enable : Y
 RTW: disable_ra : N, disable_pt : N
 RTW: is_noisy : N
 RTW: txrx_state : 0
 RTW: curr_tx_rate : CCK_1M (L)
 RTW: curr_tx_bw : 20MHz
 RTW: curr_retry_ratio : 0
 RTW: ra_mask : 0x00000000003ffff0
 RTW: recv eapol packet 1/4
 RTW: send eapol packet 2/4
 RTW: recv eapol packet 3/4
 RTW: send eapol packet 4/4
 RTW: set pairwise key camid:0, addr:04:92:26:8c:2a:f4, kid:0, type:AES
 RTW: set group key camid:1, addr:04:92:26:8c:2a:f4, kid:2, type:AES
```

That log is as clean as it gets. The adapter works well.

Can I get both of you to post a rtw.log using `save-log.sh`?

It is possible this is a `RFE Type` issue. If this is a rtl8821cu chipset, then we may need to do some testing.

Run the following and scroll down to the documentation section that talks about RFE Type:

sudo ./edit-options.sh

Playing around with the various RFE settings may be the only way to find out if this is the problem.

greengeckowizard, 2023-04-28:
> I just installed this driver on a fresh installation of Ubuntu 23.04. System is x86_64. Kernel is 6.2 gcc is 12.2
>
> I use the `save-log.sh` script that comes with the driver so as to only get `RTW` [...]

Which numbers do you suggest we try?

morrownr, 2023-04-28:
> Which numbers do you suggest we try?

Start with 7 and work your way down. This is trial and error at its best.

greengeckowizard, 2023-04-28:
> > Which numbers do you suggest we try?
> 
> Start with 7 and work your way down. This is trial and error at its best.

Thanks  I'll test in a few hours I'll get back to you then.

evshiron, 2023-04-28:
I have tried all the RFE types, and the results are here:

* `64`, less 5G SSIDs
* `7`, less 5G SSIDs, higher success rate
* `6 5 3 1 0`, only 5G SSIDs
* `4 2`, less 5G SSIDs

Btw, this is a multi-state dongle, it will first show a CD-ROM on Windows, but not on Ubuntu afaik.

A rtw log of `log_level=6 rfe_type=7` here: [rtw7v.log](https://github.com/morrownr/8821cu-20210916/files/11356668/rtw7v.log)

The driver source provided by the vendor here (for pre-5.15 kernel only, not tested): https://www.lulian.cn/download/60-cn.html

morrownr, 2023-04-28:
> I wonder if the in-kernel rtl8821cu driver supports rtl8811cu too? If so, I might try it when I have time.

It does starting with kernel 6.2. Performance on 6.2 was poor. You might want to try 6.3.

I'll comment more on the above as I have time.

greengeckowizard, 2023-04-28:
> > I wonder if the in-kernel rtl8821cu driver supports rtl8811cu too? If so, I might try it when I have time.
> 
> It does starting with kernel 6.2. Performance on 6.2 was poor. You might want to try 6.3.
> 
> I'll comment more on the above as I have time.

Fedora 38 seemed to use the inline but it had the same issue.

greengeckowizard, 2023-04-28:
Oddly after installing the driver from this repo after a reboot it worked like it connected once then after you reboot would keep having the same authentication issue .

evshiron, 2023-04-29:
> Fedora 38 seemed to use the inline by default (enabled by default) but it had the same issue (Authentication issue).

Ubuntu 22.04.1 with mainline kernel 6.3 too.

greengeckowizard, 2023-04-29:
> I have tried all the RFE types, and the results are here:
> 
> * `64`, less 5G SSIDs
> * `7`, less 5G SSIDs, higher success rate
> * `6 5 3 1 0`, only 5G SSIDs
> * `4 2`, less 5G SSIDs
> 
> Btw, this is a multi-state dongle, it will first show a CD-ROM on Windows, but not on Ubuntu afaik.
> 
> A rtw log of `log_level=6 rfe_type=7` here: [rtw7v.log](https://github.com/morrownr/8821cu-20210916/files/11356668/rtw7v.log)
> 
> The driver source provided by the vendor here (for pre-5.15 kernel only, not tested): https://www.lulian.cn/download/60-cn.html
> 
> I wonder if the in-kernel `rtl8821cu` driver supports `rtl8811cu` too? If so, I might try it when I have time.

Yeah I also got the same result.

evshiron, 2023-05-03:
Today I switch to the initial version of Ubuntu 22.04.

With its 5.15 kernel and this [rtl8821CU](https://github.com/brektrou/rtl8821CU) driver, the dongle works well so far.

I do encounter similar failures when I skip [Check the name of the interface](https://github.com/brektrou/rtl8821CU#check-the-name-of-the-interface) though.

morrownr, 2023-05-03:
FYI:

```
To disable predictable network interface names on many distros:

Boot with the kernel parameter net.ifnames=0.

sudo nano /etc/default/grub

Add net.ifnames=0 to the GRUB_CMDLINE_LINUX_DEFAULT= line as such:

GRUB_CMDLINE_LINUX_DEFAULT="quiet splash net.ifnames=0"

Remember to the run the following after saving.

sudo update-grub  

Reboot system.

sudo reboot
```

greengeckowizard, 2023-08-07:
Hi an update on the situation so the inline kernel is working okish but there are still some odd issues like it would disconnect after a few hours randomly and when you reboot sometimes and startup the computer it just doesn't connect and then after rebooting it will connect again. But most of the time it seems to be working fine connecting.
running this kernel atm 6.4.8-x64v3-xanmod1 on Linux Mint 21.2 Cinnamon.

closed 2024-03-26 by morrownr