morrownr/8821cu-20210916 #78: Armbian(23.02.2) install

LittleOn, 2023-03-22:
```bash
neofetch
root@armbian 
------------ 
OS: Armbian (23.02.2) aarch64 
Host: CM311-1a-YST 
Kernel: 5.15.98-flippy-82+o 
Uptime: 14 days, 1 hour, 56 mins 
Packages: 583 (dpkg) 
Shell: zsh 5.8 
Resolution: 720x576i 
Terminal: /dev/pts/0 
CPU: ARMv8 rev 4 (v8l) (4) @ 1.908GHz 
Memory: 813MiB / 1953MiB 

```
```
armbian:8821cu-20210916:# ./install-driver.sh                            <main>
: ---------------------------
: install-driver.sh v20230227
: aarch64 (architecture)
: 4/4 (in-use/total processing units)
: 2000832 (total system memory)
: 5.15.98-flippy-82+o (kernel version)
: gcc (Debian 10.2.1-6) 10.2.1 20210110
: dkms:2.8
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source ->
                 /usr/src/rtl8821cu-5.12.0.4

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.98-flippy-82+o (aarch64)
Consult /var/lib/dkms/rtl8821cu/5.12.0.4/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

morrownr, 2023-03-22:
Hi @LittleOn 

Please go to the driver directory , run the following and post the results of make:

make clean
clear
make

LittleOn, 2023-03-25:
armbian:8821cu-20210916:# make clean                                     <main>
#make -C /lib/modules/5.15.103-flippy-82+o/build M=/root/src/8821cu-20210916 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions

armbian:8821cu-20210916:# make                                           <main>
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.103-flippy-82+o/build M=/root/src/8821cu-20210916  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.103-flippy-82+o'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: Ubuntu clang version 14.0.0-1ubuntu1
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
  CC [M]  /root/src/8821cu-20210916/core/rtw_cmd.o
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-mno-global-merge’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
make[2]: *** [scripts/Makefile.build:289: /root/src/8821cu-20210916/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1913: /root/src/8821cu-20210916] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.103-flippy-82+o'
make: *** [Makefile:2501: modules] Error 2

morrownr, 2023-03-25:
Hi @LittleOn 

> The kernel was built by: Ubuntu clang version 14.0.0-1ubuntu1
> You are using: gcc (Debian 10.2.1-6) 10.2.1 20210110

It appears that this version of Armbian was compiled with clang and some of the settings that work for clang are not working for gcc. I am curreently moving and my Armbian system is packed away so my ability to look at this is limited in the short term. Let me recommend that you post the results is the `make` command, as you did above, over at an Armbian forum and at a clang forum with a note saying what you are doing. If you get an answer that fixed this, I'm willing to add the info to the docs.

@morrownr

LittleOn, 2023-03-26:
Thanks for telling me

LittleOn, 2023-03-26:
armbian:8821cu-20210916:# make CC=clang-14                             <main ✗>
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.103-flippy-82+o/build M=/root/src/8821cu-20210916  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.103-flippy-82+o'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: Ubuntu clang version 14.0.0-1ubuntu1
  You are using:           Debian clang version 14.0.6
  CC [M]  /root/src/8821cu-20210916/core/rtw_cmd.o
In file included from /root/src/8821cu-20210916/core/rtw_cmd.c:17:
In file included from /root/src/8821cu-20210916/include/drv_types.h:78:
/root/src/8821cu-20210916/include/rtw_recv.h:820:52: warning: variable 'buf_star' is uninitialized when used here [-Wuninitialized]
        precv_frame = rxmem_to_recvframe((unsigned char *)buf_star);
                                                          ^~~~~~~~
/root/src/8821cu-20210916/include/rtw_recv.h:818:14: note: initialize the variable 'buf_star' to silence this warning
        u8 *buf_star;
                    ^
                     = NULL
1 warning generated.
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:289: /root/src/8821cu-20210916/core/rtw_cmd.o] Error 126
make[2]: *** Deleting file '/root/src/8821cu-20210916/core/rtw_cmd.o'
make[1]: *** [Makefile:1913: /root/src/8821cu-20210916] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.103-flippy-82+o'
make: *** [Makefile:2501: modules] Error 2

-_-||

ShiinaRinne, 2023-04-30:
```
root@armbian:~/src/8821cu-20210916# uname -a
Linux armbian 5.15.109-ophub #1 SMP PREEMPT Wed Apr 26 22:21:32 EDT 2023 aarch64 GNU/Linux

root@armbian:~/src/8821cu-20210916# make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.109-ophub/build M=/root/src/8821cu-20210916  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.109-ophub'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-none-linux-gnu-gcc (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.2.1 20221205
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
  CC [M]  /root/src/8821cu-20210916/core/rtw_cmd.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:289: /root/src/8821cu-20210916/core/rtw_cmd.o] Error 126
make[2]: *** Deleting file '/root/src/8821cu-20210916/core/rtw_cmd.o'
make[1]: *** [Makefile:1907: /root/src/8821cu-20210916] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.109-ophub'
make: *** [Makefile:2501: modules] Error 2

```

Hello~It seems that  I have similar problems, could you please give me some guidance？

morrownr, 2023-04-30:
My little system that runs Armbian is not setup yet and it may be a while before I can try to duplicate the problem on Armbian.

It appears that Armbian has moved over to using CLANG insted of gcc and gcc is chocking on something because of it. Anyone in the Armbian forums talking about this?

I'd offer guidance but I have no experience with CLANG.

ShiinaRinne, 2023-05-01:
Hi~ This issue is indeed caused by the different compilers used in the kernel. I have solved it by referring to this issue:
https://github.com/ophub/amlogic-s9xxx-armbian/issues/1043

> Armbian has moved over to using CLANG insted of gcc and gcc is chocking on something because of it
> 
This issue is caused by the fact that the kernel in another repository seems to have been compiled with CLANG by the same person all along. 

> 我在内核仓库里5.15和6.1都是转存的f大编译的内核，他使用clang-14编译。但是大家在armbian里默认都是gcc-11，所以失败的日志提示，当前内核用的clang-14，而你是gcc-11。
> 
> 
> 解决办法就是统一编译器，把armbian里的编译工具更新到和安装的内核所用的版本一致了。
> 

To solve this issue, we can recompile the kernel using GCC, or directly use the **dev kernel** they later compiled:
[https://github.com/ophub/kernel/releases/tag/kernel_dev](https://github.com/ophub/kernel/releases/tag/kernel_dev)

[image]

[image]

Jibun-no-Kage, 2023-07-14:
How does on identify how the kernel was compiled, i.e. via GCC or CLANG?  If the install-driver.sh trips over the not finding the headers?  Is there a CLI command that could expose this issue?

\# cat /proc/version
Linux version 5.10.110-rockchip-rk3588-indiedroid (armbian@next) (aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Thu Mar 9 02:22:35 UTC 2023

Looks like the IndieDroid Nova recommended OS based on Ubuntu Jammy Jellyfish was compiled with GCC if I read the above right?

mcr, 2023-08-09:
```
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
  You are using:           gcc (Raspbian 12.2.0-14+rpi1) 12.2.0
  CC [M]  /home/mcr/8821cu/core/rtw_cmd.o
In file included from ./include/linux/types.h:6,
                 from /home/mcr/8821cu/include/basic_types.h:75,
                 from /home/mcr/8821cu/include/drv_types.h:26,
                 from /home/mcr/8821cu/core/rtw_cmd.c:17:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /home/mcr/8821cu/core/rtw_cmd.o] Error 1
```

Jibun-no-Kage, 2023-08-10:
Ah... so it is embedded in the make run!

morrownr, 2023-08-10:
Jibun,

I keep 4 versions of gcc loaded on my dev box. My experience is that it can be a good idea to use the same major version of gcc that your kernel was compiled with.

https://github.com/morrownr/USB-WiFi

See:

10. How to install multiple gcc compiler versions

mcr, 2023-08-10:
I think dkms really should be taking care of using the right compiler. Maybe
it doesn't know how.  One would learn the result when loading the module, and
many of us just don't there.
The raspberrypi-kernel-headers just always seem broken to me.
It sure doesn't seem like it's 8821cu's fault, but until drivers can get
upstreamed, it's where everyone stubs their toes.

Jibun-no-Kage, 2023-08-11:
And the ARM64 support applicable to IndieDroid Nova... has a gap in that the headers are not even published via APT.  But I knew that in getting the Nova and using it... I use it as an infrastructure device, and it is a beast, well suited to running QEMU/KVM and/or even Docker.

morrownr, 2023-08-12:
@mcr 

> The raspberrypi-kernel-headers just always seem broken to me.

It happens far too often.

> It sure doesn't seem like it's 8821cu's fault, but until drivers can get
upstreamed...

An in-kernel driver for the 8821/11cu has been upstreamed. It happened with kernel 6.2.

RasPiOS will probably move to kernel 6.6, if it is the next LTS, at some point in spring of next year. That will probably be when you see the support unless you want to compile and install 6.4 or so. I do have a guide for do so if interested.

closed 2024-03-27 by morrownr

Jibun-no-Kage, 2024-03-28:
Just final comment... Pi OS is not at 6.6.x

morrownr, 2024-03-28:
Yes,  it is:

https://www.raspberrypi.com/software/operating-systems/

or

$ sudo apt full-upgrade

Now, what was the topic of this thread?

Jibun-no-Kage, 2024-03-28:
Well, the inference was... that for my IndieDroid... it is still on a variant of Armbian that is only 6.1.x.  And as you may (not?) recall I had issues finding the right kernel modules at the time to be able to compile/dkms various WiFi adapter drivers.  But since I use my IndieDroid as a hypervisor, I have not been motivated to come back to the original issue... in my case.  Is that what you were suggesting?  :)

Jibun-no-Kage, 2024-03-28:
Ah... no problem... just thought if someone came across the discussion, the Pi OS kernel version changed not long ago.   I just realized... I had a typo in my comment!