morrownr/8821cu-20210916 #80: Build Error (fatal error: asm/bitsperlong.h: No such file or directory)

elockman, 2023-03-27:
I'm running into the same build error with multiple custom driver installs. When trying to build, I get the following:
```
root@sama5d2-xplained:/usr/test/8821cu-20210916# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-poky-linux-gnueabi/11.3.0/lto-wrapper
Target: arm-poky-linux-gnueabi
Configured with: ../../../../../work-shared/gcc-11.3.0-r0/gcc-11.3.0/configure --build=x86_64-linux --host=arm-poky-linux-gnueabi --target=arm-poky-linux-gnueabi --prefix=/usr --exec_prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/libexec --datadir=/usr/share --sysconfdir=/etc --sharedstatedir=/com --localstatedir=/var --libdir=/usr/lib --includedir=/usr/include --oldincludedir=/usr/include --infodir=/usr/share/info --mandir=/usr/share/man --disable-silent-rules --disable-dependency-tracking --with-libtool-sysroot=/ --with-gnu-ld --enable-shared --enable-languages=c,c++ --enable-threads=posix --enable-multilib --enable-default-pie --enable-c99 --enable-long-long --enable-symvers=gnu --enable-libstdcxx-pch --program-prefix=arm-poky-linux-gnueabi- --without-local-prefix --disable-install-libiberty --disable-libssp --enable-libitm --enable-lto --disable-bootstrap --with-system-zlib --with-linker-hash-style=gnu --enable-linker-build-id --with-ppl=no --with-cloog=no --enable-checking=release --enable-cheaders=c_global --without-isl --with-float=hard --with-build-sysroot=/ --enable-standard-branch-protection --enable-nls --with-glibc-version=2.28 --enable-initfini-array --with-arch=armv7-a+fp
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 11.3.0 (GCC)
root@sama5d2-xplained:/usr/test/8821cu-20210916# git pull
Already up to date.
root@sama5d2-xplained:/usr/test/8821cu-20210916# sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: armv7l (architecture)
: 1/1 (in-use/total processing units)
: 508216 (total system memory)
: 5.15.68-linux4microchip-2022.10 (kernel version)
: gcc (GCC) 11.3.0
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.68-linux4microchip-2022.10/build M=/usr/test/8821cu-20210916  modules
make[1]: Entering directory '/lib/modules/5.15.68-linux4microchip-2022.10/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: arm-poky-linux-gnueabi-gcc (GCC) 11.3.0
  You are using:           gcc (GCC) 11.3.0
  CC [M]  /usr/test/8821cu-20210916/core/rtw_cmd.o
In file included from ./include/asm-generic/int-ll64.h:11,
                 from ./arch/arm/include/uapi/asm/types.h:5,
                 from ./include/uapi/linux/types.h:5,
                 from ./include/linux/types.h:6,
                 from /usr/test/8821cu-20210916/include/basic_types.h:75,
                 from /usr/test/8821cu-20210916/include/drv_types.h:26,
                 from /usr/test/8821cu-20210916/core/rtw_cmd.c:17:
./include/uapi/asm-generic/int-ll64.h:12:10: fatal error: asm/bitsperlong.h: No such file or directory
   12 | #include <asm/bitsperlong.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:289: /usr/test/8821cu-20210916/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1898: /usr/test/8821cu-20210916] Error 2
make[1]: Leaving directory '/lib/modules/5.15.68-linux4microchip-2022.10/build'
make: *** [Makefile:2501: modules] Error 2
An error occurred:  2
Please report this error.
Please copy all screen output and paste it into the problem report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
```
Does anyone have insight on how to resolve this?

elockman, 2023-03-28:
I no longer need this, closing issue.

closed 2023-03-28 by elockman

morrownr, 2023-03-28:
> I no longer need this, closing issue.

I haven't seen this before. What did you do to fix it?

elockman, 2023-03-30:
This issue is not fixed, I used an inline driver.