morrownr/8821cu-20210916 #81: Arm64 cross-compiling

gorghino, 2023-03-29:
Hi! I'm trying to cross-compile your drivers for my Jetson NX (arm64). I cannot directly use neither the install script nor the manual compilation on-board, since I'm testing the RealTime kernel.

On my host (x64 - Ubuntu 20.04.5 - gcc 9.4.0) I downloaded Nvidia's kernel sources and I compiled them.
According to the [Nvidia's documentation](https://docs.nvidia.com/jetson/archives/r35.2.1/DeveloperGuide/text/SD/Kernel/KernelCustomization.html#to-build-external-kernel-modules) I shall be able to cross compile with

```
$ cd <path_to_module_source>
$ make ARCH=arm64 –C <kernel_directory> M=$(pwd)
```

after setting CROSS_COMPILE accordingly.

I tried:

```
gorgo@razer:~/8821cu-20210118$ make ARCH=arm64 CROSS_COMPILE=~/Downloads/l4t-gcc/bin/aarch64-buildroot-linux-gnu- -C /media/gorgo/PopOS/home/gorgo/NX_5_1_HYAI/Linux_for_Tegra/source/public/kernel_out M=$(pwd)
make: Entering directory '/media/gorgo/PopOS/home/gorgo/NX_5_1_HYAI/Linux_for_Tegra/source/public/kernel_out'
  MODPOST /home/gorgo/8821cu-20210118/Module.symvers
make: Leaving directory '/media/gorgo/PopOS/home/gorgo/NX_5_1_HYAI/Linux_for_Tegra/source/public/kernel_out'
```

_Make_ stops after creating 4 files only (two are 0B):

```
-rw-rw-r--   1 gorgo gorgo    0 mar 29 10:07 modules.order
-rw-rw-r--   1 gorgo gorgo  121 mar 29 10:07 .modules.order.cmd
-rw-rw-r--   1 gorgo gorgo  212 mar 29 10:07 .Module.symvers.cmd
-rw-rw-r--   1 gorgo gorgo    0 mar 29 10:07 Module.symvers
```

In this log I retried the command with -d.
[makefile.log](https://github.com/morrownr/8821cu-20210916/files/11097188/makefile.log)

Thank you

morrownr, 2023-04-29:
Hi @gorghino 

You likely cannot use the installation script but that is no big.

For manually compiling, since I have made alterations to autodetect the platform, you will need to turn the autodetection off. Find the below lines in Makefile:

CONFIG_PLATFORM_AUTODETECT = y
CONFIG_PLATFORM_I386_PC = n

Change them to the following:

CONFIG_PLATFORM_AUTODETECT = n
CONFIG_PLATFORM_I386_PC = y

Try what you did again and let me know what happens. I did not test this but I do need to document what to do for cross-compiling so if we can work it out, I will document it.

morrownr, 2023-05-15:
I am going to reach out to another user that cross compiles to see if we can get some guidance. My to-do list is long right now so my time is limited. Will let you know.

gorghino, 2023-05-15:
Tnks, no rush

morrownr, 2023-05-15:
Here is the reply from @disaster123

```
I used the following command to cross compile the driver - for sure the pathes need to be adjusted and may be adapted to the environment:

ARCH=arm64 CROSS_COMPILE=/root/BPI-M4-bsp/toolchains/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu- make KVER=4.9.119-BPI-M4 KSRC=/root/BPI-M4-bsp/linux-rtk
```

If I had time right now, I would make a short guide for those using this and the drivers here. I think it could be specific to host x86 and target arm64. Would either or both of you gents like to work on it?