morrownr/8821cu-20210916 #84: (Question) Mesh Point Mode?

elockman, 2023-04-18:
The readme states that it supports mesh, but mesh point mode is not listed in the Supported interface modes. Is this a clerical error, or am I misunderstanding something?  

I'd like to build this driver into a Linux 5.15.68 version with yocto for mesh support. Should this work for me?

morrownr, 2023-04-18:
> Should this work for me?

It likely will work.

The items I list under Supported Interface Modes are modes that I test and know first hand that they work, If it does work, let me know and I will add it to the list.

elockman, 2023-04-18:
I'm running into the issue of my RTL8821CU identifying as a DISK.
`ID 0bda:1a2b Realtek Semiconductor Corp. DISK`

**usb_modeswitch** isn't working. Any suggestions @morrownr?
```
root@sama5d2-xplained:~# /usr/sbin/usb_modeswitch -K -v 0bda -p 1a2b > /dev/null
-sh: /usr/sbin/usb_modeswitch: No such file or directory
```

morrownr, 2023-04-18:
The Main Menu for this site is located at:

https://github.com/morrownr/USB-WiFi

Menu item 7 has some info about usb_modeswitch

Let me know if that helps.

elockman, 2023-04-18:
**mesh point** was not listed, so it looks like I may be starting over...

root@sama5d2-xplained: # uname -a
Linux sama5d2-xplained 5.15.68-linux4microchip-2022.10 #1 Wed Nov 9 12:48:42 UTC 2022 armv7l armv7l armv7l GNU/Linux
root@sama5d2-xplained: # iw list | grep "Supported interface modes" -A 8
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x1862
root@sama5d2-xplained:~#

elockman, 2023-04-19:
@morrownr are there `CONFIG_*` #defs I can use to enable mesh point?

morrownr, 2023-04-19:
> Feeling out of luck.

Time to do some research.

Personally myself, I would go with adapters based on in-kernel driver. Is there anything you want me to check?

closed 2023-04-20 by elockman

morrownr, 2023-04-20:
> It is my understanding that mesh is intrinsic to WiFi 6, so driver compatibility and configuration will be the hurdles, removing the availability and functionality hurdles. If you have any suggested adapters, I'd be willing to take a look.

Absolutely avoid adapters with the rtl8852au/rtl8832au chipsets. I tried to make the initial Realtek release of the driver for these chipsets availble over a year ago. It was the worst driver I have ever seen. I had to drop the project and warn Linux users to stay away. It appears that those chipsets may be out of production at this point, possibly because of some internal problems with the design of the chipsets. I'd say those chipsets will be abandonware.

I am currently working on the out-of-kernel driver for the rtl8852bu/rtl8832bu chipsets. It is a better driver and does seem to be getting a little support. I won't go so far as to say it is a good driver but it is better. It is nowhere near as good as the mt7921au driver.

Right now, in my opinion, for usb wifi 6 and 6e you only have one viable option: mt7921au. The mt7922u should be available soon and maybe we will start seeing adapters based on it later this year. It uses the same mt7921u driver as the mt7921au chipset does as they are basically the same WiFi 6e chipset but the mt7922u chipset is capable of 160 MHz wide channels.

A little detail that gets skipped over in the WiFi 6/6e usb adapter market is that the mt7921au/mt7922u chipsets are capable of 6 GHz so they are tri-band. I'm not aware of any Realtek usb chipsets that support 6 GHz.

@morrownr

elockman, 2023-04-20:
Thanks, that's helpful. Are you aware if WiFi 6 can create a mesh without a WiFi router?  In other words, an ad hoc mesh with 3 laptops for example?

morrownr, 2023-04-20:
> Are you aware if WiFi 6 can create a mesh without a WiFi router? In other words, an ad hoc mesh with 3 laptops for example?

I have little knowledge of mesh at this point so I'm not the one to ask. You might post in Issues at USB-WiFi (the link is above). I totaled the views the other day and over the last 2 weeks, there were over 25k views so a lot of people stop by. Some are very knowledgeable. Even some kernel devs and company engineers stop by at times.

elockman, 2023-04-20:
Nice, we could both learn something. Thanks for the help.