morrownr/8821cu-20210916 #88: install-driver.sh

bigbear1007, 2023-05-03:
root@armbian:~/8821cu-20210916# ./install-driver.sh 
: ---------------------------
: install-driver.sh v20230227
: aarch64 (architecture)
: 4/4 (in-use/total processing units)
: 2000832 (total system memory)
: 5.15.110-ophub (kernel version)
: gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
: dkms-2.8.7
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source -> /usr/src/rtl8821cu-5.12.0.4
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.110-ophub (aarch64)
Consult /var/lib/dkms/rtl8821cu/5.12.0.4/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

bigbear1007, 2023-05-03:
DKMS make.log for rtl8821cu-5.12.0.4 for kernel 5.15.110-ophub (aarch64)
Wed May  3 04:04:10 PM CST 2023
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.110-ophub/build M=/var/lib/dkms/rtl8821cu/5.12.0.4/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.110-ophub'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-none-linux-gnu-gcc (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24)) 12.2.1 20221205
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_io.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_io.o] Error 126
make[2]: *** Deleting file '/var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_io.o'
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_security.o] Error 126
make[2]: *** Deleting file '/var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_security.o'
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_debug.o] Error 126
make[2]: *** Deleting file '/var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_debug.o'
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_cmd.o] Error 126
make[2]: *** Deleting file '/var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_cmd.o'
make[1]: *** [Makefile:1907: /var/lib/dkms/rtl8821cu/5.12.0.4/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.110-ophub'
make: *** [Makefile:2501: modules] Error 2

bigbear1007, 2023-05-03:
已解决

echo 'export PATH=$PATH:/opt/gcc-aarch64/bin' | sudo tee -a /etc/profile.d/gcc-aarch64.sh
source /etc/profile

ln -sf /opt/gcc-aarch64/bin/arm-none-eabi-gcc /usr/local/bin/gcc

Jibun-no-Kage, 2023-05-04:
@morrownr, BTW, I have an IdieDroid device on order, and it comes with Armbian by default, as I recall.  So I am thinking I can test as well.

morrownr, 2023-05-04:
Hi @Jibun-no-Kage 

I can use the help.

Let me step back and paint the big picture for you.

I am seeing two problems right now that are confirmed by bugs in the other repos. Neither are common but they are things that need to be sorted out.

1. Compilation is not working with Clang. It does not work with the script or manually. This is coming to a head because it appears that Armbian has switched to using Clang as the default instead of gcc. Installing gcc does not fix the problem. I am not familiar with Clang so any help to figure out how to support Clang without messing up gcc is welcome.

2. Cross compiling seems to have been affected by the mods we made in the scripts when working on this driver. I suspect it is the code to automatically detect and set the platform that is causing the problem.

I think research is the first thing to do. Find some good guides for doing both of the above so that we know how things should work is probably the first thing to do and then I should set up new separate issues for each of these problems so we can keep better track of the work.

Your thoughts?

Jibun-no-Kage, 2023-05-04:
Yup, as usual, I think you have outlined it well.  :)  I just got today my new IndieDroid.... the first one was DOA, the boot loader was failing, and the once I got that squared away, the HDMI port had a hardware fault, so the vendor sent me a new unit.  I will power it up this weekend, and if all is well, will spend some time getting familiar Armbian.

I struggle with why Clang is being used, because it is now slower that GCC, and GCC is from all I have read, considered better for C++ for example.

However, this maybe one reason why Clang is used... "Clang retains more information during the compiling process than GCC, and preserves the overall form of the original code, making it easier to map errors"  This is from the Clang Wiki.

bigbear1007, 2023-05-06:
这个错误的原因是内核用clang编译的，换成arm的gcc编译工具进行交叉编译就可以了
如有必要，我可以出个教程；

bigbear1007, 2023-05-06:
但是在armbian的5.15 110内核上测试不成功。今天我再测试一下在2023.5.4版本上测试一下。之前在110内核上测试通过，wifi可以正常使用。

bigbear1007, 2023-05-06:
[image]
armbian换成108内核可以成功

参考以下步骤
mkdir /opt/gcc-aarch64-none-elf

 tar xf arm-gnu-toolchain-12.2.rel1-aarch64-aarch64-none-elf.tar.xz --strip-components=1 -C /opt/gcc-aarch64-none-elf

echo 'export PATH=$PATH:/opt/gcc-aarch64-none-elf/bin' | sudo tee -a /etc/profile.d/gcc-aarch64-none-elf.sh

source /etc/profile

ln -sf /opt/gcc-aarch64-none-elf/bin/aarch64-none-elf-gcc /usr/local/bin/gcc

echo 'export PATH=$PATH:/opt/gcc-aarch64/bin' | sudo tee -a /etc/profile.d/gcc-aarch64.sh
source /etc/profile

ln -sf /opt/gcc-aarch64/bin/arm-none-eabi-gcc /usr/local/bin/gcc
随后就可以继续install-driver了

morrownr, 2023-05-08:
Hi @bigbear1007 

Thanks for the information. When I get to this issue on my to-do list, I will use your information.

closed 2024-03-27 by morrownr