morrownr/8821cu-20210916 #89: Any chance for Kernel 4.9 support?

disaster123, 2023-05-14:
HI there,

i've a Banana PI M4 which has the 8821cu installed. Sadly the only supported Kernel from them is 4.9 - any chance to add compat to the 4.9 kernel?

Greets,
Stefan

morrownr, 2023-05-14:
Hi @disaster123 

> Any chance for Kernel 4.9 support? 

This driver likely works with kernel 4.9. I mostly follow the support of the Linux foundation in that I don't support kernels older than what the foundation is supporting. Have I removed support for kernel 4.9? No. Do I test with it anymore? No.

You can give it a try. Follow the installation instructions.

disaster123, 2023-05-14:
Thanks! Any ideas about this issue:
```
ake[1]: Entering directory '/usr/src/linux-headers-4.9.119-BPI-M4-Kernel'
arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
arch/arm64/Makefile:49: LSE atomics not supported by binutils
  CC [M]  /home/pi/8821cu-20210916/core/rtw_cmd.o
  CC [M]  /home/pi/8821cu-20210916/core/rtw_security.o
  CC [M]  /home/pi/8821cu-20210916/core/rtw_debug.o
  CC [M]  /home/pi/8821cu-20210916/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-mcmodel=large’
scripts/Makefile.build:334: recipe for target '/home/pi/8821cu-20210916/core/rtw_cmd.o' failed

```

I found the same error here:
https://github.com/aircrack-ng/rtl8812au/issues/408

morrownr, 2023-05-14:
Post the screen output of the install-driver.sh script.

disaster123, 2023-05-14:
here we go:

```
pi@bananapi:~/8821cu-20210916 $ sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20230227
: aarch64 (architecture)
: 4/4 (in-use/total processing units)
: 1609720 (total system memory)
: 4.9.119-BPI-M4-Kernel (kernel version)
: gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.9.119-BPI-M4-Kernel/build M=/home/pi/8821cu-20210916  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.119-BPI-M4-Kernel'
arch/arm64/Makefile:27: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
arch/arm64/Makefile:49: LSE atomics not supported by binutils
  CC [M]  /home/pi/8821cu-20210916/core/rtw_cmd.o
  CC [M]  /home/pi/8821cu-20210916/core/rtw_security.o
  CC [M]  /home/pi/8821cu-20210916/core/rtw_debug.o
  CC [M]  /home/pi/8821cu-20210916/core/rtw_io.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-mcmodel=large’
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
scripts/Makefile.build:334: recipe for target '/home/pi/8821cu-20210916/core/rtw_cmd.o' failed
make[2]: *** [/home/pi/8821cu-20210916/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
gcc: error: unrecognized command line option ‘-mcmodel=large’
scripts/Makefile.build:334: recipe for target '/home/pi/8821cu-20210916/core/rtw_security.o' failed
make[2]: *** [/home/pi/8821cu-20210916/core/rtw_security.o] Error 1
gcc: error: unrecognized command line option ‘-mcmodel=large’
scripts/Makefile.build:334: recipe for target '/home/pi/8821cu-20210916/core/rtw_debug.o' failed
make[2]: *** [/home/pi/8821cu-20210916/core/rtw_debug.o] Error 1
gcc: error: unrecognized command line option ‘-mcmodel=large’
scripts/Makefile.build:334: recipe for target '/home/pi/8821cu-20210916/core/rtw_io.o' failed
make[2]: *** [/home/pi/8821cu-20210916/core/rtw_io.o] Error 1
Makefile:1613: recipe for target '_module_/home/pi/8821cu-20210916' failed
make[1]: *** [_module_/home/pi/8821cu-20210916] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.119-BPI-M4-Kernel'
Makefile:2501: recipe for target 'modules' failed
make: *** [modules] Error 2
An error occurred:  2
Please report this error.
Please copy all screen output and paste it into the problem report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

```

morrownr, 2023-05-14:
> gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516

That could be a problem.

If Banana Pi is only supporting this old distro, bad on them.

What you might want to do is investigate Armbian to see if there is a distro for your hardward.

disaster123, 2023-05-14:
Sadly yes... they really suck - but due to covid there was to alternative board available in 2021...

disaster123, 2023-05-14:
Thanks a lot - i got it all working even with this old OS and Kernel ;-) Driver is happily running!

closed 2023-05-14 by disaster123

morrownr, 2023-05-14:
What did you do?

disaster123, 2023-05-15:
BananaPi provides a special docker container to cross compile the kernel using gcc-7. I used that one to compile the kernel and manually cross compiled this driver!

morrownr, 2023-05-15:
I was thinking that old gcc v6 was going to be problematic. Glad to see support for a newer version.

I have a user in another repo that is trying to cross compile x86 to arm64 a driver and that is not something I have done, plus I am busy right now. I would be greatly if you could either assist him or post the information how to do it with this driver.

Let me know.

disaster123, 2023-05-15:
@morrownr sorry sure. I used the following command to cross compile the driver - for sure the pathes need to be adjusted and may be adapted to the environment:
```
ARCH=arm64 CROSS_COMPILE=/root/BPI-M4-bsp/toolchains/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu- make KVER=4.9.119-BPI-M4 KSRC=/root/BPI-M4-bsp/linux-rtk

```