morrownr/8821cu-20210916 #90: Alma Linux 9.2 make not work | 9.1 make work

ds2k5, 2023-05-15:
Hello,
try to build on AlmaLinux 9.2 but make fail: ( the system did a upgrade from 9.1 to 9.2 by dnf )

```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-284.11.1.el9_2.x86_64/build M=/root/8821cu-20210916  modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-284.11.1.el9_2.x86_64'
  CC [M]  /root/8821cu-20210916/os_dep/osdep_service.o
/root/8821cu-20210916/os_dep/osdep_service.c: In function ‘thread_exit’:
/root/8821cu-20210916/os_dep/osdep_service.c:1315:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
 1315 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:321: /root/8821cu-20210916/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1923: /root/8821cu-20210916] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-284.11.1.el9_2.x86_64'
make: *** [Makefile:2501: modules] Error 2

```

Kernel Version:  5.14.0-284.11.1.el9_2.x86_64

**How to fix this ?**

Did this in the paste on AlmaLinux 9.1 there it worked.

Workaround to get Internet connection: choose at boot the kernel from 9.1 

[image]

pierrefnuk, 2023-05-17:
Same issue after upgrading from RHEL 9.1 to RHEL 9.2

morrownr, 2023-05-17:
Hi @ds2k5 @pierrefnuk 

You aren't going to like my answer but here goes. From the README.md:

```
Note: Red Hat Enterprise Linux (RHEL) and distros based on RHEL are not supported due to the way kernel patches are handled. I will support knowledgable RHEL developers if they want to merge the required support and keep it current. I reserve the right to delete this support if it causes any problems.
```

I enjoy supporting Linux users but I simply don't have time to add another major project and this would be major because of how Red Hat inappropriately handles the backporting of selected patches. 

My recommendation would be for you to investigate if you can use adapters based on in-kernel drivers. Adapters based on the mt7612u and mt7610u would likely be a good place to start. You can get some ideas of the adapters that are available if you go to the Main Menu and read menu item 2:

https://github.com/morrownr/USB-WiFi

Like I said, menu item 2, Plug and Play List, could be a good place to start but given that Red Hat uses very old kernels, starting the search with section AC1200 is probably a good idea. You may want to check in with the distro support forums for advice on getting wifi going as RHEL is a server distro and some server distros need a little help to get wifi going. I have no experience with RHEL.

Good luck and if you find a good solution, please post the solution here so others may benefit.

Regards,

@morrownr