morrownr/8821cu-20210916 #92: prevent RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode) from loading at boot

mousemat86, 2023-06-23:
So I have this USB dongle that identifies as RTL8188GU which looks to have an inbuilt driver. 

I've installed 8821cu-20210916 and it's works well with the device apart from the driver isn't loading on boot and requires unplugging and re-inserting to load.

I believe this could be overcome with a modprobe blacklist but my attemps so far haven't been successful. Would any one know the answer?

```
lsusb

on boot:
RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)

After resinserting the device:
Realtek Semiconductor Corp. 802.11ac NIC
```

x931890193, 2023-07-12:
> So I have this USB dongle that identifies as RTL8188GU which looks to have an inbuilt driver.
>
> I've installed 8821cu-20210916 and it's works well with the device apart from the driver isn't loading [...]

same situation， how to do that？，replug when boot everytime！！

sanhuoO, 2023-07-13:
> So I have this USB dongle that identifies as RTL8188GU which looks to have an inbuilt driver.
>
> I've installed 8821cu-20210916 and it's works well with the device apart from the driver isn't loading [...]

same too ,eject -r 
eject: /dev/cdrom: not found mountpoint or device with the given name

sanhuoO, 2023-07-13:
lsusb

Bus 001 Device 003: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)

**usb_modeswitch -KW -v 0bda -p 1a2b** 

lsusb

Bus 001 Device 005: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

moonlightz, 2023-12-13:
So happens I have this piece of...... .... just because it supposedly has an good antenna and 802.11ac.
I unpacked the thing... I looked fine... I plugged it into the an usb extension and it got detect by Armbian box.
I did frist lsusb and I got scared... I saw 8188GU 802.11n and started looking what this thing is. Just an wireless n chip.
I picked the cardboard box and indeed it was the product I ordered. Mercusys MU6H dual band 5dbi antenna.. yada...  I confess that I never have heard of this .... manufacturer. 
The usb modeswitch was helpful.... but:
• how do I auto switch the damn thing from cdrom mode to the wireless mode?
• drivers would be appreciated... I have armbian aarch64 kernel 6.1.50...

morrownr, 2023-12-13:
@moonlightz @sanhuoO

>  Apparently the thing is autoswitching from cdrom to wireless mode and it is using RTL8821CU:
cfg80211 430080 2 8821cu,brcmfmac

Where did you see that line?

I have seen problems with the rtl8821cu regarding mode switching before.  usb_modeswitch has a little database and there could be an error in it. However, that causes me to ask why it would work on a replug. Maybe it is a timing issue. 

I've also seen other problems with this chipset. Evidently makers have various options for antennas and that is problematic on some adapters. My adapter that I use with this driver is a single-state  (no usb_modeswitch required) and single-function (no bluetooth). I have no problems but I basically do not recommend adapters with rtl8821cu or rtl8811cu as I have simply seen too many Linux users have problems.

The Main Menu of this site has a lot of good info about adapters that work well in Linux:

https://github.com/morrownr/USB-WiFi

moonlightz, 2023-12-13:
lsmod | grep 80211
the 8821cu is the driver for the usb adapter and the other brcm is the internal wifi chip of the box that runs armbian (I'm currently having problems with it... randomly most of times there's no wlan0).

When I bought this ... thing ... I wasn't aware of all of these problems. I happened to install firmware-realtek and usb-modeswitch and that's it. Only time will tells if this adapter is any good...

morrownr, 2023-12-14:
@moonlightz 

> When I bought this ... thing ... I wasn't aware of all of these problems.

I am very aware of this problem. This is why I started this site several years ago. This site, [USB-WiFi](https://github.com/morrownr/USB-WiFi) , is averaging over 10,000 hits per week these days. People all over the world use and help me maintain the information on this site. The Main Menu gives a lot of info:

Menu item 1 gives background information that all Linux users need to be aware of.

Menu item 2 provides a long list of adapters that are known to work well. Many of the listed adapters have reviews and all listed adapters include links to online locations where the adapter can be purchased.

Menu item 5 is an overview of usb chipsets. It includes a recommendation for each chipset as to whether the chipset is recommended for Linux users and many details of the support for the chipset.

I can say without reservation that if Linux users read and consider the guidance and information that is provided, the probably of a more satisfying experience is enhanced greatly.

closed 2024-03-27 by morrownr

MR0000001, 2024-04-17:
> lsusb
> 
> Bus 001 Device 003: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)
> 
> **usb_modeswitch -KW -v 0bda -p 1a2b**
> 
> lsusb
> 
> Bus 001 Device 005: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

Being a Amazon Vine reviewer, i got the change to get 2 piece of Newfast (comfast branch company) with this RTL8188GU chipset and CDROM mode.

Im using them as Client & AP mode for Extender the WiFi coverage in a TvBox running ARMbian.
The usb_modeswitch command will stop at the first device 0bda:a2b, the solution i found to make both work as RTL8188GU is to run that modeswitch and then unbind & bind the whole HUB where those are connected to.

So unbinding and re-binding 1-1 (in my case there is only 1 hub usb) both of them are in RLT8188GU mode.

I just wanted to share this solution ;)

morrownr, 2024-04-17:
@MR0000001 

Thanks for sharing.

In my opinion, the single best thing you can do when you write your review is to WARN Linux users to avoid from this product. Multi-state usb wifi adapters can cause problems...as you have found out. Having that onboard Windows driver is a bad idea even for Windows users...who is going to update that driver in the future? Also, this adapter is based on the rtl8852bu chipset for WiFi. There is no good driver support for Realtek WiFi 6 usb wifi adapters.

There is no reason for Linux users to buy a WiFi 6/e adapter with a Realtek chip when there are adapters with with Mediatek chips that are simply Plug and Play and are feature rich. You can see several such adapters in the Plug and Play List here on the main menu. Here is an example of such an adapter on Amazon:

 https://www.amazon.com/EDUP-Wireless-802-11AX-Tri-Band-Compatible/dp/B0CVVWNSH2

Please help spread the word.

Thanks again for the post ... together, we can get the information to Linux users.

@morrownr

MR0000001, 2024-04-17:
> @MR0000001
>
> Thanks for sharing.
>
> In my opinion, the single best thing you can do when you write your review is to WARN Linux users to avoid from this product. Multi-state usb wifi adapters can cause [...]

Well is not like they are hiding the fact that those products are bad on Linux, infact they dont offer a Linux Driver at all in their website.

The Linux User must be wise and always know that things are primary made for Windows and so do a good search before to buy.

The product on Windows works very well so i cant discriminate it because is made the way better for Windows-noob-users that will not find the driver if not in a CDROM memory....

While this make things harder and strange is the best way to help the lesser able people, so i dont dislike it but at same time hate it in  a Unix environment.

@morrownr thanks for your great work on drivers!

morrownr, 2024-04-17:
> The Linux User must be wise...

I must disagree with what you appear to be saying here. Linux has evolved to the point that regular users can get along just fine with many desktop distros these days. My wife's desktop has Linux on it. I have to help her with her cel phone at times but almost never with her laptop. She does not know the difference between Windows and Linux. She simply uses the laptop.

Linux users do need to be informed and that is what my github site here is all about... informing Linux users about usb wifi adapters. 

> and always know that things are primary made for Windows...

This is not true. I can start list of `things` that are ONLY made with Linux in mind. If we are only talking about usb wifi adapters, in the Plug and Play List, you will see two Alfa adapters and one Edup adapter in the WiFi 6e, mt7921au chipset, section that have characteristics that show Linux was considered and tested during development. One example is LED support and how it was done.

> The product on Windows works very well so i cant discriminate it...

I did not ask you to say anything about Windows. I did ask you to point out that it is not a good product for Linux and to give the reasons why. In doing so, you are telling the truth and you are doing Linux users a service.

MR0000001, 2024-04-18:
Update on my side:

tried on another tvbox with a single USB3 port and a HUB usb3 with the same Armbian version and your driver, the rtl8188GU doesnt go in CDROM mode.

Seems that forcing the USB3 mode in the rtl8852bu.conf directly disable the CDROM mode.

@moonlightz @morrownr

morrownr, 2024-04-18:
> a HUB usb3 with the same Armbian version and your driver, the rtl8188GU doesnt go in CDROM mode.

Armbian includes usb_modeswitch, the utility that will automatically switch modes if everything is working properly.

> tried on another tvbox with a single USB3 port

The TV box likely runs Linux and has usb_modeswitch installed.

The adapter starts up in flash/CDROM mode and the only way for it to get to wifi mode is for it to be modeswitched. Often it works well, but it does not always work as we want as you pointed out.

FYI: Your adapter is not a rtl8188GU. You simply see that because it is in the database that way. I do not know why. There is no rtl8188GU chipset or driver associated with your adapter. It uses the rtl8852/32bu chipset/driver.

Engardian, 2024-08-06:
I had a similar problem. The seller claimed that the adapter had an RTL8811 chipset, the lsusb command, in turn, displayed RTL8188GU for the adapter, and in the end the adapter used the rtw_8821cu driver (after usb_modeswitch). One way or another, the adapter started working, and did not even return to Driver CDROM mode. But the problem was different - when I try to start or reboot the computer with this adapter connected, it stops at the stage of launching the initialisation daemon (in my case it's runit, the Void Linux distro), and does nothing until I disconnect the adapter. How can I fix this?

morrownr, 2024-08-06:
@Engardian 

I wish I had a perfect solution but multi-state usb wifi adapters are problematic and it isn't just for Linux. Who is going to update that windows driver when it no longer works on what you use?

I do my best to warm Linux users away from multi-state adapters. Here is the site Main Menu:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 are good to know if you use Linux.

Engardian, 2024-08-23:
> For people stumbling onto this thread like me,
> 
> Here is something that worked for me as a permanent solution based on @sanhuoO comment above usb_modeswitch is needed for this to work.
> 
>     1. Create a config at /etc/usb_modeswitch.d/
> 
> 
> `sudo nano /etc/usb_modeswitch.d/usb1.conf`
> 
>     2. Add the contents as
> 
> 
> ```
> TargetVendor=0x0bda 
> TargetProduct=0x1a2b
> MessageContent="55534243123456780000000000000011062000000000000000000000000000"
> ```
> 
> May need to play with the MessageContent but this has worked for me.
> 
>     3. then create a switch rule as
>        `sudo nano /etc/udev/rules.d/usb_modeswitch.rules`
> 
>     4. add content as
>        `ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="/usr/sbin/usb_modeswitch -v 0bda -p 1a2b -J"`
> 
>     5. run `sudo udevadm control --reload-rules` and bobs your uncle :)
> 
> 
> Happy debugging

I tried your method. It didn't work after the first restart. But miraculously it worked after the second or third, so I guess thanks 🙃

defencedog, 2024-09-17:
@RavenOfTime @Engardian bad luck for me I bought this chamaleon USB dongle for my OrangePi3b. Here is [my experience ](https://github.com/defencedog/orangepi3b_v2.1/blob/main/peripherals/README.md) However I didn't need to create any ` /etc/usb_modeswitch.d/usb1.conf`  What does this part do?

dubhater, 2024-09-17:
@defencedog Someone is wrong on the internet. :) Your readme.md says RTL8821CU has Bluetooth 5.x, but that's not right. It has 4.2, same as RTL8723DU.

defencedog, 2024-09-18:
@dubhater Thanks for correction. Per [AskUbuntu ](https://askubuntu.com/a/591813) I got this output ...however my BT5 from Broadcom really sucks
```
hci1:	Type: Primary  Bus: USB
	BD Address: 90:DE:80:17:E3:AE  ACL MTU: 1021:8  SCO MTU: 255:12
	UP RUNNING PSCAN ISCAN 
	RX bytes:998950 acl:468 sco:0 events:49144 errors:0
	TX bytes:29268332 acl:44822 sco:0 commands:1417 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'orangepi3b'
	Class: 0x6c0000
	Service Classes: Rendering, Capturing, Audio, Telephony
	Device Class: Miscellaneous, 
	HCI Version: 4.2 (0x8)  Revision: 0x826c
	LMP Version: 4.2 (0x8)  Subversion: 0xa99e
	Manufacturer: Realtek Semiconductor Corporation (93)

hci0:	Type: Primary  Bus: UART
	BD Address: 54:78:C9:0E:17:AF  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:3268169 acl:155470 sco:0 events:392 errors:0
	TX bytes:50664 acl:97 sco:0 commands:287 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'orangepi3b #1'
	Class: 0x6c0000
	Service Classes: Rendering, Capturing, Audio, Telephony
	Device Class: Miscellaneous, 
	HCI Version: 5.0 (0x9)  Revision: 0x59
	LMP Version: 5.0 (0x9)  Subversion: 0x6606
	Manufacturer: Broadcom Corporation (15)

```

morrownr, 2025-11-09:
Hi @BellemareMederic 

> How do I flash my Realtek Wi-Fi dongle to remove this cursed CDFS partition?

I am not aware of a good way to do this as I think most adapters has write once storage. I have seen a couple of examples where a guy used a soldering iron to remove the storage.

> Alright, I need to vent a little.

Understandable. I started this site to spread information to prevent the need to vent. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

You can scroll down to see the menu. Recommended reading includes menu items 1 and 2. Menu item 2 includes information about and links to many USB WiFi adapters that work wonderfully with Linux. I eliminate adapters that will have problems. Here is the list of policies that I use:

Policy: USB WiFi adapters listed here adhere to the following policies:

```
- all adapters use modern in-kernel (mac80211) drivers that make the adapters plug and play (No Exceptions!)
- all adapters are single-state (no Windows driver inside) (No Exceptions!)
- all adapters must pass rigorous testing by me, @morrownr, or have multiply recommendations from users 
- adapters that are single-function (no Bluetooth support) are preferred and will be given listing priority
- adapters that use chip maker default device IDs, VID/PID, are preferred and will be given listing priority
- no adapters from brands that have a history of doing things that make Linux support difficult
- no white-box adapters. White box adapters are used by resellers that pop up for short periods
```

> Why on earth does this Realtek dongle pretend to be a virtual CD-ROM (CDFS) before it even shows up as a Wi-Fi adapter?

It is so that an onboard Windows driver can jump into action and install when the adapter is plugged in. You'll find Realtek based adapters do this a lot. It is very rare on Mediatek based modern adapters. 

Read the two sections that I pointed out on the Main Menu. If you have an questions, let me know. There is no need for Linux users to be tortured with this multi-state crap in this day in time as there are MANY really good adapters that work great on Linux.