morrownr/8821cu-20210916 #93: make[1]: *** /lib/modules/6.2.0-23-generic/build: No such file or directory.  Stop.

shumintao, 2023-06-23:
ubuntu: 23.04

```
root@lm-OptiPlex-3000:/home/lm/Downloads/rtl8812AU_8821AU_linux# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.2.0-23-generic/build M=/home/lm/Downloads/rtl8812AU_8821AU_linux  modules
make[1]: *** /lib/modules/6.2.0-23-generic/build: No such file or directory.  Stop.
make: *** [Makefile:1622: modules] Error 2
root@lm-OptiPlex-3000:/home/lm/Downloads/rtl8812AU_8821AU_linux# sudo sh install-driver.sh
sh: 0: cannot open install-driver.sh: No such file
```

morrownr, 2023-06-23:
Hi @shumintao 

There is a pinned issue at the top that I need you to read and follow:

(Info) Problem Reporting Guidelines - Please Read.

shumintao, 2023-08-24:
> > 
> 
> 你好，问题解决了吗

解决了，但是忘记了怎么操作了。一个多月了

closed 2024-03-27 by morrownr