morrownr/8821cu-20210916 #94: This is not issue!!! this is many thx

x931890193, 2023-07-07:
This my raspi 2 model B info.,   many thx,  rtl8811cu works will!
```bash
Linux kali-raspberry-pi 5.15.44-Re4son-v7+ #1 SMP Debian kali-pi (2022-07-03) armv7l GNU/Linux
````
```bash
dmesg -T

[Fri Jul  7 23:33:07 2023] usb 1-1.4: new high-speed USB device number 9 using dwc_otg
[Fri Jul  7 23:33:07 2023] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[Fri Jul  7 23:33:07 2023] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Jul  7 23:33:07 2023] usb 1-1.4: Product: 802.11ac NIC
[Fri Jul  7 23:33:07 2023] usb 1-1.4: Manufacturer: Realtek
[Fri Jul  7 23:33:07 2023] usb 1-1.4: SerialNumber: 123456

```

```bash
: ---------------------------
: install-driver.sh v20230227
: armv7l (architecture)
: 2/4 (in-use/total processing units)
: 892892 (total system memory)
: 5.15.44-Re4son-v7+ (kernel version)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
: ---------------------------
```

```bash
ifconfig
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.112  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::f252:f89a:6ee4:335  prefixlen 64  scopeid 0x20<link>
        ether e8:4e:06:99:16:59  txqueuelen 1000  (Ethernet)
        RX packets 564  bytes 123025 (120.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 37  bytes 3573 (3.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
````

many  many many many thx 😊 😊 😊

morrownr, 2023-07-07:
Good to hear. Enjoy.

Keep in mind that an in-kernel driver for this chipset was added in kernel 6.2. When Kali for RasPi gets to kernel 6.3 or later, you might want to give it a try as the in-kernel drivers generally have better monitor mode support.

All you need to do to fully remove this driver is:

$ sudo sh remove-driver.sh
$ sudo reboot

If you are working on security analysis or pen testing, I have a script to take you into monitor mode that you might find works better than alternatives and it is easy for you to modify the source. See `10. Monitor Mode` on the Main Menu:

https://github.com/morrownr/USB-WiFi

x931890193, 2023-07-07:
> Good to hear. Enjoy.
>
> Keep in mind that an in-kernel driver for this chipset was added in kernel 6.2. When Kali for RasPi gets to kernel 6.3 or later, you might want to give it a try as the [...]

Dear friend!!!! 
glad to hear you reply！！
I also have a verison-five2 with risc-v cpu.  I try to use this repo to install driver！ maybe we need risc-v cpu diver。 hhhh

morrownr, 2023-07-07:
Hola amigo,

>  maybe we need risc-v cpu diver...

No tengo idea.

I have no idea.

x931890193, 2023-07-07:
> Hola amigo,
> 
> > maybe we need risc-v cpu diver...
> 
> No tengo idea.
> 
> I have no idea.

Thank you my friend!  I am most grateful to you for your help， have a nice day！！

closed 2023-07-07 by x931890193