morrownr/8821cu-20210916 #98: FYI:  Install on IndieDroid Nova, OS Ubuntu 22.04 LTS Jammy Jellyfish, Not Quite

Jibun-no-Kage, 2023-07-14:
This is NOT at this point, a driver/package issue, just started an issue to track progress, and so others will see the current status of  my efforts to install my USB dongle on the IndieDroid recommended OS image.  The IndieDroid Nova is pretty new, so not surprised that there is/or will be some bumps to smooth out.

\# apt install -y build-essential dkms git iw

In my case only iw was not already present.

\# git clone https://github.com/morrownr/8821cu-20210916.git

Worked as expected.

\# ./install-driver.sh
Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.
Once the header files are properly installed, please run "sudo ./install-driver.sh"

This was done as validation, since I did not as yet have headers in place.

\# cat /etc/os-release
PRETTY_NAME="Armbian 23.05.0-trunk jammy"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.2 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

\# uname -a
Linux indiedroid-nova 5.10.110-rockchip-rk3588-indiedroid #1 SMP PREEMPT Thu Mar 9 02:22:35 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

\# uname -r
5.10.110-rockchip-rk3588-indiedroid

\# apt search linux-headers-$(uname -r)
Sorting... Done
Full Text Search... Done

\# ls -l /usr/src/linux-headers-$(uname -r)
ls: cannot access '/usr/src/linux-headers-5.10.110-rockchip-rk3588-indiedroid': No such file or directory

\# apt install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.10.110-rockchip-rk3588-indiedroid
E: Couldn't find any package by glob 'linux-headers-5.10.110-rockchip-rk3588-indiedroid'

I fired off a message to the AmeriDroid, reseller for the IndieDroid Nova, asking for details on why the headers package appears to be M.I.A.  Looks like the classic way to pull down the headers for Ubuntu is not going to work.

Now... I did find 'linux-headers-legacy-rockchip-rk3588' but? The package installation failed, has a python2 dependency, and that is just not going to fly, or is it?

\# apt install python2
\# apt install linux-headers-legacy-rockchip-rk3588

A few warnings, but actually compiled?  Ah, ok, lets see what happens!

\# ./install-driver.sh
Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.
Once the header files are properly installed, please run "sudo ./install-driver.sh"

Like I said above... No way that was going to fly.

Jibun-no-Kage, 2023-07-14:
One more nugget of information... Came across the CLANG versus GCC kernel compile issue thread... Appears IndieDroid Nova recommended OS kernel was compiled via GCC if I am reading the following right:

\# cat /proc/version
Linux version 5.10.110-rockchip-rk3588-indiedroid (armbian@next) (aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Thu Mar 9 02:22:35 UTC 2023

morrownr, 2023-07-15:
Hi @Jibun-no-Kage 

You need to stop by more often.

> Like I said above... No way that was going to fly.

I saw that coming from a mile away. These little ARM board makers seem to be a bunch of EE's that don't do software nor do they are understand that it is important. There is one thing that separates RasPi from all of the others and it is not hardware design and quality. How in the hell did the RasPi folks manage to put Broadcom and Linux together in the same project? They did not reseach what they were doing well at all.

> aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0

Yup. gcc v11.3.

I need to go pop some popcorn. This is going to be good.

Jibun-no-Kage, 2023-07-15:
Hey!  Yeah, been busy with life... and picked up a new part time career... I am doing Video/Audio recording and streaming, for various customers.  I am not a content generator myself, but someone that helps content generators get it done and online.  If you are curious, use BlackMagic 4K Pro cameras, an BlackMagic ATEM Switcher... but I also am (now) familiar with OBS and vMix streaming solutions.  Using ProPresenter to do the Lower-Thirds and such as well.  But I digress.

As for the IndieDroid Nova, I wanted to run a bunch of stuff in KVM and docker, and not need multiple Pi devices for same.  I plan to take my Pi 4s and make a cluster!  The Nova is a beast in it own right compared to a Pi 4, I have 16GB RAM/64 GB EMMC, but I am actually using 128 GB SD card and older 80GB USB 3.0 enclosed drive for DB and high I/O stuff.

I knew going in, it was going to be tricky with USB WiFi devices and such, but since I had some time this morning to play, I pulled out my USB WiFi favorite dongle for network scanning, and gave it a shot.

Oh, I still plan to get back to that solar panel tracking project, got my hands on a couple of ESP32 with long range WiFi external antennas for the project.  But that project in on Q.T.  [Cough] LOL

morrownr, 2023-07-15:
> I knew going in, it was going to be tricky with USB WiFi devices and such,

It is only tricky if you stay with Realtek based adapters. Have you thought about a Mediatek based adapter such as:

https://www.amazon.com/dp/B08NPX2X4Z/?tag=pandaw-20

It is should work right away on kernel 5.10. There are other adapters, I just through that one up to get your attention.

> I still plan to get back to that solar panel tracking project...

I would like to follow along but late in the Fall is about the earliest I might have time.

Jibun-no-Kage, 2023-07-15:
Yeah, next on the list is to get MediaTek... just don't need it as yet.  But maybe Christmas I will upgrade the WiFi router(s), then will be time do update the adapters.

My client base, uses a lot of UniFi (Ubiquiti) based networking components by the way.

The solar project is way out there... may not get any serious work done on it until Jan 2024... just to much going on right now.

closed 2024-03-27 by morrownr