morrownr/8821cu-20210916 #99: Error! Bad return status for module build on kernel: 5.14.0-284.18.1.el9_2.x86_64 (x86_64)

goyalg8, 2023-07-18:
I am trying to install the driver for wifi usb adapter rtl8821CU on fedora redhat linux 9 with kernal version 5.14.0

it is not working.

Is there any solution to this?

Thank you

morrownr, 2023-07-18:
Hi @goyalg8 

> fedora redhat linux 9 with kernal version 5.14.0

I need you to clarify what distro you are using. The above description is a mix of two different distros. There is Fedora Linux and then there is Red Hat Enterprise Linux, also known as RHEL.

goyalg8, 2023-07-19:
@morrownr , I am using fedora linux with kernal version 5.14.0

morrownr, 2023-07-19:
@goyalg8 

Can you provide a link to download the distro you are using?

Also, can I get you to do a manual compilation and provide the output that appears on screen so that I can see the errors?

morrownr, 2023-07-19:
The manual compilation:

Go to the driver directory:

$ make clean
$ make

goyalg8, 2023-07-20:
Please see the attached screenshot of the error after running the
below command

Sudo ./install-driver.sh

Thanks

closed 2024-03-27 by morrownr