morrownr/8821cu-20210916 #103: problems after installing nvidia-driver

paozival, 2023-07-20:
I'm using Ubuntu 22.04.1, and the linux kernel is 5.19.0.46-generic.
my wireless device information is   0bda : c811 Realtek Semiconductor Corp.

At first , I follow this repo and install the driver manually, and it worked , the  wifi connection showed up.
Then I install the nvidia-driver (through the setting -> Additional drivers), and I reboot the computer.

Two problems occurred ( at least, maybe there are other problems I don't notice):
1. The network was gone , even including the wired connection, I mean disappeared in the setting
2. I tried to install the driver again, but I couldn't  "make" , it said :
`make[1]: *** /lib/modules/5.19.0-46-generic/build: No such file or directory.`

After googling, I found a blog said maybe because I change the gpu driver from the default nouveau to nvidia , I run `sudo lshw -c network` ,it showed 
'*-network UNCLAIMED' . Then I run `sudo dpkg -s linux-modules-extra-$(uname -r) | grep Status`, and it said the system didn't install the package. So I downloaded the linux-modules-extra package and installed it , and the wired connection was back. But I don't really know what is the function of linux-modules-extra, and why the wired network can come back after installing it.

As for the second question , I go to the path of "/lib/modules/$(uname -r)/"  , there is no folder "build" , So I tried to reinstall the kernel by `sudo apt-get install linux-headers-$(uname -r)` , but it said it is already the latest and didn't download. I fixed it by 
`sudo apt-get --reinstall install linux-headers-$(uname -r)` eventually , and can install the driver manually by "make" again. I am a new user of linux , and I don't know if there exist conflict problems between linux kernel and nvidia-driver.

I have tried to reinstall the driver for many times , and the "8821cu.ko" exist in the `/lib/modules/$(uname -r)/kernel/drivers/net/wireless/`  , but  the wifi just never shows up. And I run `sudo modprobe 8821cu` , it said 
`modprobe ERROR: could not insert '8821cu': Invalid  argument`, I try another way `sudo insmod 8821cu.ko`, it gives
`insmod: ERROR: could not insert module 8821cu.ko: Invalid parameters`

Another thing is when I "make" , the information is 
`warning: the compiler differs from the one used to build the kernel
The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
You are using: gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0` is this supposed to be a problem? 

I really need some help , thanks!

morrownr, 2023-07-20:
Hi @paozival 

Please post the results of:

$ dkms status

paozival, 2023-07-24:
> Hi @paozival
> 
> Please post the results of:
> 
> $ dkms status

```
bcmwl/6.30.223.271+bdcom, 5.19.0-46-generic, x86_64: installed
rtl8821cu/5.12.0.4, 5.19.0-46-generic, x86_64: installed
```

morrownr, 2023-07-24:
Well, dkms things both drivers are installed.

My best guess at this point is that the installation of the nvidia driver is messing things up. I do my best to keep the installation of this driver here working in accordance with standard methods whether doing a manual installation or using the install-driver.sh script. I do not have any systems with nvidia cards or chipsets so really don't have any way to test.

My install-driver.sh script does a lot of things besides just install the driver. It also checks many things and it will remove any previous installations. Please run the script and post what it shows on the screen:

$ sudo sh install-driver.sh

paozival, 2023-07-25:
After I run the script `sudo sh install-driver.sh`,it shows 
```
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 12/12 (in-use/total processing units)
: 32669472 (total system memory)
: 5.19.0-46-generic (kernel version)
: gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
: dkms-2.8.7
: ---------------------------

Checking for previously installed drivers.
rtl8821cu/5.12.0.4, 5.19.0-46-generic, x86_64: installed
: ---------------------------

Removing a dkms installation.
Module rtl8821cu-5.12.0.4 for kernel 5.19.0-46-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

8821cu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.19.0-46-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...
Deleting module rtl8821cu-5.12.0.4 completely from the DKMS tree.
Removing 8821cu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821cu-5.12.0.4
Removal complete.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source -> /usr/src/rtl8821cu-5.12.0.4
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh.......
Signing module:
 - /var/lib/dkms/rtl8821cu/5.12.0.4/5.19.0-46-generic/x86_64/module/8821cu.ko
Secure Boot not enabled on this system.
cleaning build area...
Compile time: 122.97 seconds
The driver was built by dkms successfully.
: ---------------------------

8821cu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.0-46-generic/updates/dkms/

depmod...
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.
: ---------------------------

Do you want to edit the driver options file now? (recommended) [Y/n] n
Do you want to apply the new options by rebooting now? (recommended) [Y/n] n

```
the driver options file is default , then I reboot the computer and the wifi still don't shows up. 
Maybe I remove the current nvidia driver and install a new one manually?

morrownr, 2023-07-25:
> The driver was installed by dkms successfully.

That is a clean installation that should work.

What does the following command show?

$ iw dev

> Maybe I remove the current nvidia driver and install a new one manually?

You might as well try this.

paozival, 2023-07-25:
> What does the following command show?
> 
> $ iw dev

it shows nothing

morrownr, 2023-07-25:
No wifi interfaces.

Do you have Secure Boot turned on?

paozival, 2023-07-25:
I checked the secure boot state, and it is off. How to recover the wifi interfaces?

morrownr, 2023-07-25:
> How to recover the wifi interfaces?

If the driver is properly loading, the interfaces will appear. The fact that they are not appearing means there is a problem somewhere...but we know there is a problem. Let's continue investigating.

While we think the nvidia installation or driver seems to be the cause, it will help if we can narrow down what has been done to cause a problem with this Realtek driver.

Is the driver actually loading? Run and post the results of:

$ lsmod | grep 8821cu

paozival, 2023-07-26:
`lsmod | grep 8821cu` , it shows nothing. So the driver is not loading ,right?

I run 'lsusb -tv', it shows

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 20000M/x2
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 9: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 2: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        ID 0b05:19af ASUSTek Computer, Inc. 
    |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 12M
        ID 0b05:19af ASUSTek Computer, Inc. 
    |__ Port 9: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 480M
        ID 0bda:c811 Realtek Semiconductor Corp. 
    |__ Port 10: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
        |__ Port 1: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 046d:c534 Logitech, Inc. Unifying Receiver
        |__ Port 1: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
            ID 046d:c534 Logitech, Inc. Unifying Receiver

```

here, 
>    |__ Port 9: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 480M
>        ID 0bda:c811 Realtek Semiconductor Corp. 

we can see that this c811 has no driver for it.

morrownr, 2023-07-26:
> lsmod | grep 8821cu , it shows nothing. So the driver is not loading ,right?

Correct, the driver is not loading.

I am not sure exactly what has happened but the problem showing up when the nvidia driver was installed makes it look like the nvidia driver caused the problem.

Let me tell you how I handle the Ubuntu Additional Drivers option: I ignore it. I never even open it because many of the drivers that it can load are poorly maintained. The rtl8812au driver that it offers is very old and a little over a year ago it was causing problems for users. I wrote to the maintainer that he was welcome to use my driver here and I would help. I never even got a reply. 

Side issue: While I appreciate the work that Ubuntu has done on Linux over the years, I am currently testing Debian 12 to see if I want to change over to it for my main dev box and the operational systems here at my place. So far, Debian is looking really good and I will likely make the change soon. Ubuntu has a history of making bad decisions and they are continuing to do so.

The only solutions I can offer in this case is for you to uninstall the nvidia driver. If that does not work, you may need to save your important files and do a clean installation. There is likely another location to get the nvidia driver, if you need it. I would investigate further but I do have any nvidia video cards so I have no way to investigate.

Good luck.

closed 2024-03-27 by morrownr