morrownr/8821cu-20210916 #105: Driver sometimes doesn't load on boot, warning rtw_os_ndevs_register+0xc9/0x156

neilthomson, 2023-07-28:
I have installed the driver and it works, but only sometimes after boot. If I rmmod 8821cu and then modprobe 8821cu the device comes back up.

dmesg error after boot

```
[    5.342009] ------------[ cut here ]------------
[    5.342012] WARNING: CPU: 5 PID: 506 at /var/lib/dkms/rtl8821cu/5.12.0.4/build/os_dep/linux/os_intfs.c:3745 rtw_os_ndevs_register+0xc9/0x156 [8821cu]
[    5.342173] Modules linked in: snd_sof_pci_intel_tgl snd_sof_intel_hda_common soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci intel_rapl_msr snd_sof_xtensa_dsp snd_sof snd_soc_hdac_hda mei_hdcp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi soundwire_bus ledtrig_audio snd_soc_core snd_hda_codec_hdmi snd_compress intel_rapl_common ac97_bus snd_pcm_dmaengine intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp snd_hda_intel coretemp snd_usb_audio snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec kvm_intel snd_hda_core iwlmvm snd_usbmidi_lib snd_hwdep kvm binfmt_misc snd_seq_midi input_leds snd_seq_midi_event rapl snd_rawmidi mac80211 mc btusb intel_cstate btrtl serio_raw btbcm 8821cu(OE+) usblp snd_pcm libarc4 snd_seq btintel ftdi_sio bluetooth snd_seq_device snd_timer ecdh_generic iwlwifi usbserial ee1004 ecc snd mei_me cfg80211 soundcore mei igen6_edac mac_hid intel_hid acpi_tad acpi_pad sparse_keymap sch_fq_codel msr parport_pc
[    5.342246]  ppdev lp parport ramoops pstore_blk reed_solomon pstore_zone efi_pstore ip_tables x_tables autofs4 btrfs blake2b_generic zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear hid_cmedia hid_generic usbhid hid i915 i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops cec ahci rc_core i2c_i801 intel_lpss_pci crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd e1000e thunderbolt drm i2c_smbus igc intel_lpss libahci idma64 nvme xhci_pci xhci_pci_renesas nvme_core video pinctrl_tigerlake
[    5.342306] CPU: 5 PID: 506 Comm: systemd-udevd Tainted: G           OE     5.15.0-78-generic #85~20.04.1-Ubuntu
[    5.342311] Hardware name: Simply NUC NUC11EMi5/Rev PB01-TGL-V10, BIOS PB01 V3.2 02/18/2022
[    5.342313] RIP: 0010:rtw_os_ndevs_register+0xc9/0x156 [8821cu]
[    5.342442] Code: 87 cc 48 00 00 49 8d b6 b7 3b 00 00 84 c0 74 0d 49 8d b6 c7 3b 00 00 3c 01 49 0f 45 f7 e8 27 ec ff ff 41 89 c5 83 f8 01 74 aa <0f> 0b 48 63 c3 4d 8d 34 c4 85 db 74 6e 4c 89 e7 e8 2e 3f 02 00 e9
[    5.342445] RSP: 0018:ffffaba8c0c579c8 EFLAGS: 00010297
[    5.342449] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000048
[    5.342451] RDX: ffffaba8c0c57888 RSI: ffffaba8c0c57880 RDI: ffffaba8c0c57948
[    5.342453] RBP: ffffaba8c0c579f8 R08: 0000000000000001 R09: 0000000000000000
[    5.342455] R10: 0000000000000000 R11: 0000000000000001 R12: ffff9190177d8000
[    5.342458] R13: 0000000000000000 R14: ffffaba8c0811000 R15: ffffffffc1059314
[    5.342461] FS:  00007f50a77fe880(0000) GS:ffff919087b40000(0000) knlGS:0000000000000000
[    5.342465] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    5.342468] CR2: 0000559073883078 CR3: 00000001032d0002 CR4: 0000000000770ee0
[    5.342470] PKRU: 55555554
[    5.342471] Call Trace:
[    5.342473]  <TASK>
[    5.342477]  rtw_os_ndevs_init+0x2d/0x41 [8821cu]
[    5.342615]  rtw_drv_init+0x644/0x7aa [8821cu]
[    5.342742]  usb_probe_interface+0xea/0x290
[    5.342748]  really_probe.part.0+0xc8/0x380
[    5.342753]  really_probe+0x40/0x80
[    5.342755]  __driver_probe_device+0x119/0x190
[    5.342759]  driver_probe_device+0x23/0xb0
[    5.342762]  __driver_attach+0xc5/0x180
[    5.342765]  ? __device_attach_driver+0x140/0x140
[    5.342768]  bus_for_each_dev+0x7c/0xd0
[    5.342774]  driver_attach+0x1e/0x30
[    5.342776]  bus_add_driver+0x178/0x220
[    5.342779]  driver_register+0x74/0xe0
[    5.342782]  usb_register_driver+0x89/0x130
[    5.342787]  ? 0xffffffffc0e05000
[    5.342791]  rtw_drv_entry+0xed/0x1000 [8821cu]
[    5.342849]  do_one_initcall+0x46/0x1e0
[    5.342855]  ? __cond_resched+0x19/0x40
[    5.342861]  ? kmem_cache_alloc_trace+0x15a/0x420
[    5.342865]  do_init_module+0x52/0x230
[    5.342871]  load_module+0x1294/0x1500
[    5.342877]  __do_sys_finit_module+0xbf/0x120
[    5.342881]  ? __do_sys_finit_module+0xbf/0x120
[    5.342886]  __x64_sys_finit_module+0x1a/0x20
[    5.342890]  do_syscall_64+0x59/0xc0
[    5.342895]  entry_SYSCALL_64_after_hwframe+0x61/0xcb
[    5.342899] RIP: 0033:0x7f50a7e2973d
[    5.342902] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 23 37 0d 00 f7 d8 64 89 01 48
[    5.342905] RSP: 002b:00007ffe973179a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    5.342908] RAX: ffffffffffffffda RBX: 0000557a91774450 RCX: 00007f50a7e2973d
[    5.342910] RDX: 0000000000000000 RSI: 0000557a917873e0 RDI: 0000000000000006
[    5.342912] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000000
[    5.342914] R10: 0000000000000006 R11: 0000000000000246 R12: 0000557a917873e0
[    5.342915] R13: 0000000000000000 R14: 0000557a9177c0d0 R15: 0000557a91774450
[    5.342918]  </TASK>
[    5.342920] ---[ end trace 20b6202c98a7e79c ]---
```

Log files from both occasions

[rtw_successfull_load.log](https://github.com/morrownr/8821cu-20210916/files/12198715/rtw_successfull_load.log)
[rtw_failed_to_load.log](https://github.com/morrownr/8821cu-20210916/files/12198716/rtw_failed_to_load.log)

morrownr, 2023-07-28:
Hi @neilthomson 

There is a pinned issue at the top called Problem Reporting Guidelines...

Please read and provide the requested information.

neilthomson, 2023-07-30:
missed that sorry, info here:

```
cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```
It's not a VM install

```
sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 8/8 (in-use/total processing units)
: 3738076 (total system memory)
: 5.15.0-78-generic (kernel version)
: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
: dkms:2.8
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source ->
                 /usr/src/rtl8821cu-5.12.0.4

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh............
Signing module:
 - /var/lib/dkms/rtl8821cu/5.12.0.4/5.15.0-78-generic/x86_64/module/8821cu.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.
Compile time: 202.26 seconds
The driver was built by dkms successfully.
: ---------------------------

8821cu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-78-generic/updates/dkms/

depmod...

[... 170 lines cut ...]
        (5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
        (5815 - 5835 @ 20), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN
        ```
        
I also have a systemd network rename rule running which I can disable but I have the same issues

```
cat /etc/systemd/network/5-wifidongle.link
[Match]
Driver=rtl8821cu

[Link]
Description="BrosTrend Wi-Fi Adapter"
Name=wifidongle

```

The internal Wifi is set as an Access Point using netplan. I have a NetworkManager rule to only manage the wifi dongle

```
cat /etc/NetworkManager/conf.d/99-unmanaged-devices.conf
[keyfile]
unmanaged-devices=interface-name:*,except:interface-name:wifidongle

```

closed 2024-03-27 by morrownr