morrownr/8821cu-20210916 #107: OpenWrt Driver Install

tommy1223, 2023-08-01:
Hi, may I know how to install driver for 8811cu on OpenWrt? I am not sure which version of linux version should I follow for OpenWrt. Great if there is help on this.

Thank you!

morrownr, 2023-08-01:
Hi @tommy1223 

The best place to get an idea of what this requires is at the OpenWRT site in the forums. What you are asking takes a lot of work.

I run USB WiFi adapters with OpenWRT. None of them are adapters that use Realtek out-of-kernel drivers. The best solution is to get an adapter that uses a driver that is already available in OpenWRT. This site Main Menu has a lot of information:

https://github.com/morrownr/USB-WiFi

I recommend reading menu item 1 and then go over the lists of adapters in menu item 2. I use and recommend 3 adapters for using with OpenWRT:

Alfa AXM
Alfa ACM
Alfa ACHM

You will find them in menu item 2. Skip the sections on the rtl8812bu and rtl8811cu as the in-kernel drivers won't find their way into OpenWRT for 1 to 2 years.

Questions?

tommy1223, 2023-08-02:
Hi @morrownr,

Got it, let me get another adapter for this purpose then.

Thanks!