morrownr/8821cu-20210916 #110: Install fails on Raspberry Pi OS with kernel 6.1.42-v8+ (aarch64)

mirek-burkon, 2023-08-15:
Hi, can you please advise me on this?

sudo ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: aarch64 (architecture)
: 4/4 (in-use/total processing units)
: 3884804 (total system memory)
: 6.1.42-v8+ (kernel version)
: gcc (Debian 10.2.1-6) 10.2.1 20210110
: dkms:2.8
: ---------------------------

Checking for previously installed drivers.
rtl8821cu, 5.12.0.4: added
rtl8821CU, 5.4.1, 5.10.0-24-arm64, aarch64: installed
: ---------------------------

Removing a dkms installation.

------------------------------
Deleting module version: 5.12.0.4
completely from the DKMS tree.
------------------------------
Done.
Removing 8821cu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821cu-5.12.0.4
Removal complete.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source ->
                 /usr/src/rtl8821cu-5.12.0.4

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh..........................................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.42-v8+ (aarch64)
Consult /var/lib/dkms/rtl8821cu/5.12.0.4/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

cat /var/lib/dkms/rtl8821cu/5.12.0.4/build/make.log
DKMS make.log for rtl8821cu-5.12.0.4 for kernel 6.1.42-v8+ (aarch64)
Mon 14 Aug 2023 05:38:10 PM PDT
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.42-v8+/build M=/var/lib/dkms/rtl8821cu/5.12.0.4/build  modules
make[1]: Entering directory '/home/killbot/linux-431319d91b8584c9f28b195ab9a97d7e78905aeb'
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_mi.o
[... 155 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/phydm/rtl8821c/phydm_regconfig8821c.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/phydm/rtl8821c/phydm_rtl8821c.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/phydm/halrf/rtl8821c/halhwimg8821c_rf.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/phydm/halrf/rtl8821c/halrf_8821c.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/btc/halbtc8821cwifionly.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/btc/halbtc8821c1ant.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/hal/btc/halbtc8821c2ant.o
  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/platform/platform_ops.o
  LD [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.o
  MODPOST /var/lib/dkms/rtl8821cu/5.12.0.4/build/Module.symvers
ERROR: modpost: "__arch_copy_to_user" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "preempt_schedule_notrace" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "alt_cb_patch_nops" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "__udelay" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
ERROR: modpost: "__const_udelay" [/var/lib/dkms/rtl8821cu/5.12.0.4/build/8821cu.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:126: /var/lib/dkms/rtl8821cu/5.12.0.4/build/Module.symvers] Error 1
make[1]: *** [Makefile:1964: modpost] Error 2
make[1]: Leaving directory '/home/killbot/linux-431319d91b8584c9f28b195ab9a97d7e78905aeb'
make: *** [Makefile:2501: modules] Error 2

morrownr, 2023-08-15:
Hi @mirek-burkon 

That is a little mess you have there.

The devs of the Raspberry Pi OS are notorious for screwing up the header files and this very well could be one of those cases. However. In looking at the age of the distro as it originally was installed and the much more modern kernel, it may be hard for us to sort this out. It may be easier to backup the important files and do a clean installation with a current version of RasPiOS.

With that said, If your Pi is a Pi4B/400/CM4, be careful about installing the 32 bit version as they have a trap for you so I recommend the 64 bit version.

mirek-burkon, 2023-08-16:
Thanks @morrownr, seems like I messed up the image myself, possibly by apt full-upgrade. Yes, it's Pi 4B and the newer kernel seems necessary to get the VideoCore VI chipset working. I rolled back to the original Pi OS with 6.1.21-v8+ kernel and compiled 8821cu just fine following your description. Appreciate the help!

closed 2023-08-16 by mirek-burkon

morrownr, 2023-08-17:
Glad you are working again. I understand the desire to get that newer drivers working but adding a bleeding edge kernel sometimes can make a mess.

Cheers