morrownr/8821cu-20210916 #112: Your kernel header files aren't properly installed. Raspi4 using Raspian OS Lite Feb  2023

ashah7, 2023-08-30:
Getting the below error, installation via script or manual compile 
Only shows on the Raspi4 (3 works fine)
Using Raspian OS Lite Feb 2023

```

pi@1db291:~/8821cu-20210916 $ uname -a
Linux 10000000255db291 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux

pi@10b291:~/8821cu-20210916 $ uname -r
5.15.84-v7l+
```

```
pi@100291:~/8821cu-20210916 $ sudo apt-get install raspberrypi-kernel-headers 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
The following packages were automatically installed and are no longer required:
  dc libgnutls-dane0 libunbound8
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 49 not upgraded.

pi@100291:~/8821cu-20210916 $ sudo ./install-driver.sh 
Your kernel header files aren't properly installed.
Please consult your distro documentation or user support forums.
Once the header files are properly installed, please run "sudo ./install-driver.sh"

```
```

pi@10291:~/8821cu-20210916 $ make clean
#make -C  M=/home/pi/8821cu-20210916 clean
cd hal ; rm -fr */*/*/*.mod.c */*/*/*.mod */*/*/*.o */*/*/.*.cmd */*/*/*.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
rm -fr MOK.der MOK.priv

pi@1b291:~/8821cu-20210916 $ make
make ARCH= CROSS_COMPILE= -C  M=/home/pi/8821cu-20210916  modules
make[1]: *** M=/home/pi/8821cu-20210916: No such file or directory.  Stop.
make: *** [Makefile:2501: modules] Error 2

```

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr dc:a6:32:3d:6b:37
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 31.00 dBm
global
country 00: DFS-UNSET
	(2402 - 2472 @ 40), (6, 20), (N/A)
	(2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)
```

morrownr, 2023-08-30:
Try this:

https://forums.raspberrypi.com/viewtopic.php?p=2091532&hilit=Tp+link#p2091532

ashah976, 2023-08-30:
that didn't help
But I got it fixed by building the headers using `rpi-source`

`https://github.com/RPi-Distro/rpi-source`

not sure if this was right way to do it but it worked

morrownr, 2023-08-30:
Glad you got it fixed. The RasPiOS devs are notorious for messing up the RasPiOS header files.

closed 2024-03-27 by morrownr