morrownr/8821cu-20210916 #114: error logging?

rai68, 2023-09-08:
Is there any sort of readable log file for the drivers?
I have a program that uses gopacket, which calls https://pkg.go.dev/github.com/google/gopacket/pcap#InactiveHandle.Activate.

The program errros out with a '-1' which is a unknown error to gopacket/pcap. And im trying to figure out why, I suspected the drivers are failing or not doing something properly and wanted to see if i could find out why .Activate() errors with -1. 

I saw ./save-log.sh, but i dont understand any of it.

morrownr, 2023-09-08:
Hi @rai68 

> Is there any sort of readable log file for the drivers?

The driver adds to the main system log:

$ sudo dmesg

Default setting is low so you might need to increase the level. 

$ sudo ./edit-options.sh

That will open the configuration file for the driver and it contains some documentation as well.

> The program errros out with a '-1' which is a unknown error to gopacket/pcap.

This is an out-of-kernel driver by Realtek. Realtek's out-of-kernel drivers can be a challenge when it comes to what you are doing. You might have better luck with the new in-kernel driver but you need at least kernel 6.2 to have that support. I have not tested it yet. I have tested the new 8812bu in-kernel driver and it works well but I have not tested what you are doing.

Maybe that info can help nudge you in the right direction. Another place to post would be in Issues in the repo that contains the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

The Main Menu contains a lot of things that could be useful to users of USB WiFi adapters. I'd recommend at least reading menu item 1. There are a lot of very smart people that stop by Issues in that repo but not everyone stops by everyday as this is not a social media site. It does get about 20k hit per week.

@morrownr

rai68, 2023-09-21:
To anyone with the same issue, I had to build the specific 6.2 kernel for raspberry pi and overwrite the old modules and image. following [this](https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compiling-the-kernel). Make sure in the `make menuconfig` you select the drivers for your WLAN card (assuming they're in the 6.2 kernel)

closed 2024-03-27 by morrownr