morrownr/8821cu-20210916 #117: dhcp fails on reboot with ubuntu 22.04 lts on raspberry pi 4

blacksteel1288, 2023-09-24:
After a recent apt upgrade the wifi adapter no longer gets an IP address after reboot.  This has been working fine for about a year, and only started failing now.

Interestingly after reboot, if I run a "sudo netplan apply", the adapter does get an IP address and it works fine until the next reboot.

Here's the diagnostic info after reboot:

`sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get`

```
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0825:0001 GC Protronics X825 V1.2
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8821cu/5.12.0.4, 5.15.0-1037-raspi, aarch64: installed (WARNING! Diff between built and installed module!)
zfs/2.1.11, 5.15.0-1037-raspi, aarch64: installed
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 00:13:25:b1:4f:2d
		type managed
		txpower -100.00 dBm
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```

morrownr, 2023-09-24:
Hi @blacksteel1288 

Try this first. From your driver directory, run the following:

$ git pull
$ sudo sh install-driver.sh

blacksteel1288, 2023-09-25:
Hi @morrownr,

Yes, thank you, I did try that already.  The directory and files are all up-to-date, latest revs.

morrownr, 2023-09-25:
Let me say this first: The RasPiOS devs are notorious for breaking things related to the these Realtek out-of-kernel drivers.

I looked at the info you posted again and it is not obvious what is wrong. Can I get you to run `sudo sh install.driver` again and this time post the screen output in your replay. You do not need to run remove-driver.sh first as install-driver.sh will remove any previous driver before it compiles and installs the new compile. Maybe that will give us a hint.

blacksteel1288, 2023-09-25:
Understood on the raspi devices.  This setup has been stable for a very long time though, so something must have changed.  Also, it is running Ubuntu Server, not Raspberry Pi OS, fwiw.

Here's the output:

```
ubuntu@ragnar:~/src/8821cu-20210916$ sudo sh install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: aarch64 (architecture)
: 4/4 (in-use/total processing units)
: 1889260 (total system memory)
: 5.15.0-1037-raspi (kernel version)
: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
: dkms-2.8.7
: ---------------------------

Checking for previously installed drivers.
rtl8821cu/5.12.0.4, 5.15.0-1037-raspi, aarch64: installed (WARNING! Diff between built and installed module!)
: ---------------------------

Removing a dkms installation.
Module rtl8821cu-5.12.0.4 for kernel 5.15.0-1037-raspi (aarch64).
Before uninstall, this module version was ACTIVE on this kernel.

8821cu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.0-1037-raspi/
rmdir: failed to remove '': No such file or directory
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.......
Deleting module rtl8821cu-5.12.0.4 completely from the DKMS tree.
Removing 8821cu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8821cu-5.12.0.4
Removal complete.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source -> /usr/src/rtl8821cu-5.12.0.4
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh.........................................................................................
cleaning build area...
Compile time: 765.06 seconds
The driver was built by dkms successfully.
: ---------------------------

8821cu.ko:
Running module version sanity check.
Error! Module version v5.12.0.4-1-g9241a6516.20210916_COEX20200730-5151 for 8821cu.ko
is not newer than what is already found in kernel 5.15.0-1037-raspi (v5.12.0-16-g5229e7f95.20210118_COEX20200730-5151).
You may override by specifying --force.

depmod.......
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.
: ---------------------------

Do you want to edit the driver options file now? (recommended) [Y/n] n
Do you want to apply the new options by rebooting now? (recommended) [Y/n] 
```

I noticed the comment about override by specifying --force, but wasn't clear to me if/how I should do that.  Otherwise, no errors that I could see.

morrownr, 2023-09-25:
Hmmm...

Let me see the results of:

$ dkms status

In the mean time, I have an upgrade to install-driver.sh that I have not applied to this driver yet. It provides more information, detects more possible problems and does a better job of cleaning out old conflicting drivers. I try to get that merged today so we can see if that helps.

One last thing. These Realtek drivers do not work well in situations where there will be updates and upgrades. It is only a matter of time until something happens. We have options in this day in time and I look at the options at my site Main Menu:

https://github.com/morrownr/USB-WiFi

Take a look at menu items 1 and 2. The adapters that use the in-kernel standards compliant drivers are very stable and work well for situations where we are looking for 247/365 uptime.

blacksteel1288, 2023-09-25:
Great!  I'd be happy to check out a new build for the install-driver.sh script.  In the meantime, here's the other info:

```
ubuntu@ragnar:~$ sudo dkms status
rtl8821cu/5.12.0.4, 5.15.0-1037-raspi, aarch64: installed (WARNING! Diff between built and installed module!)
zfs/2.1.11, 5.15.0-1037-raspi, aarch64: installed
```

Also, I thought this might be helpful ---

```
ubuntu@ragnar:~$ sudo dmesg | grep 8821
[  170.222849] RTW: rtl8821cu v5.12.0-16-g5229e7f95.20210118_COEX20200730-5151
[  170.222859] RTW: rtl8821cu BT-Coex version = COEX20200730-5151
[  170.509946] usbcore: registered new interface driver rtl8821cu
[ 2171.643018] UBSAN: invalid-load in /var/lib/dkms/rtl8821cu/5.12.0/build/hal/phydm/phydm_ccx.c:1375:2
[ 2171.658527]  phydm_nhm_mntr_set+0x168/0x16c [8821cu]
[ 2171.658753]  phydm_env_mntr_trigger+0x54/0x174 [8821cu]
[ 2171.658894]  rtw_acs_trigger+0x88/0xf4 [8821cu]
[ 2171.659034]  sitesurvey_cmd_hdl+0xdc4/0x1018 [8821cu]
[ 2171.659173]  rtw_cmd_thread+0x4b4/0x690 [8821cu]
[ 2176.034204] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821cu/5.12.0/build/hal/phydm/halrf/halrf.c:445:17
[ 2176.050222]  halrf_iqk_dbg+0xa0/0x954 [8821cu]
[ 2176.050422]  phy_iq_calibrate_8821c+0x148/0x1cc [8821cu]
[ 2176.050560]  halrf_segment_iqk_trigger+0x20c/0x280 [8821cu]
[ 2176.050698]  rtw_phydm_iqk_trigger+0x54/0x64 [8821cu]
[ 2176.050835]  rtl8821c_switch_chnl_and_set_bw+0x35c/0x36c [8821cu]
[ 2176.050971]  rtl8821c_handle_sw_chnl_and_set_bw+0x1fc/0x21c [8821cu]
[ 2176.051109]  rtl8821c_set_channel_bw+0x38/0x44 [8821cu]
[ 2176.051247]  rtw_hal_set_chnl_bw+0xd0/0x15c [8821cu]
[ 2176.051384]  set_channel_bwmode+0x238/0x274 [8821cu]
[ 2176.051520]  join_cmd_hdl+0x2b0/0x320 [8821cu]
[ 2176.051656]  rtw_cmd_thread+0x4b4/0x690 [8821cu]
[ 2176.068748] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821cu/5.12.0/build/hal/phydm/halrf/halrf.c:445:17
[ 2176.084755]  halrf_iqk_dbg+0xac/0x954 [8821cu]
[ 2176.084919]  phy_iq_calibrate_8821c+0x148/0x1cc [8821cu]
[ 2176.085056]  halrf_segment_iqk_trigger+0x20c/0x280 [8821cu]
[ 2176.085193]  rtw_phydm_iqk_trigger+0x54/0x64 [8821cu]
[ 2176.085330]  rtl8821c_switch_chnl_and_set_bw+0x35c/0x36c [8821cu]
[ 2176.085467]  rtl8821c_handle_sw_chnl_and_set_bw+0x1fc/0x21c [8821cu]
[ 2176.085604]  rtl8821c_set_channel_bw+0x38/0x44 [8821cu]
[ 2176.085741]  rtw_hal_set_chnl_bw+0xd0/0x15c [8821cu]
[ 2176.085877]  set_channel_bwmode+0x238/0x274 [8821cu]
[ 2176.086014]  join_cmd_hdl+0x2b0/0x320 [8821cu]
[ 2176.086149]  rtw_cmd_thread+0x4b4/0x690 [8821cu]
```

morrownr, 2023-09-25:
> I'm not sure if those UBSAN errors matter

Not for this problem but I'll look at it later.

Things are becoming a little more clear. Follow these instructions:

$ sudo sh remove-driver.sh

Delete the driver directory

$ sudo reboot

Following the installation Steps in the README again. 

The driver in the repo is actually a new version from what you download a year ago. That may be causing problems. I have merged all the new stuff now. That probably would have handled the issue but maybe not so starting clean may be best.

blacksteel1288, 2023-09-27:
Followed the steps, but same result.  No dhcp on reboot.  There were no error messages on the build.

```
ubuntu@ragnar:~$ sudo dkms status
rtl8821cu/5.12.0.4, 5.15.0-1037-raspi, aarch64: installed (WARNING! Diff between built and installed module!)
zfs/2.1.11, 5.15.0-1037-raspi, aarch64: installed
```

morrownr, 2023-09-27:
Ben,

The hint is the follow:

installed (WARNING! Diff between built and installed module!)     < you can google this

This seems to indicate there is another copy of the driver that is installed and dkms is refusing to overwrite it. Why? I do not know yet.

To see all of the 8821cu.ko drivers in your system:

$ cd /
$ sudo grep -r "8821cu.ko*" *

I am having health problems and am going slowly. You might want to google as I posted above but be careful not to make a mess of your system. I will work on this as I have time today and am able.

blacksteel1288, 2023-09-29:
ok.  got it.  here's the output of that grep:

```
ubuntu@ragnar:/$ sudo grep -r "8821cu.ko*" *
home/ubuntu/src/8821cu-20210118/README.md:## 8821cu ( 8821cu.ko ) :rocket:
home/ubuntu/src/8821cu-20210118/.8821cu.ko.cmd:cmd_/home/ubuntu/src/8821cu-20210118/8821cu.ko := ld -r -EL  -maarch64elf --build-id=sha1  -T scripts/module.lds -o /home/ubuntu/src/8821cu-20210118/8821cu.ko /home/ubuntu/src/8821cu-20210118/8821cu.o /home/ubuntu/src/8821cu-20210118/8821cu.mod.o;  true
home/ubuntu/src/8821cu-20210118/modules.order:/home/ubuntu/src/8821cu-20210118/8821cu.ko
home/ubuntu/src/8821cu-20210118/.modules.order.cmd:cmd_/home/ubuntu/src/8821cu-20210118/modules.order := {   echo /home/ubuntu/src/8821cu-20210118/8821cu.ko; :; } | awk '!x[$$0]++' - > /home/ubuntu/src/8821cu-20210118/modules.order
home/ubuntu/src/8821cu-20210916/remove-driver.sh:# Example: /usr/lib/modules/5.15.80-rockchip64/kernel/drivers/net/wireless/rtl8821cu/8821cu.ko.xz
home/ubuntu/src/8821cu-20210916/Makefile:	@$(KSRC)/scripts/sign-file sha256 MOK.priv MOK.der 8821cu.ko
home/ubuntu/src/8821cu-20210916/README.md:## 8821cu ( 8821cu.ko ) :rocket:
home/ubuntu/src/8821cu-20210916/install-driver.sh:# Example: /usr/lib/modules/5.15.80-rockchip64/kernel/drivers/net/wireless/rtl8821cu/8821cu.ko.xz
grep: lib/modules/5.15.0-1037-raspi/modules.dep.bin: binary file matches
lib/modules/5.15.0-1037-raspi/modules.dep:updates/dkms/8821cu.ko: kernel/net/wireless/cfg80211.ko
grep: proc/sys/fs/binfmt_misc/register: Invalid argument
grep: proc/sys/net/ipv4/route/flush: Permission denied
grep: proc/sys/net/ipv6/conf/all/stable_secret: Input/output error
grep: proc/sys/net/ipv6/conf/default/stable_secret: Input/output error
grep: proc/sys/net/ipv6/conf/eth0/stable_secret: Input/output error
grep: proc/sys/net/ipv6/conf/lo/stable_secret: Input/output error
grep: proc/sys/net/ipv6/conf/wlan0/stable_secret: Input/output error
grep: proc/sys/net/ipv6/route/flush: Permission denied
grep: proc/sys/vm/compact_memory: Permission denied
grep: proc/sys/vm/drop_caches: Permission denied
grep: proc/kmsg: Resource temporarily unavailable
```

And, from the top article on that dkms warning, it looks like the solution is to add the '--force' option to the dkms install command.  I'll try modifying your install-driver.sh script to add that option and see if it handles that issue.

blacksteel1288, 2023-10-01:
I modified the ./install-driver.sh script to use the '--force' option on the dkms install line, and now the dkms status looks clean.  However, the problem still persists in that dhcp fails to get an IP address for wlan0.

```
ubuntu@ragnar:~$ sudo dkms status
rtl8821cu/5.12.0.4, 5.15.0-1038-raspi, aarch64: installed
zfs/2.1.11, 5.15.0-1038-raspi, aarch64: installed
```

Again, doing a 'sudo netplan apply' from the command line after a reboot does get dhcp to work properly and gets an IP address.

I'm wondering if there's an option in the dkms.conf that might help, but not sure if that's the right path to try.

morrownr, 2023-10-01:
Hi @blacksteel1288 

Sorry for the delay as I am only working when able.

> I modified the ./install-driver.sh script to use the '--force' option on the dkms install line...

> The dkms status is now clean, however, without an error message.

Thank you. I had considered using '--force' at one point as I suspected it would help in some situations but I could not pinpoint the reason. I think it is time for me to add it and test it in my 2 test repos. This being one of them. I need to find out if there are negative situations for using it.

> now I see 3 lines that say "EFI variables are not supported on this system". Could that be the issue?

No. All that means is that there is no support for Secure Boot and that is normal on RasPi's. I need to clean up that code. It is relatively new code and I simply need to come up with a solution. There is no problem at all with seeing this output.

> doing a 'sudo netplan apply' from the command line after a reboot does get dhcp to work properly and gets an IP address.

I am beginning to think that this problem is not a problem with the driver but rather something in the network configuration. I have no experience with netplan. Ubuntu 23.10 will be released shortly. You might consider doing a clean installation and then you would have two options. You could see if you get adequate performce from the in-kernel driver and if not, you could install this driver and check it out to make sure it is working. Then you could slowly go about configuring the system while checking to see if wifi is still working as you go along. That would allow you to pinpoint where the problem starts.

Can I get you to post the line where you used '--force' so I can see exactly what you did?

Nick

blacksteel1288, 2023-10-02:
Hi Nick,

> Can I get you to post the line where you used '--force' so I can see exactly what you did?

Sure, the line I modified is this one:

`dkms install -m ${DRV_NAME} -v ${DRV_VERSION} -k "${KVER}/${KARCH}" -c "/usr/src/${DRV_NAME}-${DRV_VERSION}/dkms.conf" --force`

I suspect you may be correct on the network configuration issue, although again, I didn't make any changes there.  I'll dig into netplan to see what I can discover.  

If there are any experts out there with experience with netplan or issues like that also on Ubuntu 22.04 LTS and Raspberry Pi 4, please comment back!

Thank you!

morrownr, 2023-10-03:
> Sure, the line I modified is this one:

Interesting. You modified the 'install' line and I modified both 'build' lines. You can see my lines below. I am going to have to relook this before merging it for testing. Thanks.

```
if command -v /usr/bin/time >/dev/null 2>&1; then
		/usr/bin/time -f "Compile time: %U seconds" dkms build -m ${DRV_NAME} -v ${DRV_VERSION} -k "${KVER}/${KARCH}" -c "/usr/src/${DRV_NAME}-${DRV_VERSION}/dkms.conf" --force
	else
		dkms build -m ${DRV_NAME} -v ${DRV_VERSION} -k "${KVER}/${KARCH}" -c "/usr/src/${DRV_NAME}-${DRV_VERSION}/dkms.conf" --force
	fi
```

> I suspect you may be correct on the network configuration issue, although again, I didn't make any changes there. I'll dig into netplan to see what I can discover.

I may be looking at this from a different angle than you are. What I am getting at is there could have been an upstream code change in one of the many parts that make up the networking system that could be causing this. You might post an issue in the distro forums and if the right person sees it, help could be at hand. I do not have any idea at this point but will keep my eyes open.

blacksteel1288, 2023-10-07:
Hi Nick,

After much research, you were correct!  There was some upstream code change in the networking system, and it is documented as a Ubuntu bug here:

[https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/2036358](https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/2036358)

I did not install the patch as directed in the thread, but instead used the workaround mentioned at the bottom to remove `'optional: true'` from my netplan yaml settings file.

Then, I had to make this update as described here: [https://unix.stackexchange.com/questions/710089/systemd-networkd-wait-online-service-consistently-overwritten-by-updates](https://unix.stackexchange.com/questions/710089/systemd-networkd-wait-online-service-consistently-overwritten-by-updates)

By doing a `sudo systemctl edit systemd-networkd-wait-online.service` and adding this section

```
[Service]
ExecStart=
ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --any --ignore eth0
```

After that, on reboot the dhcp for the wlan0 interface works correctly.  Back to normal!

I am curious though, to try out the in-kernel driver you mentioned in this thread --

Now, if a do a remove-driver.sh, the compiled driver from this repo is removed by dkms, however, the default in-kernel driver does not find the adapter anymore.  Interestingly, the in-kernel driver was working earlier in this thread when I built and removed the driver from this repo, so I'm wondering if I need to re-activate it somehow?

morrownr, 2023-10-07:
> Back to normal!

Great! Good job.

> I am curious though, to try out the in-kernel driver you mentioned in this thread --

The 8821cu.conf file that is automatically installed with 'install-driver.sh' and uninstalled with 'remove-driver.sh' has a blacklist line in it. Using remove-driver.sh will remove the file and thus the blacklist.

Important: The in-kernel driver went into mainline with kernel 6.2 so you must be using at least kernel 6.2.

morrownr, 2023-10-07:
It appears your best bet is Ubuntu 23.10 as it will be released very soon and will have kernel 6.5. Well, I guess the RasPi version is released at the same time as the x86_64 versions.

blacksteel1288, 2023-10-08:
Will do.  Thanks!

closed 2023-10-08 by blacksteel1288

morrownr, 2023-10-08:
No problem. Keep me posted.

The reason I did not mention the RasPiOS is that the new version to be released version soon will still use kernel 6.1. The RasPiOS devs seen to only upgrade with LTS versions and the next LTS will probably be 6.7. RasPi does keep versions of all new kernels for testing so if you want to go with the new version when it is available, I have a good guide for compiling your own kernel so you might update to kernel 6.5 and make sure the firmware is installed. I don't have the guide on the Main Menu so you would need to ask me for it. I guess I should put it on the Main Menu.

Cheers

blacksteel1288, 2023-10-13:
Until RasPiOS allows zfs support, I'm stuck on Ubuntu LTS but thanks!