morrownr/8821cu-20210916 #118: Porting the driver to the 6.1.x kernel

apollo80, 2023-10-03:
Greetings to all.
Is it possible to port this driver to the latest LTS version of the kernel - 6.1.x. ?
I did something
```
diff --git a/os_dep/linux/os_intfs.c b/os_dep/linux/os_intfs.c
index 47182d4..6ae54a4 100644
--- a/os_dep/linux/os_intfs.c
+++ b/os_dep/linux/os_intfs.c
@@ -2147,8 +2147,12 @@ int rtw_os_ndev_register(_adapter *adapter, const char *name)
        struct net_device *ndev = adapter->pnetdev;
        u8 rtnl_lock_needed = rtw_rtnl_lock_needed(dvobj);

-#ifdef CONFIG_RTW_NAPI
+#ifdef CONFIG_RTW_NAPI
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0)
+       netif_napi_add_weight(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
+#else
        netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
+#endif
 #endif /* CONFIG_RTW_NAPI */

 #if defined(CONFIG_IOCTL_CFG80211)
diff --git a/os_dep/osdep_service.c b/os_dep/osdep_service.c
index 494016c..f9f494d 100755
--- a/os_dep/osdep_service.c
+++ b/os_dep/osdep_service.c
@@ -2914,7 +2914,9 @@ u64 rtw_division64(u64 x, u64 y)
 inline u32 rtw_random32(void)
 {
 #ifdef PLATFORM_LINUX
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 1, 0))
+       return get_random_u32();
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
        return prandom_u32();
 #elif (LINUX_VERSION_CODE <= KERNEL_VERSION(2, 6, 18))
        u32 random_int;

```

, but stumbled on it
```
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:467:17: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
  467 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /root/8821cu-20210118/include/osdep_service_linux.h:98,
                 from /root/8821cu-20210118/include/osdep_service.h:54,
                 from /root/8821cu-20210118/include/drv_types.h:27,
                 from /root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8292:6: note: declared here
 8292 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c: At top level:
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10369:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, struct ke            y_params *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, struct key_params *)’} from incompatible pointer type ‘            int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, struct key_params *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const             unsigned char *, struct key_params *)’} [-Werror=incompatible-pointer-types]
10369 |         .add_key = cfg80211_rtw_add_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10369:20: note: (near initialization for ‘rtw_cfg80211_ops.add_key’)
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10370:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, void *, v            oid (*)(void *, struct key_params *))’ {aka ‘int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, stru            ct key_params *))’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, void *, void (*)(void *, struct key_params *))’ {ak            a ‘int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, struct key_params *))’} [-Werror=incompatible-pointe            r-types]
10370 |         .get_key = cfg80211_rtw_get_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10370:20: note: (near initialization for ‘rtw_cfg80211_ops.get_key’)
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10371:20: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *)’ {aka ‘in            t (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *)’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device             *, u8,  bool,  const u8 *)’ {aka ‘int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *)’} [-Werror=incompatible-pointer-types]
10371 |         .del_key = cfg80211_rtw_del_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10371:20: note: (near initialization for ‘rtw_cfg80211_ops.del_key’)
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10372:28: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  bool)’ {aka ‘int (*)(            struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  _Bool)’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, u8,  bool,  bool)’             {aka ‘int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  _Bool)’} [-Werror=incompatible-pointer-types]
10372 |         .set_default_key = cfg80211_rtw_set_default_key,
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10372:28: note: (near initialization for ‘rtw_cfg80211_ops.set_default_key’)
/root/8821cu-20210118/os_dep/linux/ioctl_cfg80211.c:10374:33: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, int,  u8)’ {aka ‘int (*)(struct wiphy *            , struct net_device *, int,  unsigned char)’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, u8)’ {aka ‘int (*)(struct wiphy *, struct net_dev            ice *, unsigned char)’} [-Werror=incompatible-pointer-types]
10374 |         .set_default_mgmt_key = cfg80211_rtw_set_default_mgmt_key,

```

morrownr, 2023-10-04:
Greetings

> Is it possible to port this driver to the latest LTS version of the kernel - 6.1.x. ?

This driver has been working on kernel 6.1 for nearly a year. What problem are you seeing?

What distro and version are you using?

apollo80, 2023-10-06:
Hi.
Indeed, that was my problem
Apologize.

closed 2023-10-06 by apollo80