morrownr/8821cu-20210916 #121: openwrt How to use the compilation method？

vpei, 2023-10-17:
openwrt How to use the compilation method？

morrownr, 2023-10-17:
Hi @vpei 

Are you asking how to use this driver with OpenWRT?

vpei, 2023-10-18:
> Hi @vpei
> 
> Are you asking how to use this driver with OpenWRT?

Is there a Makefile file？

morrownr, 2023-10-18:
> Is there a Makefile file？

There is.

If you do not have a lot of experience, trying to compile this driver for OpenWRT could be something that is very challenging. I do not support this driver on OpenWRT as OpenWRT already has numerous drivers for usb wifi adapters.

You can go to the site Main Menu:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 are a good place to start. Menu item 2 shows adapters that almost all already have drivers in OpenWRT... chipsets mt7921au, mt7612u and mt7610u to name a few.

Realtek out-of-kernel drivers and OpenWRT are a challenge.

Nick

closed 2023-10-19 by vpei