morrownr/8821cu-20210916 #122: Make fails:  auto.conf not found

jlpoolen, 2023-10-26:
On a Raspberry Pi 4B(8GB) using a Gentoo Linux image from the [GenPi64 project](https://github.com/GenPi64) [OpenRC version]
Trying to compile a module for the [Amazon $9 antenna:"AC600 Mbps Dual Band 2.4/5Ghz Wireless USB WiFi Network Adapter w/Antenna 802.11"](https://www.amazon.com/dp/B07Z31W68W?psc=1&ref=ppx_yo2ov_dt_b_product_details) using the Realtek chip "0bda:c811"

pizero2 /usr/local/src/8821cu-20210916 # **sh install-driver.sh** 
: ---------------------------
: install-driver.sh v20230830
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 4/4 (in-use/total processing units)
: 7997028 (total system memory)
: 5.15.61-v8-p4 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Gentoo 13.2.1_p20230826 p7) 13.2.1 20230826
: mokutil not installed
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.61-v8-p4/build M=/usr/local/src/8821cu-20210916  modules
Makefile:658: include/config/auto.conf: No such file or directory
make: *** [Makefile:2489: modules] Error 2
An error occurred:  2
Please report this error.
Please copy all screen output and paste it into the problem report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
pizero2 /usr/local/src/8821cu-20210916 # **git status**
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
pizero2 /usr/local/src/8821cu-20210916 # 

pizero2 /usr/local/src/8821cu-20210916 # **lsusb**
Bus 001 Device 007: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 006: ID 04d9:0006 Holtek Semiconductor, Inc. Wired Keyboard (78/79 key) [RPI Wired Keyboard 5]
Bus 002 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 002 Device 003: ID 0781:cfcb SanDisk Corp. SDDR-B531
Bus 002 Device 002: ID 05e3:0626 Genesys Logic, Inc. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pizero2 /usr/local/src/8821cu-20210916 # **rfkill list all**
0: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
pizero2 /usr/local/src/8821cu-20210916 # dkms status
bash: dkms: command not found
pizero2 /usr/local/src/8821cu-20210916 # **iw dev**
phy#0
        Interface wlan0
                ifindex 4
                wdev 0x1
                addr e4:5f:01:09:70:1b
                type managed
                channel 34 (5170 MHz), width: 20 MHz, center1: 5170 MHz
                txpower 31.00 dBm
pizero2 /usr/local/src/8821cu-20210916 # **iw reg get**
global
country GB: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

pizero2 /usr/local/src/8821cu-20210916 # **date**
Wed Oct 25 09:02:00 PM PDT 2023
pizero2 /usr/local/src/8821cu-20210916 # **uname -a**
Linux pizero2 5.15.61-v8-p4 #1 SMP PREEMPT Thu Aug 17 19:31:53 CDT 2023 aarch64 GNU/Linux
pizero2 /usr/local/src/8821cu-20210916 # 

_See_ "**make -d**" output at:  https://pastebin.com/FnGkm5WS

jlpoolen, 2023-10-26:
The configuration file was successfully staged under /etc/modeprobe.d:

```
jlpoole@pizero2 ~ $ ls /etc/modprobe.d/88* -lat
-rw-r--r-- 1 root root 5879 Oct 25 20:57 /etc/modprobe.d/8821cu.conf
jlpoole@pizero2 ~ $ 
```

jlpoolen, 2023-10-26:
Found the file where line 658 is faulting:

```
	pizero2 /usr/local/src/8821cu-20210916 # find /usr/src/linux-5.15.61 -type f -name "Makefile" |xargs grep -n 'auto.conf'
	/usr/src/linux-5.15.61/Makefile:588:# and from include/config/auto.conf.cmd to detect the compiler upgrade.
	/usr/src/linux-5.15.61/Makefile:658:include include/config/auto.conf
	/usr/src/linux-5.15.61/Makefile:713:include include/config/auto.conf.cmd
	/usr/src/linux-5.15.61/Makefile:726:# include/config/auto.conf (which mirrors .config).
	/usr/src/linux-5.15.61/Makefile:734:%/config/auto.conf %/config/auto.conf.cmd %/generated/autoconf.h: $(KCONFIG_CONFIG)
	/usr/src/linux-5.15.61/Makefile:739:# and include/config/auto.conf but do not care if they are up-to-date.
	/usr/src/linux-5.15.61/Makefile:740:# Use auto.conf to trigger the test
	/usr/src/linux-5.15.61/Makefile:741:PHONY += include/config/auto.conf
	/usr/src/linux-5.15.61/Makefile:743:include/config/auto.conf:
	/usr/src/linux-5.15.61/Makefile:1188:# (this can be evaluated only once include/config/auto.conf has been included)
	pizero2 /usr/local/src/8821cu-20210916 # 
```

so the variable "**need-config**" is triggering a search for a file that does not exist

```
	pizero2 /usr/local/src/8821cu-20210916 # head /usr/src/linux-5.15.61/Makefile -n 664 |nl|tail
	   550  export KBUILD_MODULES KBUILD_BUILTIN
	       
	   551  ifdef need-config
	   552  include include/config/auto.conf
	   553  endif
	       
	   554  ifeq ($(KBUILD_EXTMOD),)
	   555  # Objects we will link into vmlinux / subdirs we need to visit
	   556  core-y          := init/ usr/ arch/$(SRCARCH)/
	   557  drivers-y       := drivers/ sound/
	pizero2 /usr/local/src/8821cu-20210916 # ls -la /usr/src/linux-5.15.61/include/config
	ls: cannot access '/usr/src/linux-5.15.61/include/config': No such file or directory
	pizero2 /usr/local/src/8821cu-20210916 # 
```

jlpoolen, 2023-10-26:
I disengaged "need-config" to see what happens:

	pizero2 /usr/local/src/8821cu-20210916 # head -n 285 /usr/src/linux-5.15.61/Makefile|nl|tail
	   231                            image_name
	   232  single-targets := %.a %.i %.ko %.lds %.ll %.lst %.mod %.o %.s %.symtypes %/
	       
	   233  config-build    :=
	   234  mixed-build     :=
	   235  #need-config    := 1  # 20231026  rem'd for antenna module build
	   236  need-config     :=
	   237  need-compiler   := 1
	   238  may-sync-config := 1
	   239  single-build    :=
	pizero2 /usr/local/src/8821cu-20210916 # 

# result:

	pizero2 /usr/local/src/8821cu-20210916 # make
	make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.61-v8-p4/build M=/usr/local/src/8821cu-20210916  modules
	***
	*** The present kernel configuration has modules disabled.
	*** To use the module feature, please run "make menuconfig" etc.
	*** to enable CONFIG_MODULES.
	***
	make[1]: *** [Makefile:1839: modules] Error 1
	make: *** [Makefile:2489: modules] Error 2
	pizero2 /usr/local/src/8821cu-20210916 #
	
I'm looking into this matter further.  It may be that I have build a kernel (the kernel was built by an assembly process in the GenPi64 project) before an outside module can be compiled against it?  I'm posting here to simply preserve this exploratory attempt.

jlpoolen, 2023-10-26:
On further investigation, it looks like the GenPi64 project's kernel is built through a Gentoo package and is not the staged kernel usually under /usr/src.  I'm going to close this issue as the problem arises from a kernel configured for no modules built in a temporary directory during Gentoo's portage command of "emerge" and is making trying to compile this outside module against such a formidable task and not something the kind @morrownr should have to grapple with.  If I come up with some solution for other Gentoo users on the Raspberry pi, I'll re-open this ticket.

closed 2023-10-26 by jlpoolen

morrownr, 2023-10-26:
Hi @jlpoolen 

Sorry for the delay, I have been ill for the last couple of days. 

When I first looked at, the first thing came to mind was that something has not been installed. I ran into the following while searching but am not sure if it is related:

https://www.programmerhat.com/autoreconf-command-not-found/

jlpoolen, 2023-10-26:
> Hi @jlpoolen
>
> Sorry for the delay, I have been ill for the last couple of days.
>
> When I first looked at, the first thing came to mind was that something has not been installed. I ran into the [...]

Thank you.  No worries on "delay", this issue had not even been open for 24 hours.  I had been operating under the mistaken belief that the kernel was configured as it might for a standard Gentoo install under /usr/src.  But it turns out that for the Rapsberry Pi image, the kernel is a package which has several dependencies and possible patches.  For example, here is the ebuild (like a recipe) for my current image:  https://github.com/GenPi64/genpi64-overlay/blob/master/sys-kernel/raspberrypi-kernel/raspberrypi-kernel-5.15.61-r1.ebuild   Unfortunately, when packages are built, everything is erased except for the installed final product, so I do not have a fully fleshed out build tree available to hook into with an outside of tree module such as yours.  What I could do, for kicks, is rebuild the package and use the "keepwork" flag which will preserve everything it assembles, e.g. the linux kernel tree with all its configurations, and then try to reference that tree from your project.  

Other people encountering trying to compile an out-of-tree module for a Raspberry Pi should bear in mind this departure of building a kernel as compared to The Gentoo Handbook where everything is nicely stored under /usr/src and can be referenced.  I will take a look at autoconf, I but I did confirm I had autoconf installed and the problem is that certain "ac" files that should be in the Linux kernel source tree were not there simply because the standard way of compiling a kernel is not followed for Raspberry Pi.  So I don't hold out much hope -- the better avenue appears to be installing the kernel, again, and preserving the work and then referencing the work from this module's Makefile.