morrownr/8821cu-20210916 #123: Errors in dmesg and no connection

pbarolat, 2023-10-27:
Hi, 
I successfully compiled the drivers following the provided instructions.  However, I am not able to connect to my AP.  

Here is the information from my system:

lsusb; rfkill list all; dkms status; iw dev; iw reg get
Bus 001 Device 002: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: phy1: Wireless LAN
        Soft blocked: no
        Hard blocked: no
rtl8821cu/5.12.0.4, 6.5.0-1006-raspi, armv7l: installed
phy#1
        Interface wlxa0d768116efe
                ifindex 3
                wdev 0x100000001
                addr a0:d7:68:11:6e:fe
                ssid barolat-wlan-n
                type managed
                channel 40 (5200 MHz), width: 40 MHz, center1: 5190 MHz
                txpower 13.00 dBm
phy#0
        Unnamed/non-netdev interface
                wdev 0x3
                addr da:3a:dd:7a:23:39
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 2
                wdev 0x1
                addr d8:3a:dd:7a:23:39
                ssid barolat-wlan
                type managed
                channel 10 (2457 MHz), width: 20 MHz, center1: 2457 MHz
                txpower 31.00 dBm
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 160), (6, 20), (N/A)
        (5460 - 5860 @ 160), (6, 20), (N/A)

--------------------------------------------------------------------
Here are the errors from dmesg:
================================================================================
[  539.318953] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_wlan_util.c:1855:48
[  539.329834] index 1 is out of range for type 'u8 [1]'
[  539.335092] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C OE      6.5.0-1006-raspi #8-Ubuntu
[  539.335111] Hardware name: BCM2835
[  539.335121] Backtrace:
[  539.335135]  dump_backtrace from show_stack+0x20/0x24
[  539.335177]  r7:dc827000 r6:00000000 r5:c127d64c r4:60070113
[  539.335186]  show_stack from dump_stack_lvl+0x70/0x94
[... 406 lines cut ...]
[  618.323810] RTW: start auth
[  618.324778] RTW: auth success, start assoc
[  618.325940] RTW: assoc success
[  618.329133] RTW: ============ STA [08:60:6e:20:b4:ac]  ===================
[  618.329180] RTW: mac_id : 0
[  618.329197] RTW: wireless_mode : 0x14
[  618.329212] RTW: mimo_type : 0
[  618.329227] RTW: static smps : N
[  618.329242] RTW: bw_mode : 40MHz, ra_bw_mode : 40MHz
[  618.329259] RTW: rate_id : 10
[  618.329275] RTW: rssi : -1 (%), rssi_level : 0
[  618.329293] RTW: is_support_sgi : Y, is_vht_enable : N
[  618.329309] RTW: disable_ra : N, disable_pt : N
[  618.329325] RTW: is_noisy : N
[  618.329339] RTW: txrx_state : 0
[  618.329355] RTW: curr_tx_rate : CCK_1M (L)
[  618.329371] RTW: curr_tx_bw : 20MHz
[  618.329386] RTW: curr_retry_ratio : 0
[  618.329401] RTW: ra_mask : 0x00000000000ffff0
[  618.336601] RTW: recv eapol packet 1/4
[  618.338072] RTW: send eapol packet 2/4
[  618.346421] RTW: recv eapol packet 3/4
[  618.346892] RTW: send eapol packet 4/4
[  618.347332] RTW: set pairwise key camid:0, addr:08:60:6e:20:b4:ac, kid:0, type:AES
[  618.350724] RTW: set group key camid:1, addr:08:60:6e:20:b4:ac, kid:1, type:AES

morrownr, 2023-10-27:
What operating system and hardware are you using?

pbarolat, 2023-10-27:
Hi morrownr,

The hardware is Raspberry Pi Zero 2 W.

Below is the OS info.  It's the 32-bit version.  If moving to Ubuntu 22.04.3 will simplify or potentially resolve the issue more quickly, that's an easy change to do.

cat /etc/os-release
PRETTY_NAME="Ubuntu 23.10"
NAME="Ubuntu"
VERSION_ID="23.10"
VERSION="23.10 (Mantic Minotaur)"
VERSION_CODENAME=mantic
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=mantic
LOGO=ubuntu-logo

uname -a
Linux ubuntu 6.5.0-1006-raspi #8-Ubuntu SMP PREEMPT Mon Oct 23 17:40:55 UTC 2023 armv7l armv7l armv7l GNU/Linux

Koguni31, 2024-01-25:
Same errors with Ubuntu 22.04.3 and 6.5.0-14 kernel in VMware Workstation 17.5.0 build-22583795 VM, but adapter connects to network anyway.

```
[   26.769109] RTW: start auth
[   26.770287] RTW: auth success, start assoc
[   26.773331] ================================================================================
[   26.775575] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_wlan_util.c:1855:48
[   26.776058] index 1 is out of range for type 'u8 [1]'
[   26.776484] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE      6.5.0-14-generic #14~22.04.1-Ubuntu
[   26.776488] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 11/12/2020
[   26.776490] Call Trace:
[   26.776492]  <IRQ>
[   26.776495]  dump_stack_lvl+0x48/0x70
[   26.776505]  dump_stack+0x10/0x20
[   26.776508]  __ubsan_handle_out_of_bounds+0xc6/0x110
[   26.776514]  HT_caps_handler+0xaa/0x4e0 [8821cu]
[   26.776642]  OnAssocRsp+0x25d/0x3b0 [8821cu]
[   26.776734]  _mgt_dispatcher+0x70/0xf0 [8821cu]
[   26.776818]  mgt_dispatcher+0x296/0x300 [8821cu]
[   26.776903]  validate_recv_mgnt_frame+0x3a0/0x520 [8821cu]
[   26.777011]  validate_recv_frame+0x215/0x460 [8821cu]
[   26.777118]  recv_func_prehandle+0x19/0x50 [8821cu]
[   26.777242]  recv_func+0x2c/0x160 [8821cu]
[   26.777348]  rtw_recv_entry+0x19/0x40 [8821cu]
[   26.777453]  pre_recv_entry+0x6c/0x160 [8821cu]
[   26.777560]  recvbuf2recvframe+0x170/0x370 [8821cu]
[   26.777717]  usb_recv_tasklet+0xa1/0x2e0 [8821cu]
[   26.777855]  tasklet_action_common.constprop.0+0xfa/0x1e0
[   26.777861]  tasklet_action+0x22/0x30
[   26.777864]  __do_softirq+0xd9/0x349
[   26.777868]  ? handle_irq_event+0x52/0x80
[   26.777873]  __irq_exit_rcu+0x75/0xa0
[   26.777876]  irq_exit_rcu+0xe/0x20
[   26.777880]  common_interrupt+0xa4/0xb0
[   26.777883]  </IRQ>
[   26.777884]  <TASK>
[   26.777885]  asm_common_interrupt+0x27/0x40
[   26.777890] RIP: 0010:pv_native_safe_halt+0xb/0x10
[   26.777894] Code: 0b 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 66 90 0f 00 2d d9 1f 2f 00 fb f4 <c3> cc cc cc cc 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 90 55
[   26.777896] RSP: 0018:ffffffffa9c03d58 EFLAGS: 00000246
[   26.777900] RAX: 0000000000004000 RBX: ffff8b45419af064 RCX: 0000000000000000
[   26.777902] RDX: 0000000000000001 RSI: ffff8b45419af000 RDI: 0000000000000001
[   26.777903] RBP: ffffffffa9c03d60 R08: 0000000000000000 R09: 0000000000000000
[   26.777905] R10: 0000000000000000 R11: 0000000000000000 R12: ffff8b45419af064
[   26.777907] R13: 0000000000000000 R14: ffffffffa9ee52e0 R15: ffff8b4578600000
[   26.777911]  ? tick_nohz_next_event+0x61/0x140
[   26.777914]  ? acpi_safe_halt+0x19/0x60
[   26.777917]  acpi_idle_do_entry+0x40/0x80
[   26.777919]  acpi_idle_enter+0xb6/0x180
[   26.777922]  cpuidle_enter_state+0x8e/0x720
[   26.777926]  ? finish_task_switch.isra.0+0x85/0x2a0
[   26.777929]  cpuidle_enter+0x2e/0x50
[   26.777934]  call_cpuidle+0x23/0x60
[   26.777938]  cpuidle_idle_call+0x11d/0x190
[   26.777941]  do_idle+0x82/0xf0
[   26.777944]  cpu_startup_entry+0x1d/0x20
[   26.777946]  rest_init+0xde/0x100
[   26.777948]  ? acpi_enable_subsystem+0xe6/0x2a0
[   26.777952]  ? static_key_disable+0x1f/0x30
[   26.777957]  arch_call_rest_init+0xe/0x30
[   26.777960]  start_kernel+0x34f/0x440
[   26.777963]  x86_64_start_reservations+0x18/0x30
[   26.777965]  x86_64_start_kernel+0xa4/0xe0
[   26.777968]  secondary_startup_64_no_verify+0x17e/0x18b
[   26.777974]  </TASK>
[   26.777976] ================================================================================
[   26.778414] ================================================================================
[   26.778842] UBSAN: array-index-out-of-bounds in /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_wlan_util.c:1860:75
[   26.779322] index 2 is out of range for type 'u8 [1]'
[   26.779763] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE      6.5.0-14-generic #14~22.04.1-Ubuntu
[... 171 lines cut ...]
[   26.793209]  ? static_key_disable+0x1f/0x30
[   26.793222]  arch_call_rest_init+0xe/0x30
[   26.793231]  start_kernel+0x34f/0x440
[   26.793239]  x86_64_start_reservations+0x18/0x30
[   26.793246]  x86_64_start_kernel+0xa4/0xe0
[   26.793253]  secondary_startup_64_no_verify+0x17e/0x18b
[   26.793271]  </TASK>
[   26.793370] ================================================================================
[   26.795061] RTW: assoc success
[   26.796545] RTW: recv eapol packet 1/4

```

```
root@mrouter03:/home/ubuntu/8821cu-20210916# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

root@mrouter03:/home/ubuntu/8821cu-20210916# uname -a
Linux mrouter03 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

```

closed 2024-02-20 by morrownr

anatolyakulinski, 2024-03-14:
Hello @morrownr 
I'm facing the same issue and my adapter refuses to connect to my APs at all.
I'm on Linux 6.8.0

here's the dmesg and journal 
[dmesg.txt](https://github.com/morrownr/8821cu-20210916/files/14600924/dmesg.txt)
[journal.txt](https://github.com/morrownr/8821cu-20210916/files/14600925/journal.txt)

and thank you for your great work!

neliavaal, 2024-03-14:
Hello @morrownr
I'm facing the same issue and my adapter refuses to connect to my APs at all.
I'm on Linux 6.8.0

here's the dmesg and journal
[dmesg.txt](https://github.com/morrownr/8821cu-20210916/files/14600924/dmesg.txt)
[journal.txt](https://github.com/morrownr/8821cu-20210916/files/14600925/journal.txt)

and thank you for your great work!

morrownr, 2024-03-14:
Hi @neliavaal 

Thanks for the report. Ouch! I am going to have to configure a test system to work this issue. 

@morrownr

alkisg, 2024-03-14:
No 0bda:c811 or 8812cu in these logs...
I only see 0bda:5411, VirtualBox and rtl8812au errors.

neliavaal, 2024-03-18:
@alkisg hello and thanks for responding. 
I'm providing more logs and as you can see my system remains 'inactive' for 2 minutes during boot only _with_ the dongle plugged in:
```[    5.590071] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[    5.591642] Console: switching to colour frame buffer device 215x45
[    5.636382] EXT4-fs (nvme0n1p2): mounted filesystem 4137994d-2045-4f15-ba35-46bab48f3706 r/w with ordered data mode. Quota mode: none.
[    7.736240] ucsi_ccg 0-0008: failed to reset PPM!
[    7.736246] ucsi_ccg 0-0008: error -ETIMEDOUT: PPM init failed
[   36.202520] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   66.910456] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[   97.630088] usb 1-1: reset high-speed USB device number 2 using xhci_hcd
[  126.778255] systemd[1]: Inserted module 'autofs4'
[  126.794734] systemd[1]: systemd 249.11-0ubuntu3.12 running in system mode (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT +GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY -P11KIT -QRENCODE +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[  126.794831] systemd[1]: Detected architecture x86-64.
[  126.795218] systemd[1]: Hostname set to <eb2ad5c8>.
```

the full log is here:
[dmesg2.log](https://github.com/morrownr/8821cu-20210916/files/14640936/dmesg2.log)

then it boots - the device is in CDROM mode.
re-plugging the device switches it to the normal operation mode but it is still unable to connect to my AP no matter open or password protected one:
```
Mar 18 19:31:14 <host> kernel: wlxa047d77025ea: send auth to <mac>:4f (try 1/3)
Mar 18 19:31:15 <host> kernel: wlxa047d77025ea: send auth to <mac>:4f (try 2/3)
Mar 18 19:31:15 <host> kernel: wlxa047d77025ea: send auth to <mac>:4f (try 3/3)
Mar 18 19:31:16 <host> rtkit-daemon[1450]: Supervising 11 threads of 6 processes of 1 users.
Mar 18 19:31:16 <host> rtkit-daemon[1450]: Supervising 11 threads of 6 processes of 1 users.
Mar 18 19:31:17 <host> kernel: wlxa047d77025ea: authentication with <mac>:4f timed out
Mar 18 19:31:17 <host> wpa_supplicant[420856]: wlxa047d77025ea: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="<my_ap_name>" auth_failures=1 duration=10 reason=CONN_FAILED
```

[journal2.log](https://github.com/morrownr/8821cu-20210916/files/14640945/journal2.log)

pbarolat, 2024-03-18:
@morrownr

Sorry just seeing your question.  I've moved on to another Wi-Fi adapter so I haven't been using this driver.  At the time that I posted the issue, I am quite confident that the driver was *not* working.  I'm basing the previous statement on the fact that I decided to move onto a different adapter with a different driver; the reality is I can't specifically recall whether or not it was working.

Paolo

morrownr, 2024-03-18:
@pbarolat 

Thanks for the reply. Hardware that is not working is very frustrating. In case some else reads this issue, here is a link to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

There is a lot of information about USB WiFi adapters there.

Also, I merged a patch today that should fix this problem.

Also, a lot of work has been done on the support for this chipset in the rtw88 driver recently and it appears a lot of progress has been made.

dubhater, 2024-03-18:
@neliavaal You're using rtw88, not the driver from this repository. That's why it's not connecting to your AP. rtw88 will work better in kernel 6.9. In the meantime, you can blacklist rtw88_8821cu.

neliavaal, 2024-03-18:
@dubhater my system is a mess at this point but it doesn't matter. It behaves the same on a fresh install too.

dubhater, 2024-03-18:
@morrownr Sure.

@neliavaal:
```sh
modprobe -r rtw88_8821cu rtw_8821cu
insmod /path/to/8821cu.ko
```

Then try to connect again.

If your device has an LED, you can tell when you're using rtw88 or 8821cu.ko. With rtw88 the LED will remain dark all the time, whereas 8821cu.ko makes the LED blink.

morrownr, 2024-03-18:
@neliavaal 

> my system is a mess at this point but it doesn't matter. It behaves the same on a fresh install too.

I just reread your posts. I see a some things that raise more questions. 

- Are you running the distro you are using in a VM? 
- How many wifi drivers have you installed?

Your adapter is a multi-state adapter. There are reasons why I recommend Linux users stay away from multi-state adapters. If you are unaware of the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 provide a lot of good information. Menu item 2, the Plug and Play List provides a good list of adapters to consider and NONE of them are multi-state.

I am currently testing the Alfa AWUS036EACS (rtl8822cu chipset) and the Brostrend AC5L (rtl8811cu chipset) and both work really well with the driver in this repo. The Alfa adapter supports bluetooth and it works really well. Both adapters are single-state. rtw88 has problems in current distros but as @dubhater said, with kernel 6.9, things should be much better. They are much better in my testing here.

I merged a couple of patches last night. You can update and try the driver in this repo again by doing the following:

$ git pull
$ sudo sh install-driver.sh

The install script writes a log to screen as it is installing the driver. Please copy that log and post it here.

@morrownr