morrownr/8821cu-20210916 #124: rk3588s,ubuntu2204,linux5.10,gcc11.4,install failed

mapkkkk, 2023-10-27:
I installed the driver on my rockpi 5a arm64 board,it seems all right at the beginning,however,when i successfully installed the driver and reboot,i did not work.I run the 'nmtui' order,and it did not recognize the 8811CU wifi adapter with an empty wlan list.Howerver,when i 'lsusb',the adapter is connected.

mapkkkk, 2023-10-27:
add terminal:
root@rock-5a:~/drivers# cd 8821cu-20210916
root@rock-5a:~/drivers/8821cu-20210916# ls
8821cu.conf         dkms-make.sh     hal                Kconfig   platform          save-log.sh
core                docs             halmac.mk          LICENSE   README.md         supported-device-IDs
default-editor.txt  edit-options.sh  include            Makefile  remove-driver.sh
dkms.conf           FAQ.md           install-driver.sh  os_dep    rtl8821c.mk
root@rock-5a:~/drivers/8821cu-20210916# ./install-driver.sh
: ---------------------------
: install-driver.sh v20230830
: aarch64 (kernel architecture)
: arm64 (architecture to send to gcc)
: 8/8 (in-use/total processing units)
: 8136072 (total system memory)
: 5.10.110-12-rockchip (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
: dkms-2.8.7
: mokutil not installed
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source -> /usr/src/rtl8821cu-5.12.0.4
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh...................
cleaning build area...
The driver was built by dkms successfully.
: ---------------------------

8821cu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.110-12-rockchip/updates/dkms/

depmod....
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 2 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!

Do you want to edit the driver options file now? (recommended) [Y/n]

morrownr, 2023-10-27:
Hi @mapkkkk 

> ... install failed

Given that 'install-driver.sh' shows a successful installation, it is highly probably the driver was installed successfully. That script has a lot of checking and report that handles many issues.

> 'lsusb',the adapter is connected.

Running 'lsusb' does not mean the adapter is connected. If the adapter shows with 'lsuab", all that means is that the system is recognizing a usb adapter and it will do this whether a driver is install or not.

My thoughts: the driver is likely installed and what we need to figure out is why the driver is not being used.

I have a Rock4se. If you can point me to the distro you are using, if there is a version for the 4se, I will try to duplicate the problem as I am able. I am having health problems so this may not happen quickly.

Let's look at a few things: (run and post the results)

$ dkms status

# iw dev

mapkkkk, 2023-10-27:
root@rock-5a:~# dkms status
8852be/1.15.10.0.5.0-3, 5.10.110-12-rockchip, aarch64: installed
8852bu/1.15.7.112-2, 5.10.110-12-rockchip, aarch64: installed
r8125/9.007.01, 5.10.110-12-rockchip, aarch64: installed
rtk_btusb/20220330-1, 5.10.110-12-rockchip, aarch64: installed
rtl8821cu/5.12.0.4, 5.10.110-12-rockchip, aarch64: installed
root@rock-5a:~# iw dev
root@rock-5a:~# 
it seems it was installed,but dev is not loaded

mapkkkk, 2023-10-28:
root@rock-5a:~# iw dev
phy#0
	Interface wlx10634b013ea5
		ifindex 3
		wdev 0x1
		addr 10:63:4b:01:3e:a5
		type managed
		txpower -100.00 dBm
root@rock-5a:~# 
i changed the usb port,and then it recognize the 8811CU,but when i want to activate a connection by 'nmtui',it show nothing 
[image]

morrownr, 2023-10-28:
Okay, what I am seeing is consistent with either this being the wrong driver or the vip/pid is not in the driver. Run

$ lsusb

... and post the line for your adapter.

Once we are sure this is the correct driver and the vip/pid is in the driver, we can explore a couple of other options.

mapkkkk, 2023-10-28:
[image]

morrownr, 2023-10-28:
This driver should work with your adapter give that vip/pid.

I have seen problems with this chipset which caused me to note the following in the 8821co.conf file. You will need to add the parameter/option and experiment with settings.

```

# Select RFE type ( rtw_RFE_type )
#
# 0 = (2-Ant, DPDT), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 1 = (1-Ant, SPDT@Ant1), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 2 = (1-Ant, SPDT@Ant1) , (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 3 = (1-Ant, DPDT@Ant2), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 4 = (1-Ant, DPDT@Ant2), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 5 = (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 6 = (2-Ant), (2G_WLG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW)
# 7 = (1-Ant), (2G_BTG, iPA, iLNA, iSW), (5G, iPA, iLNA, iSW) (try this setting first)
# 64 = this appears to be the default on adapters that do not support bluetooth
#
# Note: RFE Type is used to set antenna isolation and the BT coexistence
# mechanism. Some adapters require this setting and some do not. It may be
# necessary to try different settings to determine which setting is optimal
# for your adapter.
#

```

Also, if you installed a previous driver, there could be problems/ What does the following day?

$ dkms status

Lastly, it could be that something is wrong in the distro.

If you can't get that adapter working, you might have better luck at the site Main Menu:

https://github.com/morrownr/USB-WiFi

Read menu items 1 and 2.

mapkkkk, 2023-10-28:
Thank you for your advice!
I finally solve the problem by update linux kernel to 6.2.
Every method does not work,but fortunatelly,new kernel support it and do not need a driver.
Thanks a lot!

mapkkkk, 2023-10-28:
[image]
oh,i find that it maybe a bug by nmtui,your driver work well
nmtui can only scan my wifi(connected by other ways)
your driver should work well at first,by nmtui has problem instead
finally,nice repo!

morrownr, 2023-10-28:
> Thank you for your advice!

You are welcome. Glad you got it going.

closed 2023-11-03 by mapkkkk