morrownr/8821cu-20210916 #125: Positive Single Level

ssmoss, 2023-10-29:
I have complied and installed the most recent version of this driver and installed it on a Pi 3A+. I am using the [EDUP AC600M USB WiFi Adapter](https://www.amazon.com/gp/product/B01CCMUN8C). Currently both the internal wireless `wlan0` and the usb wireless `wlan1` are connected to the same network. The usb wireless `wlan1` has an external antenna connected. Below is the output of `iwconfig` and `/proc/net/wireless`. As shown below the signal level for `wlan0` is pretty bad, which is to be expected but the signal level for `wlan1` is positive. 

```
$ lsusb
Bus 001 Device 002: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
```

```
$ iwconfig 
lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:"xxx"  
          Mode:Managed  Frequency:2.4xx GHz  Access Point: xx:xx:xx:xx:xx:xx 
          Bit Rate=24 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=41/70  Signal level=-69 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:23  Invalid misc:0   Missed beacon:0

wlan1     IEEE 802.11bgn  ESSID:"xxx"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.4xx GHz  Access Point: xx:xx:xx:xx:xx:xx   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=71/100  Signal level=35/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

```
cat /proc/net/wireless 
Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   40.  -70.  -256        0      0      0     23      0        0
 wlan1: 0000   76.   36.    0.       0      0      0      0      0        0
```

This seems a bit strange as I thought that value should always be negative. My network controller reports a strength of `-71` for `wlan0` and `-65` for `wlan1`.

morrownr, 2023-10-30:
Hi @ssmoss 

> This seems a bit strange as I thought that value should always be negative.

It should be negative as far as I know.

> My network controller reports a strength of -71 for wlan0 and -65 for wlan1.

That is more in line with what should be expected, however, both of the numbers are not that good.

I learned over time that there are a lot of things in these Realtek out-of-kernel drivers that don't make sense. If the driver is working, I would ignore it.

Before I go, since you use usb wifi adapters, you might be interested in the Main Menu:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 can be very informative.

Cheers

@morrownr

ssmoss, 2023-10-30:
Thanks for the quick reply. As far as I can tell the driver is working correctly and my connection seems solid. The previous adaptors I had tried with "in-kernel-support" would disconnect every few hours and print messages about not support VHT. I will keep monitoring but I think everything is fine.

I had opted for the one I had since it since it is about $12 compared to the other two being ~ $25 - $30.

Thank you so much for making this repo and helping those us get functioning USB wireless adaptors. I wish your hardworking could make it into the kernel.

closed 2024-02-20 by morrownr