morrownr/8821cu-20210916 #126: Support for OWE

ARiber, 2023-11-01:
Hi @morrownr 

do you have any plans to also add OWE support? Not sure if this is mainly depending on the driver or also on the linux kernel or something else?

By the way, is there any other way to contact you besides creating an issue? 

Cheers
Aaron

morrownr, 2023-11-02:
> do you have any plans to also add OWE support?

What is that?

> By the way, is there any other way to contact you besides creating an issue?

I can get contact information to you if you want.

Cheers,

@morrownr

ARiber, 2023-11-02:
Haha, i had the same thought when i hear OWE the first time, it is like an open network but with encryption, seems it was part of WPA3 standard

https://www.wi-fi.org/beacon/dan-harkins/wi-fi-certified-enhanced-open-transparent-wi-fi-protections-without-complexity

or

https://en.wikipedia.org/wiki/Opportunistic_Wireless_Encryption

As for the contact, you can maybe sent me to a.riber@instar.com :)

Cheers
Aaron

morrownr, 2023-11-02:
> Haha, i had the same thought when i hear OWE the first time, it is like an open network but with encryption, seems it was part of WPA3 standard

Have you checked to see if it works?

ARiber, 2023-11-03:
So far i did not get it to work.

I get a 

<3>CTRL-EVENT-ASSOC-REJECT status_code=40

key_mgmt i have set to IEEE8021X

Somehow according to the wpa_supplicant documentation i should be able to set key_mgmt to OWE, but that fails setting it.

morrownr, 2023-11-03:
What version of wpa_supplicant are you running?

$ wpa_supplicant -v

ARiber, 2023-11-08:
i tried it with 2.10 stable and also with the current 2.11 devel version.

Somehow both dont allow me to set 

key_mgmt to OWE

also i according to the wpa_supplicant.conf docu i should be able to set key_mgmt to WPA-EAP-SUITE-B-192 or WPA-EAP-SUITE-B, but this also does not work. Maybe you know any reason?
Does this maybe require a certain Kernel version or somthing?

morrownr, 2023-11-09:
> Does this maybe require a certain Kernel version or somthing?

I have no idea. Let me see if I can figure out how to test.

What is it you are trying to do? Managed mode, AP mode?

ARiber, 2023-11-09:
We are using this driver in our embedded devices to connect to WiFi. So for customers we want to offer the most options including the connection to OWE networks (if possible).

Currently the WiFi module is build in to an embedded device with linux kernel 4.19
It is using the latest stable wpa_supplicant 2.10 but as mentioned above also tried 2.11 (devel)

I tested it with a OpenWRT Router by selecting OWE there as the network encryption. My iPhone was able to connect without any problems, my MacBook on the other hand was not able to connect, i found later on Apples website that OWE is only supported starting with the MacBook 14" and 16" with M1. Unfortunatly I have a 13" with M1 which does not seem to be working with OWE. So i am also not a 100% sure if its just a software thing which i assume it is, but makes me wonder why Apple does not support older devices...

Let me know if I can help testing in any kind of way.

The OpenWRT Router is running as an Accesspoint and all other WiFi modes (WPA2/WPA3/WPA2 and 3 Enterprise...) are all working, just not OWE.

morrownr, 2023-11-09:
> So i am also not a 100% sure if its just a software thing which i assume it is, but makes me wonder why Apple does not support older devices...

I suspect you are correct. My experience over the last several years here helping users goes from pre-WPA3 until the present. In this case, minor modes were needed in the drivers and the in-kernel drivers had it early on. WPA3 still would not work if the various other parts of distros were not up to speed which took a little time. Once the distros were up to speed, all in-kernel drivers, all the way back to the early WiFi 4 days, worked with WPA3 find. Realtek was slow to modify their drivers and they used a different way to do support, it is what they do. Eventually 5 of the 6 Realtek out-of-kernel drivers I have up here support WPA3. rtl8814au likely never will support WPA3 as Realtek had terminated support for the driver before the need to add WPA3 support.

I am somewhat worried when I see kernel 4.19 in use as that kernel pre-dates WPA3, let alone many more modern things like OWE. The issue will be not only the kernel but the distro surrounding the kernel. I see that you know wpa_supplicant could play a part but there are other things that might play a part as well. I'd recommend nothing less than perhaps a Debian 12 base if you want to stand a chance of the more modern features working without causing you sleepless nights. The Linux wireless devs have been at this non-stop for the last 4-5 years due to the incredible amount of new stuff. WiFi 7 has been and continues to go into the kernel and I hope once it is stabilized, we can see a more normal dev pace.

I realize some sbc makers are slow to update to modern distros due to lack of manpower so you have to be careful about that because it limits what you can do.

> By the way, is there any other way to contact you besides creating an issue?

I will put it in a message below that will be deleted.

@morrownr

morrownr, 2023-11-09:
Private email: morrownr@gmail.com

closed 2024-03-26 by morrownr