morrownr/8821cu-20210916 #129: Nvidia Jetson Nano

ZGMFX20AR, 2023-11-16:
Hi, is there any way that i can make this work with  NVIDIA Jestson Nano?

OS: Ubuntu 18.04.7
Kernel: 4.9.337-tegra

I did manage to install the driver, but didn't seem to work.

morrownr, 2023-11-26:
@ZGMFX20AR 

> Hi, is there any way that i can make this work with NVIDIA Jestson Nano?

There probably is but I don't have a Jetson Nano so the testing has to be on your end.

> but didn't seem to work.

That is not much to go on.

Can you run sudo sh install-driver.sh again and post the screen output?

Also, do you have options for a more modern Linux distro? It has been a long time since I stopped testing with kernel 4.9.

mozlima, 2023-12-10:
> Hi, is there any way that i can make this work with NVIDIA Jestson Nano?
> 
> OS: Ubuntu 18.04.7 Kernel: 4.9.337-tegra
> 
> I did manage to install the driver, but didn't seem to work.

Did you try with:

https://github.com/brektrou/rtl8821CU

>  Realtek RTL8811CU/RTL8821CU USB wifi adapter driver version 5.4.1 for Linux 4.4.x up to 5.x

ZGMFX20AR, 2023-12-11:
I am sorry, It was silly mistake, mistaken with RTL8812CU.

Thank you for reply!

closed 2024-02-20 by morrownr