morrownr/8821cu-20210916 #130: No traffic for 4-5 seconds every minute

egueli, 2023-11-19:
Happening on a Raspberry Pi 2 running Raspbian 11 (bullseye).

While connected to my 5GHz network, the USB dongle sometimes loses connectivity for 4-5 seconds. It happens no more often than once a minute. Here's the output of `ping <my-router-ip> | ts`:

```
Nov 19 22:29:36 64 bytes from 192.168.0.1: icmp_seq=48008 ttl=64 time=1.15 ms
Nov 19 22:29:37 64 bytes from 192.168.0.1: icmp_seq=48009 ttl=64 time=1.09 ms
Nov 19 22:29:38 64 bytes from 192.168.0.1: icmp_seq=48010 ttl=64 time=1.83 ms
Nov 19 22:29:39 64 bytes from 192.168.0.1: icmp_seq=48011 ttl=64 time=1.18 ms
Nov 19 22:29:40 64 bytes from 192.168.0.1: icmp_seq=48012 ttl=64 time=1.28 ms
Nov 19 22:29:45 64 bytes from 192.168.0.1: icmp_seq=48013 ttl=64 time=3862 ms
Nov 19 22:29:45 64 bytes from 192.168.0.1: icmp_seq=48014 ttl=64 time=2783 ms
Nov 19 22:29:45 64 bytes from 192.168.0.1: icmp_seq=48015 ttl=64 time=1743 ms
Nov 19 22:29:45 64 bytes from 192.168.0.1: icmp_seq=48016 ttl=64 time=703 ms
Nov 19 22:29:45 64 bytes from 192.168.0.1: icmp_seq=48017 ttl=64 time=1.89 ms
Nov 19 22:29:46 64 bytes from 192.168.0.1: icmp_seq=48018 ttl=64 time=1.19 ms
Nov 19 22:29:47 64 bytes from 192.168.0.1: icmp_seq=48019 ttl=64 time=1.51 ms
Nov 19 22:29:48 64 bytes from 192.168.0.1: icmp_seq=48020 ttl=64 time=1.10 ms
Nov 19 22:29:49 64 bytes from 192.168.0.1: icmp_seq=48021 ttl=64 time=1.17 ms

[... more 1ms pings...]

Nov 19 22:30:39 64 bytes from 192.168.0.1: icmp_seq=48070 ttl=64 time=1.68 ms
Nov 19 22:30:40 64 bytes from 192.168.0.1: icmp_seq=48071 ttl=64 time=1.63 ms
Nov 19 22:30:41 64 bytes from 192.168.0.1: icmp_seq=48072 ttl=64 time=1.60 ms
Nov 19 22:30:45 64 bytes from 192.168.0.1: icmp_seq=48073 ttl=64 time=3563 ms
Nov 19 22:30:45 64 bytes from 192.168.0.1: icmp_seq=48074 ttl=64 time=2485 ms
Nov 19 22:30:45 64 bytes from 192.168.0.1: icmp_seq=48075 ttl=64 time=1445 ms
Nov 19 22:30:45 64 bytes from 192.168.0.1: icmp_seq=48076 ttl=64 time=405 ms
Nov 19 22:30:46 64 bytes from 192.168.0.1: icmp_seq=48077 ttl=64 time=1.53 ms
Nov 19 22:30:47 64 bytes from 192.168.0.1: icmp_seq=48078 ttl=64 time=1.80 ms
Nov 19 22:30:48 64 bytes from 192.168.0.1: icmp_seq=48079 ttl=64 time=2.86 ms

[... more 1ms pings...]

Nov 19 22:31:38 64 bytes from 192.168.0.1: icmp_seq=48129 ttl=64 time=1.57 ms
Nov 19 22:31:39 64 bytes from 192.168.0.1: icmp_seq=48130 ttl=64 time=1.25 ms
Nov 19 22:31:40 64 bytes from 192.168.0.1: icmp_seq=48131 ttl=64 time=1.15 ms
Nov 19 22:31:45 64 bytes from 192.168.0.1: icmp_seq=48132 ttl=64 time=4412 ms
Nov 19 22:31:45 64 bytes from 192.168.0.1: icmp_seq=48133 ttl=64 time=3374 ms
Nov 19 22:31:45 64 bytes from 192.168.0.1: icmp_seq=48134 ttl=64 time=2334 ms
Nov 19 22:31:45 64 bytes from 192.168.0.1: icmp_seq=48135 ttl=64 time=1294 ms
Nov 19 22:31:45 64 bytes from 192.168.0.1: icmp_seq=48136 ttl=64 time=254 ms
Nov 19 22:31:46 64 bytes from 192.168.0.1: icmp_seq=48137 ttl=64 time=1.08 ms
Nov 19 22:31:47 64 bytes from 192.168.0.1: icmp_seq=48138 ttl=64 time=1.17 ms
Nov 19 22:31:48 64 bytes from 192.168.0.1: icmp_seq=48139 ttl=64 time=1.27 ms
```

Note the absence of replies between seconds 40 and 44, then a burst of replies at second 45.

At the same time, the kernel emits these messages:

```
Nov 19 22:29:45 radiorobo kernel: RTW: wlan0- hw port(0) mac_addr =90:de:80:da:13:d3
Nov 19 22:30:45 radiorobo kernel: RTW: wlan0- hw port(0) mac_addr =90:de:80:da:13:d3
Nov 19 22:31:45 radiorobo kernel: RTW: wlan0- hw port(0) mac_addr =90:de:80:da:13:d3
```

It doesn't always happen every minute; sometimes it works fine for 3-5 minutes, sometimes 10.

Here's the output of `sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get`:

```
Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8821cu, 5.12.0.4, 6.1.21-v7+, armv7l: installed
rtl88x2bu, 5.8.7.1, 6.1.21-v7+, armv7l: installed
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 90:de:80:da:13:d3
		ssid <redacted>
		type managed
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 13.00 dBm
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), NO-IR
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, NO-IR
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, NO-IR
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, NO-IR
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, NO-IR
	(5735 - 5835 @ 80), (N/A, 20), (N/A), NO-IR
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

```

And here's the output of `install-driver.sh` (note that I ran it while the driver was already installed):

```
: ---------------------------
: install-driver.sh v20230830
: armv7l (kernel architecture)
: arm (architecture to send to gcc)
: 2/4 (in-use/total processing units)
: 944104 (total system memory)
: 6.1.21-v7+ (kernel version)
: gcc-8 (version of gcc used to compile the kernel)
: gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
: dkms:2.8
: mokutil not installed
: ---------------------------

Checking for previously installed drivers...
--> rtl8821cu 5.12.0.4 6.1.21-v7+

-------- Uninstall Beginning --------
Module:  rtl8821cu
Version: 5.12.0.4
Kernel:  6.1.21-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

8821cu.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.1.21-v7+/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod..........
.

DKMS: uninstall completed.

------------------------------
Deleting module version: 5.12.0.4
completely from the DKMS tree.
------------------------------
Done.
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4

Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source ->
                 /usr/src/rtl8821cu-5.12.0.4

DKMS: add completed.
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
./dkms-make.sh..........................................................................................................................
........................................................................................................................................
........................................................................................................................
cleaning build area...

DKMS: build completed.
The driver was built by dkms successfully.
: ---------------------------

8821cu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.21-v7+/updates/dkms/

depmod............

DKMS: install completed.
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 2 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!

Do you want to apply the new options by rebooting now? (recommended) [Y/n] n

```

morrownr, 2023-11-21:
Hi @egueli 

I see the issue you are talking about. However, nothing comes to mind.

Have you tried a different adapter to see if you see the same results?

Have you considered burning a clean sd card to see if it happens?

I don't see anything that tells me the problem is due to the adapter or this driver. Your Pi is a low capability setup so I'm wondering if there are other things running that could leave a lack of cpu power available to the network for short periods.

Are you using the adapter in managed mode?

Do you see the same thing if on a 2.4 GHz channel?

egueli, 2023-11-22:
Hi @morrownr ,

Thanks for your reply. It made me look into other possible causes; after some googling I stumbled upon [this suggestion](https://forums.raspberrypi.com/viewtopic.php?p=1674971#p1674971) and I could verify that the ping drops were due to a GUI panel performing a wireless scan every minute. After changing the scan interval to a really high value and restarting, I saw no more ping drops and no more weird kernel messages. So, the issue is solved for me.

One might wonder why would a wireless scan stop all traffic for 4-5 seconds, but that's another topic.

closed 2023-11-22 by egueli