morrownr/8821cu-20210916 #131: [Issue]:  Hang on  linux-6.6.5.arch1-1

mozlima, 2023-12-10:
Hi, on the latest Arch kernel `linux 6.6.5.arch1-1`, the adapter is experiencing freezing and only reverting to `linux-6.6.4.arch1-1` and `linux-headers-6.6.4.arch1-1` fix the problem

I did test:
- usb_modeswitch:  freeze
- iw dev:  freeze
- sudo dmesg: freeze
- iwd: freeze
- network manager+iwd: freeze
- network manager+wpa_supplicant: freeze

The computer only shuts down by pulling the power cable, since usb_modeswitch fail to shutdown.

The stats **on working System:**
```
Bus 001 Device 003: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC

2: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no

phy#1
	Interface wlan1
		ifindex 5
		wdev 0x100000001
		addr ca:65:58:bd:ea:80
		ssid NAME -5G
		type managed
		channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
		txpower 13.00 dBm
global
country BR: DFS-FCC
	(2400 - 2483 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 27), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS, AUTO-BW
	(5725 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57000 - 71000 @ 2160), (N/A, 40), (N/A)

```

Some logs
```
dez 10 11:47:28 router kernel: 8821cu: loading out-of-tree module taints kernel.
dez 10 11:47:28 router kernel: 8821cu: module verification failed: signature and/or required key missing - tainting kernel
dez 10 11:47:28 router kernel: RTW: module init start
dez 10 11:47:28 router kernel: RTW: rtl8821cu v5.12.0.4-1-g9241a6516.20210916_COEX20200730-5151
dez 10 11:47:28 router kernel: RTW: rtl8821cu BT-Coex version = COEX20200730-5151
dez 10 11:47:28 router kernel: RTW: module init ret=0
dez 10 11:49:40 router kernel: rtw_8821cu 1-5:1.0: read register 0x5 failed with -110
dez 10 11:49:41 router kernel: rtw_8821cu 1-5:1.0: read register 0x5 failed with -110
dez 10 11:49:42 router kernel: rtw_8821cu 1-5:1.0: read register 0x5 failed with -110
------------------------------------------------------------------------------------------
[  429.546743] 8821cu: module verification failed: signature and/or required key missing - tainting kernel
[  429.842103] RTW: module init start
[  429.842105] RTW: rtl8821cu v5.12.0.4-1-g9241a6516.20210916_COEX20200730-5151
[  429.842106] RTW: rtl8821cu BT-Coex version = COEX20200730-5151
[  429.842157] RTW: [HALMAC]55772M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 6
               HALMAC_PATCH_VER = 22
[  429.877981] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  436.046251] RTW: HW EFUSE
[  436.046259] RTW: 0x000: 29 81 00 BC  09 00 31 00  6F 04 A4 10  12 00 30 0B  
[  436.046276] RTW: 0x010: FF FF FF FF  FF FF FF FF  FF FF FF 02  FF FF FF FF  
[  436.046290] RTW: 0x020: FF FF 2A 2A  2C 29 2B 2A  29 27 28 29  29 29 2C 2E  
[  436.046304] RTW: 0x030: 24 FF FF FF  FF FF 00 FF  FF FF 33 34  34 34 34 34  
[  436.046318] RTW: 0x040: 36 36 36 36  36 22 FF FF  FF FF FF FF  FF FF FF FF  
[  436.046332] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[... 99 lines cut ...]
[  530.716306] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  530.717055] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  530.717805] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[  530.718560] RTW: [RF_PATH] ver_id.RF_TYPE:RF_1T1R
[  530.718563] RTW: [RF_PATH] HALSPEC's rf_reg_trx_path_bmp:0x11, rf_reg_path_avail_num:1, max_tx_cnt:1
[  530.718566] RTW: [RF_PATH] PG's trx_path_bmp:0x00, max_tx_cnt:0
[  530.718569] RTW: [RF_PATH] Registry's trx_path_bmp:0x00, tx_path_lmt:0, rx_path_lmt:0
[  530.718571] RTW: [RF_PATH] HALDATA's trx_path_bmp:0x11, max_tx_cnt:1
[  530.718573] RTW: [RF_PATH] HALDATA's rf_type:RF_1T1R, NumTotalRFPath:1
[  530.718575] RTW: [TRX_Nss] HALSPEC - tx_nss:1, rx_nss:1
[  530.718577] RTW: [TRX_Nss] Registry - tx_nss:0, rx_nss:0
[  530.718578] RTW: [TRX_Nss] HALDATA - tx_nss:1, rx_nss:1
[  530.718581] RTW: txpath=0x1, rxpath=0x1
[  530.718583] RTW: txpath_1ss:0x1, num:1
[  530.719208] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[  530.853252] RTW: rtw_ndev_init(wlan1) if1 mac_addr=a0:d7:68:40:4e:10
[  538.344613] RTW: wlan1- hw port(0) mac_addr =a0:d7:68:40:4e:10
[  543.849241] RTW: wlan1- hw port(0) mac_addr =a0:d7:68:40:4e:10
[  550.127746] RTW: wlan1- hw port(0) mac_addr =a0:d7:68:40:4e:10
[  564.572384] RTW: wlan1- hw port(0) mac_addr =a0:d7:68:40:4e:10
[  575.379926] RTW: ERROR Free disconnecting network of scanned_queue failed due to pwlan == NULL

```

Thanks!

morrownr, 2023-12-10:
Hi @mozlima 

> the adapter is experiencing freezing and only reverting to linux-6.6.4.arch1-1 and linux-headers-6.6.4.arch1-1 fix the problem

Over the last few days, I have noticed an abnormal number of bug reports from arch users and all seem to be related to kernel 6.6.5. It is not yet clear to me what may be the problem but this very well could be an arch issue as I am not seeing this on the distros I currently have up for testing here. You might want to post on an arch forum to see if you can find additional info. If you get more info, please let me know.

ylxdzsw, 2023-12-11:
I'm experiencing the same issue. Not sure if it is the exact same problem but [this thread](https://bbs.archlinux.org/viewtopic.php?id=290976) seems to suggest that it is an Arch kernel issue that affects more than this driver.

alkisg, 2023-12-11:
Might be related:

https://www.phoronix.com/news/Linux-6.6.6-Released

mozlima, 2023-12-11:
@morrownr Thanks!

Ill close this.
The bug is already tracked here:

https://bbs.archlinux.org/viewtopic.php?id=290976
https://www.phoronix.com/news/Linux-6.6.6-Released

closed 2023-12-11 by mozlima