morrownr/8821cu-20210916 #132: Very poor performance

DominikM60, 2023-12-10:
Hello,

I don't know if that is something you are able to do anything about but I think it is better to report it.

I am using Nobara and installed this driver for DWA-171. Unfortunately the speed of the connection is very unsatysfying. 

Today I set up Windows on another disk for the test.
While downloading a game on Steam on Linux my peak speed is a little bit less than 30 Mbps while I am lucky and usually drops below 20.
On Windows 10 peak speed is 130Mbps and usually the stable one is around 100 Mbps.

I appreciate your work because without it I would not be able to use this device at all.

mozlima, 2023-12-11:
100Mbps on USB 2.0 ?

morrownr, 2023-12-11:
@DominikM60 

I simply do not have the time to help right now due to some high priority issues. If you are still having problems is 5-6 days, ping me.

DominikM60, 2023-12-18:
It is still the same and I don't remember this working as expected ever on this setup. I can try to setup a different distribution or use other computer

morrownr, 2023-12-18:
I should have some time this week to take a look.

However, first, some questions:

What is Nobara?
What kernel is it using? Use $ uname -r
Have you tried using the in-kernel driver?
What is the vid/pid of your adapter. Use $ lsusb

DominikM60, 2023-12-23:
Nobara is a fork of Fedora made by GloriousEggroll.

However it is not important since I temporarily use a different computer with Linux Mint installed and the issue is exactly the same. It does not seem possible to go beyond 33 Mbps while my network is capable of 90-100 on Windows 10. 
When it comes to in-kernel driver I haven't heard there is one but the device works only with your driver. Without installing it I can see it in the output of lsusb and that's about it.
Below is the info you requested for the machine with Mint.

5.15.0-91-generic
Bus 007 Device 002: ID 2001:331d D-Link Corp. DWA-171

morrownr, 2023-12-23:
> 5.15.0-91-generic

You would need at least kernel 6.2 to use the in-kernel driver.

> Bus 007 Device 002: ID 2001:331d D-Link Corp. DWA-171

That vid/pid does work with this driver.

What link quality and signal level are you seeing when you run `wavemon`?

DominikM60, 2023-12-24:
79% (55/70) link quality. -55dBm

DominikM60, 2024-01-11:
Hello, just wanted to let you know that I am using kernel 6.7.0-200.fsync.fc39.x86 on Nobara 39 and the built in driver does not work at all

morrownr, 2024-02-20:
Sorry about the delay, I have been having some health problems so am operating slowly.

> I am using kernel 6.7.0-200.fsync.fc39.x86 on Nobara 39 and the built in driver does not work at all

The vip/pid of your adapter is not in the driver in the kernel yet. I am working to correct the issue. For now, let's work with this driver.

Go to the driver directory and run the following:

sudo ./edit-options.sh

Post the `options` line so I can see how it is configured.

Also, you signal strength is not very good. Run the following while connected:

$ iw dev

And post the results.

@DominikM60

closed 2024-03-26 by morrownr