morrownr/8821cu-20210916 #133: not working on 6.6.5-arch1-1

erroor234, 2023-12-12:
i cannot access shell for some reason (i dont know even why) so i managed to copy journalctl log 
[text1.txt](https://github.com/morrownr/8821cu-20210916/files/13647517/text1.txt) <- this is log file 
i am using official arch linux and its kernel. Also dont work on lts arch kernel (and CachyOS kernels and Zen kernel too)
I also tried rtl8821cu-morrownr-dkms-git, but it not working 
Also doesnt work in arch live iso

erroor234, 2023-12-12:
i forgot 
Secure Boot disabled

erroor234, 2023-12-12:
secure boot disabled

closed 2023-12-12 by erroor234

morrownr, 2023-12-12:
Hi @erroor234 

The problem is a Mainline kernel bug. Back up to the last known good kernel and then use it until Arch issues a new kernel.

erroor234, 2023-12-12:
I am using lastest kernel from core-testing and it is working. Also checked on linux-zen and linux-lts from core-testing 
but I did the following:
Disabled NetworkManager (`sudo systemctl disable NetworkManager`) 
Rebooted 
Enabled NetworkManager (`sudo systemctl enable NetworkManager`)
Rebooted 
Not it works

morrownr, 2023-12-12:
Copy all. Any releases based on mainline 6.6.6 or later should be good.