morrownr/8821cu-20210916 #136: Variable behaves unexpectedly in dkms.conf

RadxaYuntian, 2023-12-19:
We have packaged this repo into a dkms deb package to use in our custom system.

Recently, one of our engineers found that the generated image is missing the Wi-Fi driver even though the package is installed. Reinstalling the package fixed this issue.

When we check the build log, indeed, this package failed to build:

```
2023/12/19 07:33:50 apt | Setting up linux-image-6.1.68-1-stable (6.1.68-1-ba6f5fb46) ...
2023/12/19 07:33:54 apt | dkms: running auto installation service for kernel 6.1.68-1-stable:
2023/12/19 07:33:54 apt | Kernel preparation unnecessary for this kernel.  Skipping...
2023/12/19 07:33:54 apt | 
2023/12/19 07:33:54 apt | Building module:
2023/12/19 07:33:55 apt | cleaning build area...
2023/12/19 07:33:55 apt | ./dkms-make.sh...(bad exit status: 2)
2023/12/19 07:33:55 apt | Error! Bad return status for module build on kernel: 6.1.68-1-stable (aarch64)
2023/12/19 07:33:55 apt | Consult /var/lib/dkms/8821cu/5.12.0.4+git20231215.eb52ece-1/build/make.log for more information.
```

The dkms build log as follows:

```
root@mnt:~# cat /var/lib/dkms/8821cu/5.12.0.4+git20231215.eb52ece-1/build/make.log
DKMS make.log for 8821cu-5.12.0.4+git20231215.eb52ece-1 for kernel 6.1.68-1-stable (aarch64)
Tue Dec 19 07:33:55 UTC 2023
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/6.1.0-13-amd64/build M=/var/lib/dkms/8821cu/5.12.0.4+git20231215.eb52ece-1/build  modules
make[1]: *** /lib/modules/6.1.0-13-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:2495: modules] Error 2
```

From here we see a mismatch of requested kernel version and built kernel version.

Once we pass [this variable explicitly](https://github.com/radxa-pkg/8821cu/commit/3f2f3980b2b4a088ceff9c52fc98e3d4cfd41524) it starts working.

My guess is that dkms will do variable substitution itself, instead of setting it as an environmental variable. Since most users install it on a running system, the default fallback with `uname` sliently fixs this issue.

RadxaYuntian, 2023-12-19:
Above hypothesis could be tested with `dkms build 8821cu/5.12.0.4+git20231215.eb52ece-1 -k xxx` with a kernel version different than the running one on a running system. It should always build for the running system.

Package is available from [here](https://github.com/radxa-pkg/8821cu/releases/tag/5.12.0.4%2Bgit20231016.f227602-1).

morrownr, 2023-12-20:
@RadxaYuntian 

I appreciate the report. I am going to see if I can figure out what causing this issue. There is another person that packages this driver as a deb and he has not reported any problems.

By the way, when are we going to see Debian 12 available for the Rock 4 SE?

@morrownr

alkisg, 2023-12-20:
Nick the dkms.conf we're using is:

```
PACKAGE_NAME="rtl8821cu"
PACKAGE_VERSION="#MODULE_VERSION#"
BUILT_MODULE_NAME[0]="8821cu"
MAKE="arch='$arch' kernelver='$kernelver' ./driverctl make all"
CLEAN="make clean"
DEST_MODULE_LOCATION[0]="/updates/dkms"
AUTOINSTALL="YES"
```

I.e. I agree with RadxaYuntian on this.

morrownr, 2023-12-20:
Alkis,

Thanks for jumping in. Let me study this. I have to be careful. I have tried to keep my settings and the scripts such that they work with the various standards but this gets complicated so if a change to dkms.conf would work better for all, I am all for it.

@morrownr

alkisg, 2023-12-20:
I was replying using my phone; here's a more detailed analysis.
DKMS doesn't export `$kernelver` into the environment, that's why it's not inherited to the MAKE process. It does have a `$kernelver` variable set, so you can see it with e.g. `MAKE="set >/tmp/set.txt"`, but it's not exported, so you can't see it with e.g. `MAKE="env >/tmp/env.txt"`.
So that `MAKE="var1=value var2=value command"` syntax exports the variables to the environment, so that they're accessible by `command`.

morrownr, 2023-12-20:
Alkis,

Thanks for the further explanation. 

After looking things over, it appears to me that a single change should correct the situation:

```
PACKAGE_NAME="rtl8821cu"
PACKAGE_VERSION="5.12.0.4"
BUILT_MODULE_NAME[0]="8821cu"
MAKE="kernelver=$kernelver ./dkms-make.sh"
CLEAN="'make' clean"
DEST_MODULE_LOCATION[0]="/updates/dkms"
AUTOINSTALL="yes"
```

The only change I made was with the MAKE line. Please correct me if you think additional changes should be made.

@RadxaYuntian ,

I noticed that you have changed the package name. The reason I use `rtl8821cu` is for consistency as that is the name of the driver/package. The name of the module is `8821cu." Both are by long standing Realtek practice.  I try to keep things as consistent as possible and I do my best to keep things where things work in a many situations as possible as there are many users that do manual compilation and installation and there are users that cross-compile but that is a manual thing and should not be a problem with a dkms installation.

I'll test and continue to look things over locally for a few days and if I don't see any issues, I will merge the change and let you know.

Cheers,

@morrownr

RadxaYuntian, 2023-12-21:
> By the way, when are we going to see Debian 12 available for the Rock 4 SE?

We should have access of Rockchip Linux 6.1 SDK soon, which supports Bookworm. But the upgrade is another matter since it takes a lot of resources to validate on all existing devices.

Use Armbian for now if you want Debian 12.

> I noticed that you have changed the package name. The reason I use `rtl8821cu` is for consistency as that is the name of the driver/package.

When I started packaging various network drivers for our system, I used r8125-dkms from Debian as a reference (which was already used on ROCK 5B), and over there they [used the module name as the package name](https://github.com/radxa-pkg/r8125-dkms/blob/debian/sid/debian/r8125-dkms.dkms).

Might be a coincidence for that driver only, but that's why we are where we are today /shrug

morrownr, 2023-12-22:
I have merged the change to the MAKE line in dkms.conf as noted above. The new MAKE line is as follows:

MAKE="kernelver=$kernelver ./dkms-make.sh"

Interestingly, the issue is a result of me missing the use case I see here so I wasn't testing for it. I had assumed that if dkms was to be used, it would be used with my script `install-driver.sh`.

> We should have access of Rockchip Linux 6.1 SDK soon, which supports Bookworm. But the upgrade is another matter since it takes a lot of resources to validate on all existing devices.

When it comes to distros in general and specifically Debian, so many changes in the world of wifi took place in the 3-4 years prior to Debian 12 being released that using wifi for modern things like WPA3, WiFi 6 and much more is a plain in the ass. With Debian 12, many things are stabilized. When you get to kernel 6.6, you won't ever need this driver anymore.

If you need someone to do testing on a Rock 4 SE, let me know as my Rock 4 SE will gather dust until a distro based on Debian 12 is available. I might try Armbian again but Armiban does some things I don't care for. My email address is morrownr@gmail.com in case you want to add me to a list of testers.

> When I started packaging various network drivers for our system, I used r8125-dkms from Debian as a reference (which was already used on ROCK 5B), and over there they used the module name as the package name.

I understand the non-standardization that is out there. One of the biggest problems I have had to try to overcome in the out-of-kernel drivers that I have up here is users that install drivers from other repos and then they don't remove them prior to trying to install this driver. To help with this, my `install-driver.sh` script has detection for previously installed drivers for the same chipset. This detection and removal could be easy if the driver/package name was used as it is declared in the driver but this is not the case. I have to have specific sections in the script for Armbian and PClinuxOS and some creative code in the main dkms removal section so changing the name from the name that Realtek gave the driver is not without pain.

In the end, we would not have this problem if Realtek was following Linux Wireless Standards like Mediatek does. There is good news in that the Linux kernel devs have merged code into the kernel that will stop Realtek dead in their tracks if they continue to try to use drivers for WiFi 7 based on the same idea as what we see now. Mediatek already has it new mt7925 WiFi 7 driver for USB and PCIe in the Linux kernel as of kernel 6.7. I have no idea what Realtek will do but they are getting further and further behind Mediatek.

Cheers,

@morrownr

RadxaYuntian, 2023-12-25:
Haha I saw the beef with Armbian. Although I now realize that `DEST_MODULE_LOCATION` in `dkms.conf` for our own packages is kinda all over the place as well, so this will be something to clean up in the future as well.

I have added this issue link in my TO-DO once we have a ROCK 4SE Debian 12 build.

Closing as the suggested fix is now merged.

closed 2023-12-25 by RadxaYuntian