morrownr/8821cu-20210916 #139: Adding new Device ID

RaphielAS, 2024-01-20:
1. 0bda:1a2b
2. https://ventuz-technology.com/product/ventuz-usb-wifi-5-ac-bluetooth-5-0-2-in-1-wireless-adapter/ or https://blossomzones.com/shop/modem-networking/ventuz-wifi-5-ac-bluetooth-5-0/

RaphielAS, 2024-01-20:
somehow it works on its own, at first it couldn't because it read 1a2b (disc driver), I just went to sleep suddenly it changed to c820

can you explain this sir?
or how to do that?

morrownr, 2024-01-20:
Hi @RaphielAS 

> can you explain this sir?

Yes I can. What you have is a multi-state adapter. I recommend against purchasing multi-state adapters as they can be problematic. If you can return the adapter and get your money back, I would suggest that.

More information is available at:

https://github.com/morrownr/USB-WiFi

That link is the Main Menu for this site. Menu item 1 provides good information about buying USB WiFi adapters and Menu item 2 is a list of Plug and Play adapters that work well with Linux.

If you have additional questions, please ask.

@morrownr

RaphielAS, 2024-01-20:
> Yes I can. What you have is a multi-state adapter. I recommend against purchasing multi-state adapters as they can be problematic. If you can return the adapter and get your money back, I would suggest that.

its worked normally on windows so i didn't do a return, on other ways i have ethernet connected to router so its no problem, the problem is only controller xbox one x|s using bluetooth for wireless XD

and i have another wifi dongle so thanks anyway, your information about USB-WiFi is great information anyway

morrownr, 2024-01-20:
> its worked normally on windows

It will for now but what happens in the future when that in-adapter driver does not support a new version of Windows. Multi-state adapters have one purpose, keep Windows users satisfied for a limited amount of time. Other than that, they are problematic.

closed 2024-02-19 by morrownr