morrownr/8821cu-20210916 #141: Not working on Mint 21.3

DominikM60, 2024-02-19:
Hello,

I have tried to install both manually and via script and unfortunately my DWA-171 does not work.
I have tried plugging in the different port, rebooting etc.

Below are the needed information:
Kernel: 5.15.0-94-generic 
dominik@mint:~/src/8821cu-20210916$ sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get
[sudo] password for dominik:          
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 003: ID 0c45:8508 Microdia USB DEVICE
Bus 007 Device 002: ID 1b1c:1b3c Corsair Corsair Gaming HARPOON RGB Mouse
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 22d9:276a OPPO Electronics Corp. SDM720G-IDP _SN:DB157DA1
Bus 001 Device 008: ID 2001:331d D-Link Corp. DWA-171
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rtl8821cu/5.12.0.4, 5.15.0-94-generic, x86_64: installed
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
dominik@mint:~/src/8821cu-20210916$ sudo ./install-driver.sh 
: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 8/8 (in-use/total processing units)
: 24545636 (total system memory)
: 5.15.0-94-generic (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
: dkms-2.8.7
: SecureBoot enabled
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source -> /usr/src/rtl8821cu-5.12.0.4
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
kernelver=5.15.0-94-generic ./dkms-make.sh.................
Signing module:
 - /var/lib/dkms/rtl8821cu/5.12.0.4/5.15.0-94-generic/x86_64/module/8821cu.ko
Adding '/var/lib/shim-signed/mok/MOK.der' to shim:
cleaning build area...
Compile time: 265.54 seconds
The driver was built by dkms successfully.
: ---------------------------

8821cu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-94-generic/updates/dkms/

depmod...
The driver was installed by dkms successfully.
: ---------------------------

Info: Update this driver with the following commands as needed:

$ git pull
$ sudo sh install-driver.sh

Note: Updates to this driver SHOULD be performed before distro
      upgrades such as Ubuntu 23.10 to 24.04.
Note: Updates can be performed as often as you like. It is
      recommended to update at least every 2 months.
Note: Work on this driver, like the Linux kernel, is continuous.

Enjoy!

Do you want to edit the driver options file now? (recommended) [Y/n] n
Do you want to apply the new options by rebooting now? (recommended) [Y/n] n
dominik@mint:~/src/8821cu-20210916$ cat /proc/version
Linux version 5.15.0-94-generic (buildd@lcy02-amd64-096) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #104-Ubuntu SMP Tue Jan 9 15:25:40 UTC 2024
dominik@mint:~/src/8821cu-20210916$ gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 

I am not sure why it does not work as it used to in the past with no issue at all.

morrownr, 2024-02-19:
Hi @DominikM60 

The problem is likely related to:

: SecureBoot enabled

Exactly what triggered the problem, I don't know. My to-do list is long and I have some hot issues I am working on with other drivers right now so I do not have time to investigate. I suspect, based on your post, that if you go into your bios and turn secureboot off, the driver will work so please test that and if secureboot in not totally necessary in the short run, maybe you can leave it off for a while.

@morrownr

morrownr, 2024-02-19:
@DominikM60 

FYI: I am working with some other programmers to make sure the in-kernel driver for your adapter is fully up to speed. You can follow along in issue #115 .

Hopefully the work can make it into kernel 6.8 for Ubuntu 24.04 which will be the basis for the next release of Mint. In-kernel drivers that just work are so much better that these Realtek out-of-kernel drivers.

DominikM60, 2024-02-19:
Many thanks, this workaround works as expected!
Can't wait for the results of your work in the new kernel, definitely would like to buy a coffee once done :)

pvbouwel, 2024-06-18:
Just as an FYI and if it is noise feel free to ignore it but I was experiencing issues on Linux Mint 20.3 Una with secure boot as well. When I tried to troubleshoot it a bit I did NOT see any dmesg message so I wanted to see if anything was actually installed:
```
$ sudo dkms status
rtl8821cu, 5.12.0.4, 5.4.0-186-generic, x86_64: installed
```

Still I got suspicious about the kernel module being used as I did not have any loggings and network manager only showed my wired connection. So I verified whether I could find the kernel module:
```
$ lsmod | grep 88
nls_iso8859_1          16384  1
raid6_pq              114688  1 btrfs
```

So as expected no kernel module was loaded, I thought perhaps I get some errors when trying to load it so I tried to load it manually:
```
$ sudo modprobe 8821cu
```
But no errors appeared and from this moment on wireless starts working decently.

And as expected now I can see the module:
```
$ lsmod | grep 88
nls_iso8859_1          16384  1
8821cu               2371584  0
cfg80211              712704  1 8821cu
raid6_pq              114688  1 btrfs
```

This was a once-off manual modprobe and after reboots everything kept on working. 

@morrownr  I am not sure what is expected? Was I supposed to do the modprobe manually and did I miss that instruction somehow? If not is it beneficial to add FAQ item? Because I guess this issue isn't ideal for users to find it and likely makes sense to be closed as workaround works for DominikM60 as well.

Thanks btw for the work on this having bought a WiFi stick with supposed Linux support that actually comes with instructions to go to the sellers website with only a broken link to windows drivers made me think I'd had to return the item but luckily I found your repo which is truly a gem :bowing_man:

morrownr, 2024-06-18:
Hi @pvbouwel 

> Was I supposed to do the modprobe manually and did I miss that instruction somehow?

That should not be necessary. Once you start `install-driver.sh` , everything should be done for you. It will end by asking if you want to reboot.

Obviously that did not happen. 

> This was a once-off manual modprobe and after reboots everything kept on working.

I can't explain it.

> btw for the work on this having bought a WiFi stick with supposed Linux support 

Well, the internet being the internet, there is a lot of bad information. That is the primary reason I started my github site:

https://github.com/morrownr/USB-WiFi

That is the Main Menu for this site. You are in one of the driver repos. I recommend reading menu items 1, 2 and 3. If you decide to keep your adapter, I can help you with that or if you decide to send it back and get something else, I can help you with that. I will say that adapters with 8821/11cu chips are a little more problematic than some others but I won't bore you with the details unless you ask.

Regards,

@morrownr

pvbouwel, 2024-06-19:
So the install-driver.sh seemed to have gone through fine. In the end it asked to reboot but once rebooted nothing had happened.

I then ran `[edit-options.sh](https://github.com/morrownr/8821cu-20210916/blob/main/edit-options.sh)` and added `rtw_RFE_type=64` as my stick should be WiFi only and let it reboot once once more. And then finally I did go for the modprobe mentioned earlier. 

Oh well it's a mystery that is likely not worth solving :) .

I starred your Github site and will surely go through it in detail when I need another WiFi stick for now I'll keep on using this one as it is working well since the probe.

morrownr, 2024-06-19:
> Oh well it's a mystery that is likely not worth solving :) .

It could be. Hopefully I will be able to retire this driver within the next year. A lot of work has gone into the in-kernel rtw88 driver for this chip recently. In fact, you should find when you upgrade to Mint 22 that this adapter just works. No need to install a driver.

I can point you to the updated driver if you are interested in installing it for testing.