morrownr/8821cu-20210916 #142: Mention that a driver apperared in kernel 6.2+

mikhailnov, 2024-03-15:
Hello!

I sthink it would be a good idea to mention that this driver may not ne needed on kernels >=6.2 or that an upstream driver must be blacklisted.

https://github.com/torvalds/linux/blob/master/drivers/net/wireless/realtek/rtw88/rtw8821cu.c

morrownr, 2024-03-15:
Hello @mikhailnov 

Would you mind writing a paragraph and tell me where you think it should go?

If you look in 8821cu.conf, you will see reference to the driver and the blacklist that automatically happens when this driver is installed. I have hesitated to point out this specific driver in a way that will give the impression that is is a good driver in its current state. Five patches were very recently finished and are now going into kernel 6.9. Those five patches take this in-kernel driver from being problematic to solid and stable, at least for basic function in managed mode. I have not had time to test AP or monitor modes.

The patch in the link you posted is one of the 5 patches. I did the editing on that file. It would be appreciated if you could test with kernel 6.9 when able. Problems can be reported in issue https://github.com/morrownr/8821cu-20210916/issues/115 .

@morrownr

mikhailnov, 2024-03-18:
Hello! Does https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/log/drivers/net/wireless/realtek?id=64be03575f9e9772ebdebc7f067d533348602083 have all needed patches?

morrownr, 2024-03-18:
Yes

13 days	wifi: rtw88: 8821c: Fix false alarm count
13 days	wifi: rtw88: 8821c: Fix beacon loss and disconnect
13 days	wifi: rtw88: 8821cu: Fix connection failure
13 days	wifi: rtw88: 8821cu: Fix firmware upload fail
13 days	wifi: rtw88: Add missing VID/PIDs for 8811CU and 8821CU

closed 2024-03-26 by morrownr