morrownr/8821cu-20210916 #144: Install on Fedora 39

Hodes, 2024-03-31:
Hello, 

This might be a stupid thing, but i'm having a problem that I never had before:
```
sudo ./install-driver.sh 
sudo: unable to execute ./install-driver.sh: No such file or directory
```

I'm currently using fedora 39:  Linux fedora 6.7.10-200.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Mar 18 18:56:52 UTC 2024 x86_64 GNU/Linux

I folowed the steps to install:

```
user@fedora:~$ mkdir -p ~/src
user@fedora:~$ cd ~/src/
user@fedora:~/src$ git clone https://github.com/morrownr/8821cu-20210916.git
Cloning into '8821cu-20210916'...
remote: Enumerating objects: 1585, done.
remote: Counting objects: 100% (151/151), done.
remote: Compressing objects: 100% (55/55), done.
remote: Total 1585 (delta 111), reused 114 (delta 96), pack-reused 1434
Receiving objects: 100% (1585/1585), 13.89 MiB | 18.67 MiB/s, done.
Resolving deltas: 100% (800/800), done.
user@fedora:~/src$ cd ~/src/8821cu-20210916
user@fedora:~/src/8821cu-20210916$ sudo ./install-driver.sh
[sudo] password for user: 
sudo: unable to execute ./install-driver.sh: No such file or directory

```
File Permissions:
```
user@fedora:~/src/8821cu-20210916$ ls -al | grep install-driver.sh 
-rwxr-xr-x. 1 user user 14210 mar 31 18:08 install-driver.sh
```

Anyone have clues what might be? I'm kinda giving up on this device, I have tryed others drives [RinCat](https://github.com/RinCat)
/
[RTL88x2BU-Linux-Driver](https://github.com/RinCat/RTL88x2BU-Linux-Driver) and [kelebek333](https://github.com/kelebek333)
/
[rtl8188fu](https://github.com/kelebek333/rtl8188fu), but since I han `lshw -c Network` I saw that it is referring to rtw_8822cu driver which the module is not found on my system.
With other two drivers I han see and try to connect to wi-fi but the authentication fails and after 3 times this timeouts.
For me the last hope is this repo driver.

Much appreciate!

morrownr, 2024-03-31:
Have you tried the following?

$ sudo sh install-driver.sh

Hodes, 2024-04-01:
Yes, another error happens:
```
user@fedora:~/src/8821cu-20210916$ sudo sh install-driver.sh 
install-driver.sh: linha 2: $'\r': command not found
install-driver.sh: linha 29: $'\r': command not found
install-driver.sh: linha 32: $'\r': command not found
install-driver.sh: linha 34: $'\r': command not found
install-driver.sh: linha 38: $'\r': command not found
install-driver.sh: linha 40: $'\r': command not found
install-driver.sh: linha 70: syntax error near token `$'in\r''
'nstall-driver.sh: linha 70: `	case $1 in
```

morrownr, 2024-04-01:
Try a menual install:

$ make clean
$ make
$ sudo make install
$ sudo reboot

Hodes, 2024-04-01:
Thankyou @morrownr, now I think it installed. Is there a way to check? 
So the installation issue is solved!

However unfortunately my problem continues the device cannot connect to the Wi-Fi.
I have these modules instaled:
```
user@fedora:~$ lsmod | grep rtw
rtw88_8822cu           12288  0
rtw88_8822c           487424  1 rtw88_8822cu
rtw88_usb              28672  1 rtw88_8822cu
rtw88_core            249856  2 rtw88_8822c,rtw88_usb
mac80211             1572864  2 rtw88_core,rtw88_usb
cfg80211             1335296  2 rtw88_core,mac80211
``` 
The problem I'm having is (trace from `dmesg`):
[image]

But, maybe this device cannot work on linux, don't know. On windows it auto install the driver as it also has a PenDrive Mode on itself.

Also it may not relate to this package, so this issue can be closed.

Again thankyou very much @morrownr for your time!

morrownr, 2024-04-01:
> rtw88_8822cu           12288  0
> rtw88_8822c           487424  1 rtw88_8822cu
> rtw88_usb              28672  1 rtw88_8822cu

Okay. This explains a lot. You are installing the wrong driver. Evidently the chipset in your adapter is a rtl8822cu which is very different from the rtl8821cu. I don't have a driver up for that chipset and don't know of anyone that does.

The drivers shown above are the in-kernel support for the rtl8822cu. Several Realtek drivers are contained in the Linux kernel in rtw88. It has picked up on your adapter and the drivers are loading. I have no idea if the drivers are good but what you can do is the following:

$ iw dev

Does that show a wireless interface for your adapter?

If it does, then connect to an AP and enjoy yourself...and don't forget to get rid of any previously installed drivers:

$ sudo sh remove-driver.sh
$ sudo make uninstall

Delete the folder. Do the same for any other drivers you have tried to install.

FYI: The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 could be helpful. In fact, if the in-kernel driver is not working and you just bought it and can return it, I would return it. That is a late cycle WiFi 5 adapter that may never have great support. Not sure why Realtek released it. The Plug and Play List is designed to help users find adapters that will work well with Linux.

If you have questions, let me know.

@morrownr

Hodes, 2024-04-01:
Thankyou @morrownr, 
also this is a AC1300 generic chinese dongle, a multi function device (which is not recomended as of the main site), bought this because I needed a quick (delivery) and cheap device to add theese functions to my PC.
So indeed in the package it says driver free, which it is as of in windows it auto installs and on linux it uses the in kernel, unfortunately it doesn't authenticate.

I'm closing this issue, if needed I'll asc on main site.
Thanks again!

@Hodes

closed 2024-04-01 by Hodes

dubhater, 2024-04-01:
It looks like the same problem some RTL8821CU and RTL8822BU devices had. The fix is in the kernel. I think 6.7.12 and 6.8.3 will have it. While you wait for your distro to update, you can blacklist rtw88_8822cu, rtw88_8822c, rtw88_usb, and rtw88_core and install the latest rtw88 from here: https://github.com/lwfinger/rtw88/