morrownr/8821cu-20210916 #149: Difference between native kernel driver vs. 8821cu-20210916?  More networks found with Nick variant?

Jibun-no-Kage, 2024-06-28:
Does this make sense?  When I use the kernel driver and scan for networks, my USB WiFi dongle finds about 14 to 17 networks.  But when I use the Nick variant driver, I find often 40 or more networks?

Pi OS, latest, kernel 6.6.x...
```
\# uname -r
6.6.31+rpt-rpi-v8

\# lsmod | grep rtw88
rtw88_8821cu           12288  0
rtw88_8821c            81920  1 rtw88_8821cu
rtw88_usb              20480  1 rtw88_8821cu
rtw88_core            180224  2 rtw88_8821c,rtw88_usb
```

I am I mistaken that the 8821cu lernel driver is considered better than Nick driver?  The actual dongle I am testing with is actually 8811cu variant...

```
\# lsub
Bus 001 Device 004: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
```
Am I making some type of boneheaded mistake here?

dubhater, 2024-06-28:
It makes sense. You're not doing anything wrong. This driver configures the RX gain for maximum coverage when it's scanning. rtw88 doesn't do it.

Jibun-no-Kage, 2024-06-28:
Yup... figure such when I saw the difference.  And... I have not yet starting looking for the technical information change the in kernel driver configuration details/settings.  Just wanted to confirm that I was not missing something.  Again, thanks for the confirmation.

Am I also correct that in kernel code is going forward superseding the driver in this repo at some point?  I want to say, it is the 8812AU kernel driver is still a work in progress?  Or am I confusing drvers here, 8821CU versus 8821AU?

morrownr, 2024-06-28:
@Jibun-no-Kage 

> Am I also correct that in kernel code is going forward superseding the driver in this repo at some point?

That is the plan for this repo, 8821cu, and the 8821au repo for sure. I'll have to wait and see regarding the 8812au repo as that driver has some capabilities that may never appear in the in-kernel driver... such as DFS channel support in AP mode.

Longer term, I'd also like to be able to take the 88x2bu and 8814au repos down once 8822bu, in-kernel, is fully up to speed and 8814au is added and up to speed. This would allow me to concentrate on the primary purpose of this site, which is to make sure really good information is available to Linux USB WiFi users. And I would like to concentrate on learning mac80211 programming.

Status of each rtw88 driver (Dubhater is welcome to correct me):

rtw_8821au - really good shape. Wish we could send it upstream now.
rtw_8821cu - Already in the kernel as rtw88_8821cu. After 5 patches earlier this year, managed mode is usable. Other modes need work.
rtw_8812au - in testing.
rtw_8822bu and rtw8822cu: both are already in the kernel as rtw88_8822bu and rtw88_8822cu but USB3 support as well as support in various modes is lacking.
rtw_8814au - not started yet... as far as I know.

It appears there are some core problems with rtw88 that needs technical help from Realtek. We are seeing messages that may indicate this help is coming. It would be extremely helpful is Realtek would provide at least one dedicated dev to help and support rtw88 and rtw89. Realtek probably does not care much for me because I am rather public and outspoken about my advice to Linux users to buy adapters with Mediatek chips because their driver support is so much better than Realtek at this point. It is astounding at this point to me. Mediatek already has its WiFi 7 USB driver in the kernel and adapters are not even on the market yet... that basically in the same league with Intel and AMD... and what is Realtek doing with USB support? Shit!