morrownr/8821cu-20210916 #150: kernel >=6.8  will not  work

lab37, 2024-06-29:
Perhaps due to changes in the header file, it was not possible to compile successfully in 6.8

morrownr, 2024-06-29:
Hi @lab37 

What distro are you using?

lab37, 2024-06-30:
> Hi @lab37 
> 
> What distro are you using?
> 
> 

kali-2024.2

phnik, 2024-06-30:
check your compiler version, I am using kali 2024 2 and it still working

lab37, 2024-06-30:
> check your compiler version, I am using kali 2024 2 and it still working

```
###################################################################gcc-13
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.8.11-common/scripts/Makefile.build:248: /home/kali/rtl8821cu/os_dep/linux/usb_intf.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.11-common/Makefile:1946: /home/kali/rtl8821cu] Error 2
make[1]: *** [/usr/src/linux-headers-6.8.11-common/Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.11-amd64'
make: *** [Makefile:2495: modules] Error 2
An error occurred:  2
Please report this error.
Please copy all screen output and paste it into the problem report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
                                                                                                                                                                                                                                            
┌──(kali㉿kali)-[~/rtl8821cu]
└─$ gcc --version                                    
gcc (Debian 13.2.0-24) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

#####################################################################################GCC-14
home/kali/rtl8821cu/os_dep/linux/usb_intf.c:1084:11: warning: no previous prototype for ‘rtw_usb_primary_adapter_init’ [-Wmissing-prototypes]
 1084 | _adapter *rtw_usb_primary_adapter_init(struct dvobj_priv *dvobj,
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.8.11-common/scripts/Makefile.build:248: /home/kali/rtl8821cu/os_dep/linux/usb_intf.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.11-common/Makefile:1946: /home/kali/rtl8821cu] Error 2
make[1]: *** [/usr/src/linux-headers-6.8.11-common/Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.11-amd64'
make: *** [Makefile:2495: modules] Error 2
An error occurred:  2
Please report this error.
Please copy all screen output and paste it into the problem report.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh
                                                                                                                                       
┌──(kali㉿kali)-[~/rtl8821cu]
└─$ gcc --version
gcc (Debian 14-20240330-1) 14.0.1 20240330 (experimental) [master r14-9728-g6fc84f680d0]
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

                                                                                                                                       
┌──(kali㉿kali)-[~/rtl8821cu]
└─$ 

```
same err.
I will  try  reinstall  kali-2024.2。

phnik, 2024-06-30:
You should try this brother:
ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali7-amd64/build M=/home/kali/rtl8812au  modules
make[1]: * /lib/modules/5.16.0-kali7-amd64/build: No such file or directory.  Stop.
make: * [Makefile:2251: modules] Error 2

Make sure you:    sudo apt dist-upgrade
After make sure to: sudo reboot now
BUT also i found that sometimes this command can be needed:   sudo apt-get install build-essential linux-headers-$(uname -r)
OR:  sudo apt-get install linux-headers-$(uname -r)

I got this script from this video comment:
https://www.youtube.com/watch?v=hEXwOkyYNL0

morrownr, 2024-07-01:
@lab37 

> $ gcc --version

> gcc (Debian 14-20240330-1) 14.0.1 20240330 (experimental) [master r14-9728-g6fc84f680d0]

Did this latest version Kali come with gcc 14 or did you install it? 

I ask because I am not using or testing 14 yet.

lab37, 2024-07-01:
> ```
> ###################################################################gcc-13
> cc1: some warnings being treated as errors
> make[3]: *** [...]

gcc-13  and   14
same  error

morrownr, 2024-07-01:
I'll try to find time to test with kernel 6.8 again later today or tomorrow. 

A question I have is: Generally users running Kali are doing security analysis or something closely related. If that is what you are doing, adapters based on the 8821/11cu are not what I would recommend. You can take a look at the Plug and Play List that is on the site Main Menu at the following location:

https://github.com/morrownr/USB-WiFi

Exactly which adapter would be best depends on exactly what you are doing and what range you need so ask questions.

lab37, 2024-07-01:
> I'll try to find time to test with kernel 6.8 again later today or tomorrow.
>
> A question I have is: Generally users running Kali are doing security analysis or something closely related. If that is [...]

Thank you.
Yes, I am using Kali.
8811au/cu works  well under Kali.
I understand that many people  recommends some MediaTek network cards (in kernel).
But many usb cards of MediaTek chips are not easy to use after testing, possibly because in China, the quality of many products is not as rigorous. 7601,5572, 5370, 7610, 7612, 7921 and so on. Especially when encountering USB 3.0, there will be more problems. **This may not be a technical issue, it's just a product quality issue.**
But using a realtek network card is much more stable.

morrownr, 2024-07-01:
>  the quality of many products is not as rigorous. 7601,5572, 5370, 7610, 7612, 7921 and so on.

I have one or more usb adapters based on the 7601, 5370, 7610, 7612 and 7921 chips. I also have numerous adapters with Realtek chips. It kinda comes with running this site. My opinion is that there are poorly made adapters with Mediatek and Realtek chips. There are also very good adapters made with Mediatek and Realtek chips. The mission of this site, as you will see by looking over the information on the Main Menu is help Linux users find the good quality adapters that are available from dependable dealers and provide Linux users with a problem free experience. 

There is no way to have a problem free experience with Realtek's out-of-kernel drivers. My first bit of evidence is this thread. Realtek's usb wifi support is subpar. This specific chip, 8821/11cu is problematic on Linux and it is likely problematic on other platforms as well. It works well in some cases and in others, there are problems. I work with a small team trying to improve the rtw88 support in the Linux kernel. We pushed 5 patches earlier this year and performance is now better in managed mode. Focus right now is on 8821/11au and 8812au. The 8821/11au driver is in really good shape and will be a new in-kernel driver when it is upstreamed. Would you like to test it?

Last word for this message: monitor mode support is far better with Mediatek drivers. Realtek and Mediatek chips are both made in Taiwan. USB adapters are made in Taiwan, China and some other places. Alfa adapters are made in Taiwan. Panda adapters are made in China. Even companies with a good reputation for building good quality adapters make a bad product at times. I started this site in the hopes of helping Linux users have a better experience with USB adapters. I get paid exactly nothing to do this so there is no incentive for me to push specific products. I can't own and test all adapters so I ask users to provide input, which they regular do, and I use that information to update the information that is available.

Regards

lab37, 2024-07-02:
Yes, thank you very much. I also searched for a suitable network card with a suitable chip from your repo to use on Linux. Because I sell network cards myself, I won't mention the brand here. In general, we only focus on chips and rarely pay attention to other things. In China, each of us can easily create a network card. For example, there is a company called Jialichuang, and everyone in China has two free opportunities to make PCBs every month. There is another opportunity for free SMT. So it is easy for individuals to use ready-made schematics to create their own electronic products. Another issue caused by this is that there are various different solutions for the same chip's network card on the market. Each type of network card uses different electronic components, and many have not undergone RF testing. The most crucial thing is the price. Some network cards with the same chip sell for $5, while others only sell for $0.1.
I have encountered many problems, and some network cards require a while to be plugged in before they can be used. Some cannot find descriptors due to the probability of unqualified capacitors. Some may exhibit different issues on different computers due to the issue of USB power supply voltage. Some work on USB 2.0 and keep reporting errors after switching to USB 3.0. Most of these issues are not driver issues or chip issues. It is the quality issue of product production and design.
My job is to refer to your repo, find the supported chips, and then find suitable and available network cards from the market. I continuously test them to find the lowest priced and usable product. The testing focus for each type of network card is also different, such as finding a network card for Centos to use for internet access, finding a network card suitable for Kali to listen to, and so on. Brand network cards are indeed of better quality, but their prices are often very expensive, so we rarely use mainstream brand products.
Your work has been of great help to me, thank you very much.
This is the result of machine translation, there may be some grammar errors.

morrownr, 2024-07-03:
> Your work has been of great help to me, thank you very much.

Thank you for the kind words. 

> Brand network cards are indeed of better quality, but their prices are often very expensive, so we rarely use mainstream brand products.

Even the brand name makers kick out cards that are problematic at times. This site more or less acts as a filter to help Linux users find adapters that are more likely to make them happy. While the Realtek out-of-kernel drivers may work for embedded programmers in some situations, they are totally unsuited for regular use by most Linux users. That is why we see almost no Realtek based adapters in the Plug and Play List. If the driver support is not good, then the quality of the adapter does not matter

Realtek seriously needs to reconsider their usb wifi support and the clock is ticking. Linux users are learning, the word is spreading. I purchased a low cost M.2 card based on the new mt7925 chip recently. I really did not need it but can use it and I was curious about the mt7925 in-kernel driver because the driver has been in the kernel since the beginning of this year... well before cards were available. I installed the card and started up Ubuntu 24.04 and the card came to life and I was able to connect. In managed mode connected to 5 GHz, I have seen no problems. I'll test more as I have time. This Mediatek driver also support usb adapters but none are on the market yet which is normal for how new the chip is. Now we get to Realtek. Where is Realtek's usb wifi support? I do not know. Realtek needs to do something because the clock is ticking and the word is spreading. They can lose the usb adapter and module business for Linux unless they take action. They need to do what Mediatek has done but they are about 6 years behind Mediatek at this point.

morrownr, 2024-10-02:
Hi @creatxrgithub 

Thanks for this information. I have a good idea how to fix this issue. Please start a new issue to report this as I will lose track of it down here at the end of an old issue.