morrownr/8821cu-20210916 #151: Creating a VIF hangs

LatinSuD, 2024-07-02:
This command never returns:
`iw dev wlxa0nnnnnn interface add kismon type monitor`

Moreover, from that moment all other interface commands (ifconfig, iwconfig, iw) will hang too.

I'm using 8211cu driver.

LatinSuD, 2024-07-02:
This could be a duplicate of #63

morrownr, 2024-07-05:
Hi @LatinSuD 

Working with VIF with these Realtek out-of-kernel driver can be frustrating. I use Mediatek based adapters for this kind of situation.

Let me see if I can help with an options or two. What kernel are you using?

LatinSuD, 2024-07-09:
Debian 12 default kernel 6.1.0-21-amd64

morrownr, 2024-07-09:
> Debian 12 default kernel 6.1.0-21-amd64

Okay, good. This modern distro supports many adapters. 

I went looking to see what Kismet recommends:

https://www.kismetwireless.net/amazon-hardware/

They are recommending adapters based on the mt7921au and mt7612u chips. That comes as no surprise to me. There is nothing based on a Realtek chip mentioned. That is also not a surprise.

Here is the link to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Menu items 1, 2 and 3 should be of interest. Menu item 2 is the Plug and Play List and it shows many good adapters that should work well with Kismet. In the mt7921au section, the Edup and Fenvi adapters are lower cost options. The mt7612u also has many good adapters.

Good luck.

@morrownr