morrownr/8821cu-20210916 #152: Help with CDROM mode in RTL8188GU chipset

z0mig, 2024-07-08:
Buenos días, me gustaría solicitar ayuda para lo que mi equipo y yo creemos que es un "bug" o una "incompatibilidad" con el chipset RTL8188GU.
We need to set up a device based on Raspberry. We have used Raspberry 3, as well as Raspberry 4 and 5.

On these Raspberry devices, we install a series of programs for which we need to use USB multi-band Wifi 2.4 GHz + 5GHz + Bluetooth devices.

Imagine a Raspberry with 4 connected USBs working. Specifically with these devices: [Amazon](https://www.amazon.es/RAGE-Adaptador-Bluetooth-inal%C3%A1mbrica-Compatible/dp/B0C4MDZBBN/ref=sr_1_2?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2AYO2NLMBPRBX&dib=eyJ2IjoiMSJ9.PHz_ZdmX9Y4z43EMMEl4etWrfhyKpw3nGjRTUQLLJ7HATCQeotS4lcMvNZTvVBRiyd2-z174TiYxqvBJx38tsW88WEyuZvwRRtx342DCJBe41Tw4VIS1YRBcf_FBTJEiuSvjue-WOkRvdnKeat0N_nCHq34Zg0PZKqjszN6-j7F53VyNxjAIfl472D6lZ4J3NCc2AH7m40QtQuYCmdAETAkje4L0VPH_908g9uLjpGJrzpELtrT_Kz5JbD3YSwn_AZsPXmSrIjSDMNCT1PyVFlkTwOUqhIjMIDjZNAuBlUQ.JyrTuIxP13YdtrH3_nXw7qDIim9NbfrADZSUwLW5DaM&dib_tag=se&keywords=rtl8188gu+usb+wifi+bluetooth&qid=1720439594&sprefix=rtl8188gu+usb+wifi+bluetooth%2Caps%2C109&sr=8-2)

The problem is that every time we turn on the Raspberry with the USBs plugged in, when running lsusb we get this:
[image]

We have sometimes managed to exit this CDROM mode with the following script:

`sudo service NetworkManager restart

counter=$(lsusb | awk '/CDROM/' | wc -l)
if [ $counter -ne 0 ]; then
 for (( i=1; i<= counter; i++ )); do
     sudo usb_modeswitch -KW -v 0bda -p 1a2b &>/dev/null
 done
fi

interfaces_list=($(sudo airmon-ng | awk '/8821cu/{print $2}'))

if [ ${#interfaces_list[@]} -eq 0 ]; then
  echo "No rtl8821cu interfaces found." >&2  # Print error message to stderr.
  exit 1  
fi
`

However, we find that we have to keep plugging and unplugging the USBs to get them out of CDROM mode, and this is not feasible since a bad reboot can bring back the problem.

morrownr, 2024-07-08:
Hola @z0mig 

Yo hablo espanol (English keyboard) but English is better for this site as users from all over the world stop by. Thanks for editing.

Let me point you to some information before I address the issue. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu items 1, 2 and 3 are good information for Linux users wanting to use USB WiFi adapters. You will notice in menu item 1 that I recommend that Linux users avoid multi-state adapters. I have been helping Linux users with USB adapter problems for years. My experience is that the only way to solve multi-state issues is to buy single-state adapters. 

While usb_modeswitch works well in some cases, some of the time, there are inconsistencies in how multi-state is implemented. Over the years, I have seen similar problems to what you are having. I am not aware of a solution in this specific case... well, my recommended solution is reevaluate and buy adapters that meet your needs that are single-state.

Menu item 2 is called Plug and Play List. It contains many adapters and links that work very well under Linux due to their use of in-kernel, standards compliant drivers. Not one single adapter in the list is multi-state. If you tell me what your requirements are, I can make suggestions.

Saludos,

Nick

z0mig, 2024-07-09:
Thank you very much for your quick response Nick!!! <3

First of all, my team and I hold you in very high regard. Thank you very much for your efforts with the community.

It would be great if you could help us with our interfaces. We are looking for “cheap” all-in-one USB Wifi 2.4GHz + 5 GHz + Bluetooth interfaces. The reason for this is that our product scans those frequencies. As I told you, we have selected the USB Wifi in the last Amazon link but we have found the CDRom problem... We have developed multiple scripts to avoid this situation but without guarantees because sometimes they work and other times they don't and we must reconnect the USB to Avoid this CDRom state. This is a problem for us since the product we are designing is designed so that it does not require human intervention, or at least the minimum possible.

Could you tell us if there is a multiband USB interface that meets our expectations? We would like to avoid having to fill the device with Alpha antennas everywhere since they are very large.

morrownr, 2024-07-09:
> Thank you very much for your efforts with the community.

De nada. You are welcome.

> We are looking for “cheap” all-in-one USB Wifi 2.4GHz + 5 GHz + Bluetooth interfaces.

Would a USB WiFi dual band + separate USB nano Bluetooth adapter work?

> ...but we have found the CDRom problem... We have developed multiple scripts to avoid this situation but without guarantees because sometimes they work and other times they don't and we must reconnect the USB to Avoid this CDRom state.

You are not the first person to run into this. I have been able to dig in and help in some situations but what you are describing is not a situation that I have been able to find a reliable solution. 

> the product we are designing is designed so that it does not require human intervention...

I understand. I've been working with folks that need extreme reliability since the early 90's.

What distro are you using? Do you have flexibility?
What hardware are you using?
Does the wifi device need to be an adapter or could it be a usb wifi module... or even a M.2 card?
Are you using monitor mode?

Get back to me with some answers and I'll have a better idea what your options are.

Nick

z0mig, 2024-07-10:
> De nada. You are welcome.
>
>
> Would a USB WiFi dual band + separate USB nano Bluetooth adapter work?
>
>
> You are not the first person to run into this. I have been able to dig in and help in some [...]

Thanks for your quick response again Nick.

Ideally, the USB should have "all in one". I mean, the 2.4Ghz, 5Ghz + BT frequency bands. If this is not possible... We are open to new ideas.

We are currently using the Raspi distribution via the command line. But we can change the distro if you think this could be better for us since we are still in development phases.

As I told you, we are working with Raspberry 3-4-5. Those devices have multiple USB ports and we can also use the motherboard pins of the devices.

Muchas gracias ;)

morrownr, 2024-07-10:
> Ideally, the USB should have "all in one". I mean, the 2.4Ghz, 5Ghz + BT frequency bands. If this is not possible... We are open to new ideas.

Let me throw some ideas out:

Your options for good combined adapters that are single-state are somewhat limited. The following is the best one that I know of and I have one:

https://store.rokland.com/collections/802-11ac-wi-fi-clients-receivers/products/alfa-awus036eacs-low-profile-802-11ac-ac600-dual-band-wifi-bt-usb-adapter

It was sent to me and another dev when we started working on improving the in-kernel rtw88 support for the chip, 8821cu. With that said, I am hesitant to recommend adapters with the 8821/11cu chips as they have a history of being problematic. When we were working on the rtw8821cu driver, there were issues we never did get to the bottom of but we were able to send 5 patches in earlier this year which has improved managed mode performance greatly. AP mode is broken. Tell me what modes you will be using.

Another option for a usb2 AC600 capable setup would be separate usb devices. For wifi, the following is very good:

https://store.rokland.com/collections/802-11ac-wi-fi-clients-receivers/products/alfa-awus036acs-802-11ac-ac600-dual-band-wifi-usb-adapter-rp-sma

That adapter is based on the 8811au chip which is rock solid, unlike the 8821/11cu. If fact, we are almost ready to upstream a new 8821au driver to be included in the kernel in rtw88. It is a really good driver. When it hits the kernel, it will immediately be the best driver for a Realtek chip in the kernel. That will probably happen for kernel 6.12. I realize the cost is a little high but can likely be negotiated if you are buying more than one.

Another good chip is the mt7610u. The in-kernel driver has been in place since kernel 4.18 and is full features and mature. I can forward possible adapters if you want.

With the wifi only adapters, that leaves finding a really good bluetooth only nano usb adapter and I can help with that if you want. I do think that overall, the most stable, problem free setup would be to go with separate adapters.

So, more questions:

What kind of range do you need?
What is your budget?
Do your need high uptime and stability?
What modes do you need? AP, Managed, Monitor, P2P?

The RasPis are fine for USB2 AC600 operations. The RasPiOS current release should work well.

z0mig, 2024-07-15:
> Let me throw some ideas out:
>
> Your options for good combined adapters that are single-state are somewhat limited. The following is the best one that I know of and I have one: [...]

Wow Nick! Thanks!! <3

We'll take a look at all of these options. I can't explain much about the product we are developing, but to put you in context in a few words. It will be an intelligent 2.4+5+BT frequency scanner.

`
  So, more questions:
  
  What kind of range do you need? --> We can use the mini-USB alpha adapters that you told us about.
  What is your budget? --> We would like the adapters to not cost more than €30.
  Do your need high uptime and stability? --> Yes. It's the most important thing. It will be a device that will be sweeping frequencies for days.
  What modes do you need? AP, Managed, Monitor, P2P? --> AP, Managed and Monitor.
`
As I told you, we are not tied to a specific operating system but right now we are using RaspiOS on the command line. It will be a device "unattended" by the user.

morrownr, 2024-07-15:
> As I told you, we are not tied to a specific operating system but right now we are using RaspiOS on the command line.

I would definitely stick with RasPiOS. It is based on a rock solid Debian 12 distro. There is no other small board maker that supports their hardware with a distro like RasPi does. That is where the other small board makers make a big mistake.

I understand the desire not to share specific information. I often provide consulting to companies that include usb wifi modules and adapters in their products. My advice gets better with more details. I do have an email address available if you need it.

Good luck.