morrownr/8821cu-20210916 #155: Compilation failing

chowbok, 2024-08-03:
Hi, trying to build this on a "Linux from Scratch" build and getting an error. Here's my output:
```
: ---------------------------
: install-driver.sh v20231115
: x86_64 (kernel architecture)
: x86_64 (architecture to send to gcc)
: 8/8 (in-use/total processing units)
: 32471708 (total system memory)
: 6.10.2 (kernel version)
: gcc (version of gcc used to compile the kernel)
: gcc (GCC) 14.1.0
: dkms-3.0.13
: mokutil not installed
: ---------------------------

Checking for previously installed drivers...
Finished checking for and removing previously installed drivers.
: ---------------------------

Starting installation.
Installing 8821cu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8821cu-5.12.0.4
Creating symlink /var/lib/dkms/rtl8821cu/5.12.0.4/source -> /usr/src/rtl8821cu-5.12.0.4
The driver was added to dkms successfully.
: ---------------------------

The kernel is built without module signing facility, modules won't be signed

Building module:
Cleaning build area...
Building module(s)............(bad exit status: 2)
Failed command:
kernelver=6.10.2 ./dkms-make.sh
Error! Bad return status for module build on kernel: 6.10.2 (x86_64)
Consult /var/lib/dkms/rtl8821cu/5.12.0.4/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

The first error in make.log looks like this (I can attach the entire log if you like):
```
CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/core/monitor/rtw_radiotap.o
/var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_mlme_ext.c: In function 'OnAuth':
/var/lib/dkms/rtl8821cu/5.12.0.4/build/core/rtw_mlme_ext.c:2570:41: error: suggest braces around empty body in an 'if' statement [-Werror=empty-body]
 2570 |                                         ;/* TODO: STA re_auth within expire_to */
      |                                         ^
```
Thanks for taking a look.

morrownr, 2024-08-03:
@chowbok 

I have pushed a patch that I think fixes the problem. Can I get a report from you?

$ git pull
$ sudo sh install-driver.sh

Also, it you are interested in testing, read the first message in this issue:

https://github.com/morrownr/8821cu-20210916/issues/115

chowbok, 2024-08-03:
That fixed it. Thanks!

closed 2024-08-03 by chowbok