morrownr/8821cu-20210916 #157: Wifi stops working each ~5 minutes

BurAndBY, 2024-08-21:
Hello! I own a 0bda:c811 Wi-Fi usb dongle. I tried installing Arch on the second PC with exact specs (the first one seemed to work on the same exact dongle without custom drivers) and the Wifi did not work. So I installed this driver for it to work and it works, but once every 
5 minutes it turns off and stops working.

Logs:
NetworkManager: https://0x0.st/XJn1.txt
wpa_supplicant: https://0x0.st/XJnQ.txt

I could not figure out a way to monitor this well, so I used Uptime Kuma, here is what I get:
[image]

BurAndBY, 2024-08-21:
huh, it seems like it's not just every 5 minutes. when i start downloading something, it goes down. example:

[image]

morrownr, 2024-08-21:
Hi @BurAndBY 

Let me recommend that you test with the following driver:

https://github.com/morrownr/8821cu-20210916/issues/115

That repo is what we are using to improve the in-kernel driver. I is a totally different driver that is Linux standards compliant. If you see similar issues with it, then there is an issue with your distro.

Let me know.

BurAndBY, 2024-08-21:
> Hi @BurAndBY
> 
> Let me recommend that you test with the following driver:
> 
> #115
> 
> That repo is what we are using to improve the in-kernel driver. I is a totally different driver that is Linux standards compliant. If you see similar issues with it, then there is an issue with your distro.
> 
> Let me know.

For some reason, after installing the driver you mentioned (with the arch linux dependency) the dongle doesn't show up in nmcli

morrownr, 2024-08-21:
I am seeing reports from Arch users in other repos I have here. Are you on kernel 6.10.2?

BurAndBY, 2024-08-21:
> I am seeing reports from Arch users in other repos I have here. Are you on kernel 6.10.2?

Yes, I am currently on 6.10.5-arch1-1. Should I try LTS?

morrownr, 2024-08-21:
> Should I try LTS?

If it is not 6.10.x, yes. 6.10 is really problematic right now. Patches are flowing but that takes time.

Let me know what happens and which kernel you try.

BurAndBY, 2024-08-21:
> > Should I try LTS?
> 
> If it is not 6.10.x, yes. 6.10 is really problematic right now. Patches are flowing but that takes time.
> 
> Let me know what happens and which kernel you try.

I've installed the driver succesfully(?), but now it just says `device wlp0s29f7u5 not available because device is not available`

morrownr, 2024-08-21:
What are the outputs of:

$ lsmod | grep 88

and 

$ iw dev

and

$ dkms status

morrownr, 2024-08-21:
Oh, tell me which driver you installed. Is it the one in this repo or the new driver I suggested?

BurAndBY, 2024-08-21:
> What are the outputs of:
> 
> $ lsmod | grep 88
> 
> and
> 
> $ iw dev
> 
> and
> 
> $ dkms status

lsmod: https://termbin.com/zuz7
iwdev: https://termbin.com/5eri
dkmss: https://termbin.com/80fo

BurAndBY, 2024-08-27:
I have tried running a clean arch linux install that uses the kernel driver and it still disconnects for 5 minutes on download on wpa-psk networks, but works fine on non-secured.